stnolting/neorv32 #1: ProASIC3 port

salmansheikh, 2020-10-01:
Hi Stephan, 

I used your NEO some years back when developing a CAN Bus core. I loved it. I saw this under RISCV and decided to give it a try. I am porting it to the ProASIC3 starter board. I got it synthesized successfully in LiberoSOC v11.9. Just synplicity but it was bringing out all the wishbone signals. I tried to change the Booleans for the neorv32_top.vhd to the following (cut from file, not all of them) but I still see the signals on the RTL block diagram as I guess they are on the top level and need to be commented out of the entity definition and put in as signals in the body of the architecture. Otherwise lots of pins created and the compiler chokes on the signals not being driven. 
   
    -- External memory interface --
    MEM_EXT_USE                  : boolean := false;  -- implement external memory bus interface?
    IO_GPIO_USE                  : boolean := false;   -- implement general purpose input/output port unit (GPIO)?

salmansheikh, 2020-10-01:
The compiler tools are linux only. Do I need WSL or should I be able to find some pre-built windows executables?

stnolting, 2020-10-01:
When you disable a certain optional module (like the GPIO unit) only the actual core logic gets discarded. Of course the top level ports are still there but there is no more logic driving/reading them.

I would suggest to use a wrapper instead of directly using `neorv32_top.vhd` as top entity.
There are some wrappers available in  [`https://github.com/stnolting/neorv32/tree/master/rtl/top_templates`](https://github.com/stnolting/neorv32/tree/master/rtl/top_templates)

For example, the [neorv32_test_setup.vhd](https://github.com/stnolting/neorv32/blob/master/rtl/top_templates/neorv32_test_setup.vhd) is a very simple wrapper only propagating the UART and GPIO ports to FPGA pins.

> The compiler tools are linux only. Do I need WSL or should I be able to find some pre-built windows executables?

I have tried WSL and it is a bit unstable... I am using [Ubuntu on Windows](https://ubuntu.com/tutorials/ubuntu-on-windows#1-overview) and that works pretty well.

There are pre-built toolchains for Windows out there. Maybe you could use the version from [SiFive](https://www.sifive.com/software), but I haven't tried that yet.

rcaproni, 2020-10-01:
Hi Guys:

About the compiler, I´m using windows only.
I Installed the compiler from here:
https://xpack.github.io/riscv-none-embed-gcc/install/
The version I´m using is 8.3.0-1.2.1
To compile using the makefile, you will need some modification:
- Install gcc riscv using XPACK
- Compile image_gen for windows (I used mingw64) and put in the image_gen
folder
- Edit makefile

# Compiler toolchain
XPACK_TOOLCHAIN ?=
C:/Users/riscv/AppData/Roaming/xPacks/@xpack-dev-tools/riscv-none-embed-gcc/8.3.0-1.2.1/.content
RISCV_TOOLCHAIN ?= $(XPACK_TOOLCHAIN)/bin/riscv-none-embed

Disable image_gen compilation
#$(IMAGE_GEN): $(NEORV32_EXG_PATH)/image_gen.cpp
# @echo Compiling $(IMAGE_GEN)
# @$(CC_X86) $< -o $(IMAGE_GEN)

With these modifications you can use windows to compile and you will have
the .bin file to upload to the board. I´ve been using it for months.

Regards
Rodolfo

salmansheikh, 2020-10-01:
> Hi Guys: About the compiler, I´m using windows only. I Installed the compiler from here: https://xpack.github.io/riscv-none-embed-gcc/install/ The version I´m using is 8.3.0-1.2.1 To compile using the makefile, you will need some modification: - Install gcc riscv using XPACK - Compile image_gen for windows (I used mingw64) and put in the image_gen folder - Edit makefile # Compiler toolchain XPACK_TOOLCHAIN ?= C:/Users/riscv/AppData/Roaming/xPacks/@xpack-dev-tools/riscv-none-embed-gcc/8.3.0-1.2.1/.content RISCV_TOOLCHAIN ?= $(XPACK_TOOLCHAIN)/bin/riscv-none-embed Disable image_gen compilation #$(IMAGE_GEN): $(NEORV32_EXG_PATH)/image_gen.cpp # @echo Compiling $(IMAGE_GEN) # @$(CC_X86) $< -o $(IMAGE_GEN) With these modifications you can use windows to compile and you will have the .bin file to upload to the board. I´ve been using it for months. Regards Rodolfo Em qui., 1 de out. de 2020 às 11:12, Stephan <notifications@github.com> escreveu:
> […](#)
> -- "Quis custodiet ipsos custodes?" ("Quem vigia os vigilantes?") Juvenal - Sátiras livro VI

Thanks but I am using a work computer and don't have elevated privileges so I may go with WSL for now. This is more involved. On my home machine, I can do these easily .Heck my home machine is an Ubuntu. No windows here..

salmansheikh, 2020-10-01:
I got the prebuilt toolchain installed per your datasheet instructions on
WSL for Ubuntu 18.04. When I try to run make check on the blink_led I get
an error message like so:

ssheikh@GSLWL2020070597:/mnt/c/Project/neorv32/sw/example/blink_led$ make
check
Compiling ../../../sw/image_gen/image_gen
make: gcc: Command not found
makefile:153: recipe for target '../../../sw/image_gen/image_gen' failed
make: *** [../../../sw/image_gen/image_gen] Error 127

And when I tried a make install, I got this error. I hope Ubuntu 18.04
isn't too old and the lipbmp.so.3 isn't available. Else, I have to build
from scratch as our WIndows 10 build doesn't allow WSL Ubuntu 20.04 to be
installed

ssheikh@GSLWL2020070597:/mnt/c/Project/neorv32/sw/example/blink_led$ make
all
/opt/riscv/bin/../libexec/gcc/riscv32-unknown-elf/10.1.0/cc1: error while
loading shared libraries: libmpc.so.3: cannot open shared object file: No
such file or directory
makefile:170: recipe for target 'main.c.o' failed
make: *** [main.c.o] Error 1

salmansheikh, 2020-10-02:
I didn't have gcc or g++ installed. I thought they would be installed by default. now the toolchain works and I can build examples. Onwards!!!

closed 2020-10-02 by salmansheikh

stnolting, 2020-10-07:
@rcaproni Nice solution for Windows! I did not know https://xpack.github.io/riscv-none-embed-gcc/install/. I definitively have to try that out.

rcaproni, 2020-10-08:
> @rcaproni Nice solution for Windows! I did not know https://xpack.github.io/riscv-none-embed-gcc/install/. I definitively have to try that out.

Yes, good solution provided by mcu-eclipse guys. Works fine for NEORV.