stnolting/neorv32 #5: ghdl simulation failure

salmansheikh, 2021-01-11:
I tried running the script ghdl_sim.sh on Ubuntu (WSL 18.04) and it gave and error about --max-sstacck-alloc=0' so I tried running it with that option and it output this message after running starting the simulation that at GHDL Bug occurred.

 I was using GHDL 0.35.  I got 4 output files 2 empty output files and one for the uart.out with a single statement. Is this the correct way this simulation should run?

-rw-rw-rw- 1 ssheikh ssheikh 8992 Jan 11 11:44 neorv32-obj93.cf
-rwxrwxrwx 1 ssheikh ssheikh    0 Jan 11 11:45 neorv32.uart.sim_mode.text.out
-rwxrwxrwx 1 ssheikh ssheikh    0 Jan 11 11:45 neorv32.uart.sim_mode.data.out
-rwxrwxrwx 1 ssheikh ssheikh   26 Jan 11 12:01 neorv32.testbench_uart.out
ssheikh@GSLWL2020070597:~/neorv32/sim/ghdl$ more neorv32.testbench_uart.out
Blinking LED demo program

 ghdl -r --work=neorv32 neorv32_tb --ieee-asserts=disable --assert-level=error $SIM_CONFIG
/home/ssheikh/neorv32/sim/ghdl/../../rtl/core/neorv32_cpu.vhd:170:3:@0ms:(assertion warning): NEORV32 CPU CONFIG WARNING! Atomic operations extension (A) only supports >lr.w< and >sc.w< instructions yet.
/home/ssheikh/neorv32/sim/ghdl/../../rtl/core/neorv32_cpu.vhd:179:3:@0ms:(assertion note): NEORV32 CPU CONFIG NOTE: Implementing physical memory protection (PMP) with 2 regions and 65536 bytes minimal region size (granulartiy).
/home/ssheikh/neorv32/sim/ghdl/../../rtl/core/neorv32_wishbone.vhd:141:3:@0ms:(assertion note): NEORV32 PROCESSOR CONFIG NOTE: Implementing external memory interface using STANDARD Wishbone protocol.
/home/ssheikh/neorv32/sim/ghdl/../../rtl/core/neorv32_wishbone.vhd:145:3:@0ms:(assertion note): NEORV32 PROCESSOR CONFIG NOTE: Using BIG-ENDIAN byte order for external memory interface.
/home/ssheikh/neorv32/sim/ghdl/../../sim/neorv32_tb.vhd:297:15:@1342745ns:(report note): NEORV32_TB_UART.TX: B
/home/ssheikh/neorv32/sim/ghdl/../../sim/neorv32_tb.vhd:297:15:@1915955ns:(report note): NEORV32_TB_UART.TX: l
/home/ssheikh/neorv32/sim/ghdl/../../sim/neorv32_tb.vhd:297:15:@2489165ns:(report note): NEORV32_TB_UART.TX: i
/home/ssheikh/neorv32/sim/ghdl/../../sim/neorv32_tb.vhd:297:15:@3062375ns:(report note): NEORV32_TB_UART.TX: n
/home/ssheikh/neorv32/sim/ghdl/../../sim/neorv32_tb.vhd:297:15:@3635585ns:(report note): NEORV32_TB_UART.TX: k
/home/ssheikh/neorv32/sim/ghdl/../../sim/neorv32_tb.vhd:297:15:@4208795ns:(report note): NEORV32_TB_UART.TX: i
/home/ssheikh/neorv32/sim/ghdl/../../sim/neorv32_tb.vhd:297:15:@4782005ns:(report note): NEORV32_TB_UART.TX: n
/home/ssheikh/neorv32/sim/ghdl/../../sim/neorv32_tb.vhd:297:15:@5355215ns:(report note): NEORV32_TB_UART.TX: g
/home/ssheikh/neorv32/sim/ghdl/../../sim/neorv32_tb.vhd:297:15:@5928425ns:(report note): NEORV32_TB_UART.TX:
/home/ssheikh/neorv32/sim/ghdl/../../sim/neorv32_tb.vhd:297:15:@6501635ns:(report note): NEORV32_TB_UART.TX: L
/home/ssheikh/neorv32/sim/ghdl/../../sim/neorv32_tb.vhd:297:15:@7074845ns:(report note): NEORV32_TB_UART.TX: E
/home/ssheikh/neorv32/sim/ghdl/../../sim/neorv32_tb.vhd:297:15:@7648055ns:(report note): NEORV32_TB_UART.TX: D
/home/ssheikh/neorv32/sim/ghdl/../../sim/neorv32_tb.vhd:297:15:@8221265ns:(report note): NEORV32_TB_UART.TX:
/home/ssheikh/neorv32/sim/ghdl/../../sim/neorv32_tb.vhd:297:15:@8794475ns:(report note): NEORV32_TB_UART.TX: d
/home/ssheikh/neorv32/sim/ghdl/../../sim/neorv32_tb.vhd:297:15:@9367685ns:(report note): NEORV32_TB_UART.TX: e
/home/ssheikh/neorv32/sim/ghdl/../../sim/neorv32_tb.vhd:297:15:@9940895ns:(report note): NEORV32_TB_UART.TX: m
/home/ssheikh/neorv32/sim/ghdl/../../sim/neorv32_tb.vhd:297:15:@10514105ns:(report note): NEORV32_TB_UART.TX: o
/home/ssheikh/neorv32/sim/ghdl/../../sim/neorv32_tb.vhd:297:15:@11087315ns:(report note): NEORV32_TB_UART.TX:
/home/ssheikh/neorv32/sim/ghdl/../../sim/neorv32_tb.vhd:297:15:@11660525ns:(report note): NEORV32_TB_UART.TX: p
/home/ssheikh/neorv32/sim/ghdl/../../sim/neorv32_tb.vhd:297:15:@12233735ns:(report note): NEORV32_TB_UART.TX: r
/home/ssheikh/neorv32/sim/ghdl/../../sim/neorv32_tb.vhd:297:15:@12806945ns:(report note): NEORV32_TB_UART.TX: o
/home/ssheikh/neorv32/sim/ghdl/../../sim/neorv32_tb.vhd:297:15:@13380155ns:(report note): NEORV32_TB_UART.TX: g
/home/ssheikh/neorv32/sim/ghdl/../../sim/neorv32_tb.vhd:297:15:@13953365ns:(report note): NEORV32_TB_UART.TX: r
/home/ssheikh/neorv32/sim/ghdl/../../sim/neorv32_tb.vhd:297:15:@14526575ns:(report note): NEORV32_TB_UART.TX: a
/home/ssheikh/neorv32/sim/ghdl/../../sim/neorv32_tb.vhd:297:15:@15099785ns:(report note): NEORV32_TB_UART.TX: m
/home/ssheikh/neorv32/sim/ghdl/../../sim/neorv32_tb.vhd:295:15:@15672965ns:(report note): NEORV32_TB_UART.TX: (13)
/home/ssheikh/neorv32/sim/ghdl/../../sim/neorv32_tb.vhd:295:15:@16246165ns:(report note): NEORV32_TB_UART.TX: (10)

******************** GHDL Bug occured ****************************
**Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: 0.35 (Ubuntu 0.35+dfsg-1~build3) [Dunoon edition]
Compiled with GNAT Version: 7.3.0
Target: x86_64-linux-gnu
In directory: /home/ssheikh/neorv32/sim/ghdl/
Command line:
/usr/bin/ghdl-mcode -r --work=neorv32 neorv32_tb --ieee-asserts=disable --assert-level=error
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : grt-processes.adb:770 overflow check failed
Call stack traceback locations:
0x7f51e547c42e 0x7f51e547cbd4 0x7f51e547ce4a 0x7f51e547d02c 0x7f51e5412a4a 0x7f51e54bc4bd 0x7f51e56f2fea 0x7f51e56df51f 0x7f51e56d0dcd 0x7f51e56fcb9e 0x7f51e541266f 0x7f51e3631b95 0x7f51e5411498 0xfffffffffffffffe**
******************************************************************

stnolting, 2021-01-12:
A stack overflow error in GHDL can occur if you are running out of memory due to too many simulated signals. GHDL uses several bytes for each signal and you might get problems when simulating large memories. There are simulation-optimized versions of the IMEM and DMEM components, that use constants/variables instead of the 'signals' for the memory array: `sim/rtl_modules`

The default testbench features a simulated UART receiver. Eachprintable char that is received from the processor's UART will be printed to the console and also written to `neorv32.testbench_uart.out`. 

If the UART is in **simulation mode** (setting bit 12 in the UART control register), the actual UART transmitter logic will be disabled. All data written to the UART's TX data register will be "printed" using VHDL text.io (which is way faster than using the testbench simulated UART receiver):
* The printable ASCII char will be printed to the console and also to `neorv32.testbench_uart.out`.
* The full 32-bit word written to the UART'S TX data register will be printed as hex word to `neorv32.uart.sim_mode.data.out`.

You can automatically activate the simulation mode by using `USER_FLAGS+=-DUART_SIM_MODE` when compiling your applications.

stnolting, 2021-01-21:
Is your simulation now working as expected? 🤔

salmansheikh, 2021-01-21:
You can close this. I haven't been able to run it yet. I was thinking more
of getting the Xilin Arty or Terasic DE0 board from work to put the SOC on
them and take it for a whirl.

Salman

stnolting, 2021-01-21:
I'm also working with Arty and E0-nano. So if you need any help, feel free to get in contact again. :wink:

closed 2021-01-21 by stnolting