stnolting/neorv32 #6: How can I easily add IP for an SRAM, EEPROM and MRAM

salmansheikh, 2021-01-21:
I have a custom daughter board for a Microsemi A3P3-Starter Kit development board (ProASIC3E). I have a 32Kx8 EEPROM, 512Kx32 SRAM and a 2Mx8 MRAM. Are there any IP for connecting these easily (opencores.org or other) so I could use these with the ProASIC3 board.

stnolting, 2021-01-21:
If your memory components provide a **serial interface** (I²C or SPI) you can directly connect them to the processors TWI or SPI modules, respectively. Obviously, this will *not* map the memories into the CPU's address space.

If your memories provide **parallel interfaces**:

I am not familiar with the Microsemi synthesis framework. But I suppose the tools provide an FPGA-optimized IP library that might also feature IP blocks to interface your memories. If these IPs support an AXI infrastructure you can use the NEORV32 AXI4-Lite wrapper (`/rtl/top_templates/neorv32_top_axi4lite.vhd`) that provides an AXI4-Lite master.

Right now there is no specific memory interface IP on GitHub/opencores/wherever that I can recommend as I am (mostly) using the IPs provided by the FPGA vendors. However, there are a lot of IPs out there. But I know that there are some people, who have successfully connected open-source memory controller - but I do not know which one they have used :wink:

However, as long as these IPs support the (classic/standard) Wishbone protocol it should be no problem to attach them to the NEORV32. Check out the processor's test bench (`sim/neorv32_tb.vhd`) which features a minimalist Wishbone "interconnection fabric". 

Last but not least: Interfacing SRAM is pretty simple and I assume that your MRAM might be as simple as well. :wink:

salmansheikh, 2021-01-21:
They are all parallel. Most of the microsemi IP are AHB/AHB-Lite so not
sure how easy that is to connect to a WB bridge.

stnolting, 2021-01-21:
Seems like each FPGA vendor has its own taste of on-chip buses...
Intel -> Avalon
Xilinx -> AXI
Lattice -> Wishbone
Microsemi -> AHB

Anyway. Looking at the AHB-Lite specifications AHB seems to be quite similar to AXI4-lite (obviously, its AXI's predecessor). The NEORV32 Wishbone-to-AXI4-Lite bridge is pretty simple, so it shouldn't be a big deal to write/convert a simple Wishbone-AHB bridge to start with.

[github.com/vfinotti/ahb3lite_wb_bridge/blob/master/wb_to_ahb3lite.v](https://github.com/vfinotti/ahb3lite_wb_bridge/blob/master/wb_to_ahb3lite.v) might be a good starting point - maybe it works out of the box :thinking:

closed 2021-03-10 by stale[bot]

GTrannoy, 2021-06-02:
> [github.com/vfinotti/ahb3lite_wb_bridge/blob/master/wb_to_ahb3lite.v](https://github.com/vfinotti/ahb3lite_wb_bridge/blob/master/wb_to_ahb3lite.v) might be a good starting point - maybe it works out of the box 🤔

=> I would have an AHB interface directly instead of AXI4, because it is lighter, but 'wb_to_ahb3lite.v' doesn't work and I find nothing else which work easily.
In simulation, 'wb_to_ahb3lite.v' seems working, but in practice the core freeze when I want to access external memory.

stnolting, 2021-06-02:
If you need a really "light" interface, then I would suggest to use the native Wishbone interface of the NEORV32. It is much simpler than AXI4-Lite and quite similar to AHB.

I have not worked with AHB for a long time, but as far as I can remember a Wishbone-to-AHB bride might only require some simple logic for controlling AHB's address and data phases (but maybe this is not really required at all).

Compare the "Read transfer" diagram from https://developer.arm.com/documentation/ihi0033/a/Transfers/Basic-transfers
with the (NEORV32's) Wishbone read transfer:

[image]

The Wishbone interface of the NEORV32 can be configured to wait a specific number of cycles (`MEM_EXT_TIMEOUT` generic) before it invalidates the transfer. You can even configure an unlimited number of cycles to wait for by `MEM_EXT_TIMEOUT` = 0.
So it should be no problem to issue AHB's address and data phases within one single Wishbone transfer.

By the way, please open a new issue or discussion if you have further questions :+1:

GTrannoy, 2021-06-02:
Thank you for your answer.

Yes, I see it is very similar and in simulation it seems to be ok, but something block the core. And I checked the MEM_EXT_TIMEOUT but is set at '0'.

I will investigate and trying to do my own bridge. Probably some mistakes I did...

I will see that and maybe openning another discussion.

Thx