stnolting/neorv32 #8: The project with the core is not going to Vivado 2019.1

diy4hobby, 2021-02-13:
**Describe the bug**
I tried to reproduce the project from the description in Vivado 2019.1, but got parameter type conversion errors:
`[IP_Flow 19-1837] Failed to convert long value '0' to HDL value. Missing bit length for conversion.`
`[IP_Flow 19-3814] Failed to get HDL value for model parameter 'IO_CFS_CONFIG'.`
`[IP_Flow 19-3814] Failed to get HDL value for model parameter 'HW_THREAD_ID'.`
`[IP_Flow 19-3814] Failed to get HDL value for model parameter 'IO_CFS_CONFIG'.`
Perhaps this is due to the fact that my block project is converted to Verilog (and yours to VHDL).

**To Reproduce**
To reproduce it, it is enough to specify in the Vivado settings that convert the block project to Verilog.

**Expected behavior**
For compatibility with different languages, it would like to fix this problem.

Thank you for the project!

AWenzel83, 2021-02-13:
I have the same problem with vhdl, when trying to generate the output products from a block design using the axi top module. 
It works after I open the elaborated design, then it synthesises the design, and it works as intended.

diy4hobby, 2021-02-13:
Thank you, Andreas Wenzel!
Indeed, your method works!
But I still think that this problem should be solved, so that the synthesis and implementation of the project with the core would take place without additional actions.

AWenzel83, 2021-02-13:
Yes you are right, it should really work without this workaround.

stnolting, 2021-02-13:
Thanks for the hint! I haven't noticed that yet.

Seems like Vivado cannot deal with aggregates using just `others` for the default initialization of generics:
https://github.com/stnolting/neorv32/blob/809af1f9804badaf6248bf0c422c01fb43afc084/rtl/top_templates/neorv32_top_axi4lite.vhd#L90

Using a fixed-size initialization like `x"00000000"` instead of `(others => '0')` might solve this problem.

I will evaluate on that and fix it ASAP :wink:

diy4hobby, 2021-02-14:
It seems that the problem is not fixed. I created a completely new core with new sources and getting errors.
[image]

stnolting, 2021-02-14:
That is strange. I have also created a new IP-packaging project, a new IP repository and also a new block design using Vivado 2019.2. So far, I do not get any errors.

Which step issued the errors you have mentioned? During "generate output products" or already during "validate design"?

stnolting, 2021-02-17:
@diy4hobby are you still encountering the same error?

stnolting, 2021-03-28:
I am closing this. Feel free to open a new issue if you are still having troubles with the IP packaging. 😉

closed 2021-03-28 by stnolting