stnolting/neorv32 #11: UART RTS / CTS HW Flow Control

vhdlnerd, 2021-02-19:
It would be great if the UART ports supported HW flow control.  I would like to run the UARTs at much higher baud rates and transfer large amounts of data. This will require flow control.

BTW: Nice work on this project!  I had it up and running on a Digilent Nexys A7 board quickly without issues.

stnolting, 2021-02-20:
Interesting idea!

Obviously, the hardware flow control signals could be implemented using the GPIO ports and some software overhead. But this overhead might compromise some of the "very high" data throughput.

So having dedicated control signals seems to be a nice solution. I have checked the RTS / CTS specifications and if I got them right
`neorv32.uart.RTS_o` (output) would indicate the UART's receiver being ready to receive another byte while `neorv32.uart.CTS_i` (input) would indicate the UART's transmitter is allowed to transmit another byte, correct?

Right now I am not sure about the polarity of these signals, but this isn't really a big deal.

vhdlnerd, 2021-02-22:
Yes, that is the way I believe it works: CTS is an input and RTS is an output. I think they are high active signals (at least to the TTL level).
Thanks for considering this enhancement.

stnolting, 2021-02-22:
I have added (experimental) RTS/CTS hardware flow control with the last repository update. Since I have no full-scale UART-USB adapter available right now, I could only simulate the flow control so far - but it looks pretty good.

Maybe you can give some feedback if you plan to test the new features. :wink:

By the way, RTS and CTS seem to be low-active (they *are* low-active in the current NEORV32 version).

vhdlnerd, 2021-02-23:
I'll try it out this weekend and give you any feedback.

vhdlnerd, 2021-02-28:
It worked great for me on my Nexys A7 board -- RTS/CTS worked as expected.
I'm now running the UART at 460,800 baud, in the bootloader and app code, with no issues.

stnolting, 2021-03-01:
I'm glad to hear that! 👍

closed 2021-03-01 by stnolting