stnolting/neorv32 #24: neorv32_wdt component not VHDL-08 compliant

tmeissner, 2021-05-12:
During experiments to synthesize the project with GHDL & Yosys I found that the Watchdog timer unit cannot analyzed when VHDL-08 standard is used. The record type for the control register uses the now reserved keyword `force`:

```vhdl
  -- control register --
  type ctrl_reg_t is record
    enable  : std_ulogic; -- 1=WDT enabled
    clk_sel : std_ulogic_vector(2 downto 0);
    mode    : std_ulogic; -- 0=trigger IRQ on overflow; 1=trigger hard reset on overflow
    rcause  : std_ulogic; -- cause of last system reset: '0' = external, '1' = watchdog
    reset   : std_ulogic; -- reset WDT
    force   : std_ulogic; -- force action
    lock    : std_ulogic; -- lock control register
  end record;
```

Changing `force` to a non-reserved word resolves the problem and neorv32 can be synthesized with Yosys & GHDL.
Nice project by the way 🙂

stnolting, 2021-05-13:
Thanks for the hint! I will fix that.

> If you're interested I could contribute to synthesize the project with Yosys & GHDL.

That would be great! Experimenting with the GHDL Yosys plugin is already on my to-do list, but I haven't really looked at it yet...

> Nice project by the way 🙂

Thanks again 😉

tmeissner, 2021-05-13:
I have a IceZero board with a Lattice ICE40HX on it which is supported by the open source flow. I will try to implement neorv32 on it and make a contribution to the `boards/` folder.

umarcor, 2021-05-14:
I have an icestick (maybe too small), a FOMU (UP5K) and an EOS S3 (maybe too small). I have the icestick and FOMU toolchains ready. Either on Windows (https://hdl.github.io/MINGW-packages/) or through containers (https://hdl.github.io/containers). I helped setup CI in [antonblanchard/microwatt](https://github.com/antonblanchard/microwatt). I'd be happy to jump in and help enhance the CI plumbing in this regard, after Torsten is done with his contributions (so that we don't overlap).

Moreover, in the mid term, I would be interested in testing neorv32 on PYNQ-Z1, microzed and KC705.

@stnolting I pinged you in [gitter.im/vhdl/General](https://gitter.im/vhdl/General). Would be nice to have a chat!

stnolting, 2021-05-14:
@tmeissner 

> I have a IceZero board with a Lattice ICE40HX on it which is supported by the open source flow. I will try to implement neorv32 on it and make a contribution to the boards/ folder.
The ICE40HX device is pretty limited regarding embedded RAM. Maybe I simply try to port the existing UPduino v3 project to the open source implementation flow.

I'm very excited to hear that and I am curious what you will come up with! 👍 

@umarcor 

> I have an icestick (maybe too small), a FOMU (UP5K) and an EOS S3 (maybe too small). I have the icestick and FOMU toolchains ready. Either on Windows (https://hdl.github.io/MINGW-packages/) or through containers (https://hdl.github.io/containers). I helped setup CI in antonblanchard/microwatt. 

Now that's a great collection of EDA tools I was not aware of. I definitely have to check things out!

> Moreover, in the mid term, I would be interested in testing neorv32 on PYNQ-Z1, microzed and KC705.

Should not be too hard to port one of the existing Xilinx setups - but feel free to start a new issue or discussion if you are encountering any problems.

> I pinged you in gitter.im/vhdl/General. Would be nice to have a chat!

Cool! Thank you very much! Unfortunately, I have never used gitter before...
 I'm not at home right now but as soon as I am back I will dig into it and respond :wink:

stnolting, 2021-05-16:
I think this can be closed?

tmeissner, 2021-05-16:
Yes, this can be closed.

closed 2021-05-16 by tmeissner