stnolting/neorv32 #30: neorv32_boot_rom.vhd:69:12: index not within bounds

tmeissner, 2021-05-20:
With the latest commits a bug was introduced in neorv32_boot_rom unit:

```
../../rtl/core/neorv32_boot_rom.vhd:69:12: index not within bounds
      mem_v(i) := init(i)
           ^
```

It seems that the range of `mem_v` is smaller than the range of `init` in the following code snippet. If type `boot_img_t`range is increased by one, it works. 

```vhdl
  -- local types --
  type boot_img_t is array (0 to BOOTROM_SIZE/4-1) of std_ulogic_vector(31 downto 0);

  -- init function --
  -- impure function: returns NOT the same result every time it is evaluated with the same arguments since the source file might have changed
  impure function init_boot_rom(init : bootloader_init_image_t) return boot_img_t is
    variable mem_v : boot_img_t;
  begin
    mem_v := (others => (others => '0'));
    for i in 0 to init'length-1 loop -- init only in range of source data array
      mem_v(i) := init(i);
    end loop; -- i
    return mem_v;
  end function init_boot_rom;
```

I'm not a git expert, but I assume there is some magic git command to find out which commit introduced the bug.

stnolting, 2021-05-20:
Oh crap! 😄 
I had this problem way too often! The bootloader code occupies exactly 4096 bytes - so it fits into the boot ROM. But the "image" file - basically a `case` with 4096 `when` entries - also provides an `other` entry. I really need to fix the image generator...
As a dirty workaround, I will remove a `nop` from crt0 making the bootloader image 4095 bytes large.

stnolting, 2021-05-20:
Should be fixed now. 🤔

commit 2c1e75b9 (2021-05-20): [bootloader image] reduced size to fix #30