stnolting/neorv32 #34: Handle riscv-arch-test as a git submodule

umarcor, 2021-05-21:
Rather than a feature request, this is a topic for discussion. It seems that [riscv/riscv-arch-test](https://github.com/riscv/riscv-arch-test) might be handled as a submodule in subdir `riscv-arch-test/riscv-arch-test` or `riscv-arch-test/work/riscv-arch-test`. That would provide better control over the version of it that is retrieved. @stnolting any reason not to use submodules?

stnolting, 2021-05-21:
The current setup of the RISC-V architecture tests is some kind of *leftover* from the time where I was porting `riscv-arch-test` to verify the RISC-V compliance of the CPU. Including this as a submodule seems to be good idea. 👍

umarcor, 2021-05-25:
riscv-arch-test was added as a submodule in https://github.com/stnolting/neorv32/commit/a9f80b2b324b850473ee45b1a3cc3a9cab34e2e0. Hence, I'm closing this issue.

@stnolting you might want to add a dependabot configuration file such as https://github.com/im-tomu/fomu-workshop/blob/master/.github/dependabot.yml. That will check the upstream of the submodule periodically (you set the interval) and it will open PRs for having it updated.

closed 2021-05-25 by umarcor