stnolting/neorv32 #58: mtime_o could be instable

GTrannoy, 2021-06-04:
You have export the mtime_o to top instance and I thank you for that ! It helps me a lot anf avoid to add a timer in my project.

But I observed instabilities, probably because 'time_o' in the file "neorv32_mtime.vhd" is not in a synchronous process...
Although, I read it in a synchronous process. It is always when the msb words 'mtime_hi' updates that it read a bad value.

To validate a good reading whithout changing something in the core, I added a process to check mtime_o change only by '1' between two clock ticks. But it is a little messy to add this process to validate the ntime output.

I don'k know why it is this instability... Maybe putting the line 167 (' time_o <= mtime_hi & mtime_lo(31 downto 00); ') in a snchronous process ?

stnolting, 2021-06-04:
What do you mean with "instabilities"?

The 64-bit MTIME counter is split in two 32-bit counter (timing issues), so there might be an _inconsistency_ when the lower word overflows into the higher word as the carry propagation requires an additional cycle.

Let me illustrate that:
```
MTIME
0x0000.0000_ffff.fffe
0x0000.0000_ffff.ffff
0x0000.0000_0000.0000 <- inconsistency
0x0000.0001_0000.0001
```
For the CPU that is no deal at all as it reads/writes MTIME only in chunks of 32-bit.

Is that what you mean with "instability"?

stnolting, 2021-06-04:
Should be fixed now :wink:🚀

GTrannoy, 2021-06-05:
Yes, that is this inconsistency which disturb other modules in my project.
Thank you if you have fixed it !

GTrannoy, 2021-06-09:
The problem of inconsistency is not corrected.

I have made some test with testbench, and the most simple is just to put the line with 'mtime_lo_ovfl' outside the process, with the mtime_lo increment:

```
  -- mtime.time_LO increment --
  mtime_lo_nxt <= std_ulogic_vector(unsigned('0' & mtime_lo) + 1);
  mtime_lo_ovfl(0) <= mtime_lo_nxt(32); -- overflow (carry) ```

And with this modification, 'mtime_lo_ovfl' is updated before the  same clock cycle which update 'mtime_lo' and 'mtime_hi'.

stnolting, 2021-06-09:
> And with this modification, 'mtime_lo_ovfl' is updated before the same clock cycle which updates 'mtime_lo' and 'mtime_hi'.

Right, by this the LOW and HIGH words of the mtime counter are always "sync". The problem here is that this will introduce a **_64-bit_** carry chain, which will reduce maximum clock frequency. That is why I split that up into two 32-bit carry chains (one for the HIGH word, one for the LOW word) that are coupled via the registered `mtime_lo_nxt(32)`.

> I have made some tests with testbench,

Could you share or specify your simulation scenario?

I have also done some simulation. I have set mtime to `0x00000000.FFFFFF00` to generate a carry from the LOW word to the HIGH word after 256 cycles. The waveform below shows this carry. `mtime_lo` and `mtime_hi` are out of sync as there is one cycle "offset" due to the registered carry. However, the processor's `mtime_o` signal uses a simple register buffer for the lower 32-bit to re-sync HIGH and LOW words _in that signal_ (white signal in waveform).

[image]

There is no inconsistency in `mtime_o`. At least I cannot find any. 🤔

GTrannoy, 2021-06-09:
> I have also done some simulation. I have set mtime to `0x00000000.FFFFFF00` to generate a carry from the LOW word to the HIGH word after 256 cycles.

This is strange, I have done the same and I obtain with ModelSim the datagram which shows the inconsistency:
[image]

> The problem here is that this will introduce a **_64-bit_** carry chain, which will reduce maximum clock frequency.

Ok. I didn't take care of that when I suggested the modification.
Anyway, with modification I wrote before, I have now this datagram with correct transition:
[image]

Moreover, Libero (IDE for Microsemi fpga) can synthesize and I don't have the error again on my real fpga. 

> Could you share or specify your simulation scenario?

This is simple, I have just specify the command  'neorv32_mtime_set_time(   0x00000000FFFFFF00ULL);' before infinite loop in 'main'. And after I have only somme reads/writes continuously in the loop.

I don't now we have not the same behavior...

umarcor, 2021-06-09:
We've seen mismatches between GHDL and ModelSim due to delta cycle ordering issues: VUnit/vunit#692. Might this be the case here?

stnolting, 2021-06-09:
Oh wait, I think this is just a misunderstanding 😄 

I think you are checking `time_o` in your testbench:
https://github.com/stnolting/neorv32/blob/74ee02612e7f484c1ce077ec104f0560f9a970f3/rtl/core/neorv32_mtime.vhd#L55-L56

`time_o` is the output of the MTIME **module** and this output does **not** provide a synchronization for LOW and HIGH words since this is irrelevant for the CPU.

I was talking about the `mtime_o` output of the **whole processor**:
https://github.com/stnolting/neorv32/blob/74ee02612e7f484c1ce077ec104f0560f9a970f3/rtl/core/neorv32_top.vhd#L187-L189

`mtime_o` provides the "low-word synchronization" that is implemented in `neorv32_top.vhd` (not in the MTIME module itself):
https://github.com/stnolting/neorv32/blob/74ee02612e7f484c1ce077ec104f0560f9a970f3/rtl/core/neorv32_top.vhd#L944-L961

GTrannoy, 2021-06-09:
Yes !!! I understand what happens !
You are right, I was checking `time_o`.

And I have the problem because I updated only `neorv32_mtime.vhd` but no the top instance.
And I just saw you have adding the process `mtime_sync` to handle with this problem...

I am sorry, it is my mistake ! 😕

closed 2021-06-09 by GTrannoy

stnolting, 2021-06-09:
> I am sorry, it is my mistake ! 😕

Absolutely no problem.
I'm happy that you setup works now! :tada: :+1: