stnolting/neorv32 #77: Use ProcessorTop templates in non osflow examples

umarcor, 2021-06-16:
Currently, the non osflow examples (radiant, quartus) do not use a ProcessorTop or SystemTop template. Quartus is trying to use `neorv32_test_setup.vhd` (which was renamed and does not exist) and radiant is instantiating the NEORV32 top directly.

It would be nice to update these examples for using `neorv32_ProcessorTop_Test` and `neorv32_ProcessorTop_UP5KDemo` (or MinimalBoot).

Vivado examples use `neorv32_ProcessorTop_Test` already.

umarcor, 2021-06-16:
While I acknowledge that Minimal vs MinimalBoot might be confusing, the main idea is that Minimal has the bootloader disabled and no other output than the PWM. Conversely, in MinimalBoot the usage of the bootloader is optional, as it is connecting the UART or GPIO. Therefore, I think that "indirectboot" and "directboot" are not descriptive enough. The purpose of MinimalBoot is to be used for both indirect and direct boot. It is correct that Minimal is direct only.

Other than that, I think that Tiny, Small, Medium, etc. might be as good as any other naming you pick.

> maybe we could add _wishbone and _axilite to highlight connectivity if a certain setup provides a bus interface

We are already doing that: https://github.com/stnolting/neorv32/blob/master/rtl/templates/system/neorv32_SystemTop_axi4lite.vhd. So, yes.

stnolting, 2021-06-16:
I was just thinking out loud 😅 

> Other than that, I think that Tiny, Small, Medium, etc. might be as good as any other naming you pick.

I am still not sure about this...
Anyway, I think it is a good idea to change the current Intel and Xilinx setups so they use one of the template instances. 👍

stnolting, 2021-06-29:
Fixed ✔️ 

> Other than that, I think that Tiny, Small, Medium, etc. might be as good as any other naming you pick.

I am still struggling with this 😅 but that might be stuff for a different issue.

I think this can be closed now.

closed 2021-06-30 by umarcor