stnolting/neorv32 #85: Issue in uploading .elf file into Altera DE2 board

karoonlee, 2021-06-20:
Hi, 
I have problems in compiling when using Quartus 20.1 prime lite, but it didn't show any error message when using 16.1 lite version😅. So currently I am using Quartus16.0 lite version, I can load the .sof file into Altera DE2-115 board, however I could not find a way to load the .elf file into FPGA board. 
Must appreciate if can get anyone of you to help help. 
Thanks a lot i say to you all.
Regards,
Lee.

stnolting, 2021-06-21:
Hey there!

> however I could not find a way to load the .elf file into FPGA board.

So you want to upload a software executable, right?

Basically, there are two different ways (depending on the configuration of the processor):
1. Upload an executable via a UART connection (after programming your FPGA): https://stnolting.github.io/neorv32/ug/#_uploading_and_starting_of_a_binary_executable_image_via_uart
2. Install an executable to memory during synthesis (before programming your FPGA): https://stnolting.github.io/neorv32/ug/#_installing_an_executable_directly_into_memory

If your FPGA board provides a UART interface, then I would recommend option 1.

Are you using one of the pre-configured templates as top entity or do you instantiate the processor's top into you design?

stnolting, 2021-06-28:
Do you still have troubles uploading the executable?

karoonlee, 2021-06-30:
> Do you still have troubles uploading the executable?

Hi, Thank You so much for the concern. I appreciate it a lot. So sorry for the late reply, recently I was busy with my school assignment as my semester is about to end.
I am using Altera De2 board, and I using Quartus to compile and upload the program. First i was trying to upload executable file via UART, but i can't the UART connection inside Quartus.

Then I tried the second method, but only single LED is blinking as the output. Is this the correct output?

stnolting, 2021-06-30:
> So sorry for the late reply,

So worries! :wink:

> 

> I am using Altera De2 board

Are you using this board? [Terasic Altera DE2-115](https://www.terasic.com.tw/cgi-bin/page/archive.pl?Language=English&CategoryNo=139&No=502&PartNo=1)

If you are using [`rtl/templates/processor/neorv32_ProcessorTop_Test.vhd`](https://github.com/stnolting/neorv32/blob/master/rtl/templates/processor/neorv32_ProcessorTop_Test.vhd) as top entity, you can use the `uart0_txd_o` and `uart0_rxd_i` signals for the establishing a UART connection.

According to the DE2-115 board's user manual there is a RS232 transceiver chip available, which you can use: `uart0_txd_o` -> `G9`, `uart0_rxd_i` -> `G12`. Of course this requires an external RS232 adapter / cable for your computer

[image]

> Then I tried the second method, but only single LED is blinking as the output. Is this the correct output?

This is the internal bootloader blinking a simple heart-beat LED and waiting for a program to be send via UART.

If you want to "directly install" an application into the processor's internal memory (so, no bootloader, no upload via UART), you will need to disable the bootloader. In [`rtl/templates/processor/neorv32_ProcessorTop_Test.vhd`](https://github.com/stnolting/neorv32/blob/master/rtl/templates/processor/neorv32_ProcessorTop_Test.vhd) you can disable the bootloader by changing the line

```vhdl
INT_BOOTLOADER_EN => true,
```

to

```vhdl
INT_BOOTLOADER_EN => false,
```

To compile and "install" an application, like the [`blink_ed`](https://github.com/stnolting/neorv32/tree/master/sw/example/blink_led), just execute `make clean_all install`. Without the bootloader, the program will be executed directly after reset.

karoonlee, 2021-06-30:
Hi, Thank You for helping. 
> Are you using this board? [Terasic Altera DE2-115](https://www.terasic.com.tw/cgi-bin/page/archive.pl?Language=English&CategoryNo=139&No=502&PartNo=1)

Yes.

> Of course this requires an external RS232 adapter / cable for your computer

Unfortunately, I don't have RS232 right now. 

> like the [`blink_ed`](https://github.com/stnolting/neorv32/tree/master/sw/example/blink_led),

It's Working. Thank You so much. 

I have one more question, if I want to run the [hellworld](https://github.com/stnolting/neorv32/blob/master/sw/example/demo_pwm/main.c) ,  Where will the "Hello world! " be printed? Is it on a screen through VGA connection, or the LED?

stnolting, 2021-06-30:
> Unfortunately, I don't have RS232 right now.

I do not understand why they keep making FPGA dev boards with RS232 interface but without any UART-USB chip... 😄 

Anyway, if you have any USB-UART converter (for example an FTDI) around, you could connect that to some user pins on the board. Intel also provides a JTAG-UART IP so you can have a virtual UART instance accessible via the programming USB port, but I have no experience with that.

Another option would be to use the on-chip debugger (via JTAG) to track the UART output. But that is quite tricky.

> It's Working. Thank You so much.

Good to hear! 👍 

> I have one more question, if I want to run the hellworld , Where will the "Hello world! " be printed? Is it on a screen through VGA connection, or the LED?

Unfortunately, this example program also uses the UART for printing everything.

The DE2-115 provides some nice on-board peripherals. You could use the processor's GPIO output pins to drive those. Especially the LCD is cool. This could be used as an UART alternative for printing text - requiring "just" some software :wink:

karoonlee, 2021-06-30:
> requiring "just" some software

Please could you provide me the name of the software or any keyword so that I get the info from google? I want to try, I feel it's really cool😁
How about the [hex_viewer}(https://github.com/stnolting/neorv32/tree/master/sw/example/hex_viewer) ? I just tried, but what i get the output from hex[0] to hex[6] is always '1'. Although I pressed reset, it's still "1" . Any details of the setting that I should change?😵

stnolting, 2021-06-30:
> Please could you provide me the name of the software or any keyword so that I get the info from google? I want to try, I feel it's really cool😁

Check out the data sheet of your FPGA board: https://www.terasic.com.tw/cgi-bin/page/archive.pl?Language=English&CategoryNo=139&No=502&PartNo=4
The LCD uses a simple HD44780 controller, which is a very very common LCD controller (it even has a wikipedia article: https://en.wikipedia.org/wiki/Hitachi_HD44780_LCD_controller)

There are several example codes/libraries out there - so you don't have to worry about that. Most of them are from "Arduino Space", but there are also some general purpose libraries available. And even if not - the interface protocol is really simple :wink:

For example, this one looks quite promising: https://github.com/Matiasus/HD44780

You just need to define low-level functions for writing a data or command byte to the LCD (so, writing 8 bit data to the GPIO port and toggling the according control signals like "enable") and for some more general things like generating exact timings (busy wait).

> How about the [hex_viewer}(https://github.com/stnolting/neorv32/tree/master/sw/example/hex_viewer) ? I just tried, but what i get the output from hex[0] to hex[6] is always '1'. Although I pressed reset, it's still "1" . Any details of the setting that I should change?😵

Most of the provided example programs use the UART for user interaction. The hex-viewer program is something like a "memory inspection tool" that allows to read/write the processor address space via UART.

stnolting, 2021-07-24:
This can be closed I think.
Feel free to open a new issue or start a new [discussion](https://github.com/stnolting/neorv32/discussions) if you have further questions (or troubles with setting up the core). :wink:

closed 2021-07-24 by stnolting