stnolting/neorv32 #94: HDL documentation (teroshdl-documenter?)

stnolting, 2021-06-30:
Currently, there is no _explicit_ documentation of the (V)HDL source files beyond a lot of in-code comments. I have tried using DOXYGEN also for the HDL files - but that is not really pretty: the actual source files would be stuffed with DOXYGEN macros and the resulting documentation is somehow _special_... Since DOXYGEN is targeting software, this is ok.

Recently, I stumbled across this: https://github.com/TerosTechnology/teroshdl-documenter-demo
They even used this project as an example: https://terostechnology.github.io/teroshdl-documenter-demo/NEORV32_doc/index.html :heart: And I really have to say: this looks very good! Finally a nice (and pretty!) way of documenting HDL sources :+1:

Does anyone already have some experience with @TerosTechnology and especially with their documenter? Is there any option (yet) to include that into a Github workflow? 🤔

umarcor, 2021-06-30:
The main limitation for any documentation generator such as Teros' is parsing VHDL. There are very few usable VHDL parsers, and almost none of them has complete support for VHDL 2008. I believe that teros has an internal JS parser based on regular expressions, and it also uses VUnit for extracting the dependencies. VUnit's parser is also based on regular expressions. It is quite complete and it works reasonably well, but it has some limitations. In fact, replacing the parser in VUnit with rust_hdl or pyVHDLParser has been discussed before. Let me know if you want me to find some references to discussions about open source VHDL parsers.

The summary is that we have a new tool in the party: pyVHDLModel and pyGHDL.dom. pyVHDLModel was split from pyVHDLParser in Dec 2020. Integration into pyGHDL started in Jan 2021 and it is almost complete. @Paebbels has been working so hard on that during the last weeks. As explained in [VHDL/pyVHDLModel](https://github.com/VHDL/pyVHDLModel), some of the use cases that pyGHDL.dom is expected to cover are precisely replacing the dependency analyser in VUnit, the symbol data extraction in Symbolator and providing a VHDL domain for Sphinx. [VHDL/sphinx-vhdl](https://github.com/VHDL/sphinx-vhdl) is a placeholder for that.

Currently, NEORV32 is being used as one of the projects for testing pyGHDL.dom (see results of the master branch in https://ghdl.github.io/extended-tests/ and the development branch in https://paebbels.github.io/extended-tests/). Moreover, I do have a branch in my fork of this repo. See https://github.com/umarcor/neorv32/commits/pyVHDLModel and https://github.com/umarcor/neorv32/runs/2953525135?check_suite_focus=true#step:5:460. I didn't create an issue for this topic yet, because I'm waiting for the latest enhancements to pyGHDL.dom :wink:.

I believe that the most sensible way forward is for TerosHDL and VUnit to use pyGHDL.dom. However, I know there are users of VUnit who don't use GHDL. By the same token, I think that TerosHDL developers were trying to provide a JS only solution, which does not rely on Python or compiled dependencies. Therefore, the usage of pyGHDL.dom might need to be opt-in in both cases, and it might not be a priority for the maintainers.

Anyway, I think I can enhance [TerosTechnology/teroshdl-documenter-demo](https://github.com/TerosTechnology/teroshdl-documenter-demo) for providing it as a GitHub Action (similarly to [VUnit/vunit_action](https://github.com/VUnit/vunit_action)). Then, we can add it as an step in https://github.com/stnolting/neorv32/blob/master/.github/workflows/Documentation.yml and have the result pushed to GitHub Pages along with the other pieces. Alternatively, we could just copy their workflow, but I think it'd be better for the community to provide it as a "clean" step. What do you think?

/cc @qarlosalberto @smgl9 @LarsAsplund

qarlosalberto, 2021-07-01:
@umarcor thanks for the mention :) 

Currently in TerosHDL we use 2 parsers: tree-sitter-verilog (https://github.com/tree-sitter/tree-sitter-verilog) and tree-sitter-vhdl (https://github.com/alemuller/tree-sitter-vhdl). They don't use regex, they use tree-sitter (https://tree-sitter.github.io/tree-sitter/). Besides we have embedded them in our project with web-tree-sitter (https://www.npmjs.com/package/web-tree-sitter). So the only dependency is nodejs.

I think that tree-sitter-vhdl is a complete parser, but I'm not a VHDL language expert. I don't usually use VHDL 2008. With web-tree-sitter there is a problem, it's very very slow because it based in wasm.

We use VUnit for the dependency graph (https://terostechnology.github.io/teroshdl-documenter-demo/NEORV32_doc/index.html). But it's optional.

From my point of view the main feature in TerosHDL Documenter is the ability to parse the comments:

[image]

[image]

It's possible to use pyGHDL.dom as TerosHDL parser backend. We have a defined json format between parser and documenter: https://github.com/TerosTechnology/colibri/blob/master/test/examples/vhdl/example_1.json If pyGHDL.dom can provide this format it's easy to integrate it. Maybe the only problem is to parse the comments (https://terostechnology.github.io/terosHDLdoc/features/documenter.html)

I think that the best would be to provide it as a GitHub Action. Could you do it? I have no experience in that :/

qarlosalberto, 2021-07-01:
Only one clarification, the command line documenter (https://github.com/TerosTechnology/colibri) is still beta. We could first to release a version and later to integrate it in Github Actions. What do you think?

smgl9, 2021-07-01:
I totally subscribe to @qarlosalberto. 

@umarcor The only reasons not having a GitHub Action are the lack of experience and free time. 
I have created this repo for Github Actions: https://github.com/TerosTechnology/teroshdl-documenter-action

There is also a docker with TerosHDL backend: `docker pull terostech/colibri`
But is probably outdated because the documenter CLI is changing a lot in the last days and is not still stable.  Hopefully in the next weeks there will be a well documented release :)

umarcor, 2021-07-01:
@smgl9 @qarlosalberto take it easy mates! 😃 I was not criticising the lack of Action, but willing to explain Stephan why we might want to wait some days/weeks before adding the feature to this repo.

In fact, as you already know, that's the area I feel comfortable with, so I created some PRs to get the CI in TerosTechnology in shape:

- TerosTechnology/teroshdl-documenter-demo#1
- TerosTechnology/teroshdl-documenter-demo#2
- TerosTechnology/teroshdl-documenter-demo#3
- TerosTechnology/colibri#225
- See also https://github.com/umarcor/teroshdl-documenter-demo/commits/ci-action

However, I was thinking about it and I feel that the action belongs in TerosTechnology/colibri, rather than teroshdl-documenter-demo or a separated repo. That's why I did not open that PR yet. Let's deal with those 4 PRs first and we will then discuss the details of the Action. 

> They don't use regex, they use tree-sitter

@qarlosalberto I'm really sorry, I keep forgetting about tree-sitter again and again....

> From my point of view the main feature in TerosHDL Documenter is the ability to parse the comments:

I agree, but I think that's an orthogonal feature. The purpose of pyVHDLModel is understanding the language, not custom comment metadata. Conversely, the focus of TerosHDL is interpreting multiple custom comments, while the language support is secondary. That's why it'd be interesting to combine them together. pyVHDLModel can ignore the comments and TerosHDL can process the comments only.

> We have a defined json format between parser and documenter: https://github.com/TerosTechnology/colibri/blob/master/test/examples/vhdl/example_1.json If pyGHDL.dom can provide this format it's easy to integrate it.

Absolutely. Currently, the pretty printer in pyGHDL.dom produces the following output: https://github.com/umarcor/neorv32/runs/2953525135?check_suite_focus=true#step:5:460. So, the info is there already. We just need to adapt the output for generating a JSON instead of using YAML syntax. See a very minimal example about how to list the entities, generics and ports: https://github.com/vhdl/pyVHDLModel#list-all-entities-with-generics-and-ports. The complete pretty print is https://github.com/ghdl/ghdl/blob/master/pyGHDL/dom/formatting/prettyprint.py.

> Maybe the only problem is to parse the comments (https://terostechnology.github.io/terosHDLdoc/features/documenter.html)

That might be tricky in the first iteration. GHDL and libghdl do ignore the comments by default. Therefore, although pyVHDLModel might handle them, currently the only implementation is pyGHDL.dom. Hence, I think it will be necessary to start with combining both pyGHDL.dom and TerosHDL's own solution.

umarcor, 2021-07-01:
> We could first to release a version and later to integrate it in Github Actions. What do you think?

I heard @mithro say: release fast and improve iteratively; release before you think you are ready for it.

The last release of TerosHDL was in November 2020, but you've been doing much work these last months. Hence, I believe you could tag pre-releases maybe once a month, or more frequently. Tagging frequently allows users to pin known versions while not being left behind.

As commented in [gitter.im/hdl/community](https://gitter.im/hdl/community) I would like to see an Sphinx demo combining:

- pyGHDL.dom for the structure, dependency diagrams and cross-references as in https://vhdl.github.io/pyVHDLModel/LanguageModel/index.html.
- colibri for the registers/bitfields and the waveforms
  - [sphinx-wavedrom](https://github.com/bavovanachte/sphinx-wavedrom)
- [sphinxcontrib-hdl-diagrams](https://github.com/SymbiFlow/sphinxcontrib-hdl-diagrams) for the RTL diagrams of the synthesisable components.

Then, we might wrap it in a single directive which provides all the features in a bundle.

qarlosalberto, 2021-07-01:
Thanks @umarcor ! :D

So is it not possible to extract the comments with pyGHDL.dom? That could be a problem, since we would need a double parser (for comments and for HDL "tags").  How you can see in the terosHDL parser output (https://github.com/TerosTechnology/colibri/blob/master/test/examples/vhdl/example_1.json#L18) it provide the description (comments). 
Later the documenter interprets the comments (description field) and it generates the wavedrom image... or a plan text description.

Is there a way to associate comments to ports, signals, generics... after to execute pyVHDLModel?

Paebbels, 2021-07-01:
@qarlosalberto currently libghdl does not preserve comments. So the limitation is not pyVHDLModel or pyGHDL.dom based. The idea of pyVHDLParser is to preserved exactly this and to serve as a second input source for pyVHDLModel.

umarcor, 2021-07-01:
> So is it not possible to extract the comments with pyGHDL.dom?

For now... rust_hdl and pyVHDLParser do preserve comments. Hence, when/if those "backends" are used for pyVHDLModel, the model itself can contain the comments.
On the other hand, GHDL removed the comments because those are not relevant for simulation or synthesis. However, since it is now being used for other use cases, it might be changed so that comments are preserved. I don't know if @Paebbels discussed this with Tristan already.

Currently, pyGHDL.dom can provide "name", "type", "direction", "default_value" and "line". The "type" and "default_value" are not just plain strings, but you can optionally get the range/limits, etc. Alternatively, you can get an string representation if you don't want to handle those details. So, TerosHDL would need to parse the comments only, for providing the "description" field to be added by name matching.

qarlosalberto, 2021-07-01:
What could be the best option to parse the comments and associate them to a VHDL tag? regex + pyVHDLModel line field?

umarcor, 2021-07-01:
To start with something easy, I'd do it "*a lo burro*":

1. `colibri > out.colibri.json`
2. `ghdl-dom --colibri > out.dom.json` (naturally, `ghdl-dom --colibri` does not exist yet; that variation of the prettyprinter needs to be implemented)
3. Overwrite each field in `out.colibri.json` with the content of `out.dom.json`.

This Python snippet will do step 3:

```py
data = {}

colibri_file = Path(__file__).resolve().parent / 'out.colibri.json'
if not colibri_file.is_file():
    print('Ups...')
    exit(1)
data = loads(colibri_file.open('r').read())

dom_file = Path(__file__).resolve().parent / 'out.dom.json'
if dom_file.is_file():
    print('Ups...')
    exit(1)
data.update(loads(dom_file.open('r').read()))
```

That will allow to check the compabilitity of the fields/data generated by `ghdl-dom --colibri` and the built-in parser. Then, it'd be a matter of removing fields from the colibri output (maybe support `colibri --comments-only`).

qarlosalberto, 2021-07-01:
Then you want to combine the colibri descriptions with ghdl-dom, right?

qarlosalberto, 2021-07-01:
Ok, I see a problem. Then we are running 2 parsers: tree-sitter-vhdl + ghdl-dom. tree-sitter-vhdl is very very slow and we will and the latency of ghdl-dom. I think that we are deleting the advantage to use ghdl-dom. 
The main problem is that in colibri we need to parse the complete document to extract the comments. we can't separate it.

In my opinion the best way would be to use ghdl-dom + a simple regex parser for only the comments and too associate the comments to the VHDL tags (we could use the line??)

umarcor, 2021-07-01:
@qarlosalberto, agree. I'm not suggesting it as a definitive solution, but just so we can start with something usable. Rather than implementing a regex parser for the comments, I think we can all beg Tristan to pleeeeeeeeeeease let us get the comments through libghdl. I don't mind if we need to wait some weeks/months for that. I've been waiting YEARS for something such as pyVHDLModel to exist on top of GHDL 😄.

The proposal of using 2 parsers is because it allows us not to add a lot of new code now, but to use most of what is written already, and combine it as is. We can do that in a feature branch, not in main/master. Once the prototype works, we have further issues to discuss. Precisely, while I see the value in generating markdown, I don't think it's worth the effort of generating raw HTML. Instead, the HTML output should be meant to be combined with Sphinx, asciidoctor, or other static site generator engines. That allows significantly reducing the burden, since you don't need to care about styling, cross-referencing, etc. Moreover, we get PDF output "for free". Even if you don't use static site generators, the usage of templates (say jinja2, go templates, or whatever equivalent in JS) can help reuse and customisation.

qarlosalberto, 2021-07-01:
okok, it's easy to do it. A command line interface with the colibri parser.

stnolting, 2021-07-04:
This is not urgent at all. I was just amazed by the documenter's features. :heart:

So I agree with @umarcor:

> I was not criticising the lack of Action, but willing to explain Stephan why we might want to wait some days/weeks before adding the feature to this repo.

smgl9, 2021-07-12:
Sorry if I sound bad. Don't get me wrong. 
I love this project and also @umarcor knows that I love the way he keep us all updated :) 
I know we are colaborating and learning, continuously. And I Iove that :D

closed 2023-03-16 by stnolting