stnolting/neorv32 #97: Having trouble building the iCESugar example

jeremyherbert, 2021-07-03:
I'm having some trouble trying to build the iCESugar example. From the `setups/examples` folder, I run the command:

`BOARD=iCESugar DESIGN=MinimalBoot VERBOSE=1 make iCESugar`

But GHDL complains:

```
make \
  BITSTREAM=neorv32_iCESugar_MinimalBoot.bit \
  run
make[1]: Entering directory '/home/jeremy/Desktop/neorv32/setups/examples'
make -C ../osflow/iCESugar/ \
  BOARD_SRC=../../examples/neorv32_iCESugar_BoardTop_MinimalBoot.vhd \
  TOP=neorv32_iCESugar_BoardTop_MinimalBoot \
  ID=MinimalBoot \
  clean neorv32_iCESugar_MinimalBoot.bit
make[2]: Entering directory '/home/jeremy/Desktop/neorv32/setups/osflow/iCESugar'
rm -f *.{asc,bit,cf,dfu,history,json,o} *-report.txt
ghdl -a --std=08 --work=iCE40 ../devices/ice40/sb_ice40_components.vhd
ghdl -a --std=08 --work=neorv32 ../../../rtl/core/neorv32_package.vhd ../../../rtl/core/neorv32_application_image.vhd  ../devices/ice40/neorv32_imem.ice40up_spram.vhd ../devices/ice40/neorv32_dmem.ice40up_spram.vhd ../../../rtl/core/neorv32_bootloader_image.vhd ../../../rtl/core/neorv32_boot_rom.vhd ../../../rtl/core/neorv32_bus_keeper.vhd ../../../rtl/core/neorv32_busswitch.vhd ../../../rtl/core/neorv32_cfs.vhd ../../../rtl/core/neorv32_cpu.vhd ../../../rtl/core/neorv32_cpu_alu.vhd ../../../rtl/core/neorv32_cpu_bus.vhd ../../../rtl/core/neorv32_cpu_control.vhd ../../../rtl/core/neorv32_cpu_cp_fpu.vhd ../../../rtl/core/neorv32_cpu_cp_muldiv.vhd ../../../rtl/core/neorv32_cpu_cp_shifter.vhd ../../../rtl/core/neorv32_cpu_decompressor.vhd ../../../rtl/core/neorv32_cpu_regfile.vhd ../../../rtl/core/neorv32_debug_dm.vhd ../../../rtl/core/neorv32_debug_dtm.vhd ../../../rtl/core/neorv32_fifo.vhd ../../../rtl/core/neorv32_gpio.vhd ../../../rtl/core/neorv32_icache.vhd ../../../rtl/core/neorv32_mtime.vhd ../../../rtl/core/neorv32_neoled.vhd ../../../rtl/core/neorv32_pwm.vhd ../../../rtl/core/neorv32_slink.vhd ../../../rtl/core/neorv32_spi.vhd ../../../rtl/core/neorv32_sysinfo.vhd ../../../rtl/core/neorv32_top.vhd ../../../rtl/core/neorv32_trng.vhd ../../../rtl/core/neorv32_twi.vhd ../../../rtl/core/neorv32_uart.vhd ../../../rtl/core/neorv32_wdt.vhd ../../../rtl/core/neorv32_wishbone.vhd
ghdl -a --std=08 --work=work  ../../examples/neorv32_iCESugar_BoardTop_MinimalBoot.vhd
../../examples/neorv32_iCESugar_BoardTop_MinimalBoot.vhd:134:29:error: unit "neorv32_processortop_minimalboot" not found in library "work"
  neorv32_inst: entity work.neorv32_ProcessorTop_MinimalBoot
                            ^
make[2]: *** [../synthesis.mk:8: work-obj08.cf] Error 1
make[2]: Leaving directory '/home/jeremy/Desktop/neorv32/setups/osflow/iCESugar'
make[1]: *** [Makefile:16: run] Error 2
make[1]: Leaving directory '/home/jeremy/Desktop/neorv32/setups/examples'
make: *** [Makefile:35: iCESugar] Error 2
```

If I run `BOARD=iCESugar DESIGN=MinimalBoot VERBOSE=1 make`, then all that happens is that my `iCESugar` folder is moved from the `osflow` folder to the `examples` folder:

```
make -C ../osflow/iCESugar/ \
  BOARD_SRC=../../examples/neorv32_iCESugar_BoardTop_MinimalBoot.vhd \
  TOP=neorv32_iCESugar_BoardTop_MinimalBoot \
  ID=MinimalBoot \
  clean 
make[1]: Entering directory '/home/jeremy/Desktop/neorv32/setups/osflow/iCESugar'
rm -f *.{asc,bit,cf,dfu,history,json,o} *-report.txt
make[1]: Leaving directory '/home/jeremy/Desktop/neorv32/setups/osflow/iCESugar'
mv ../osflow/iCESugar/ ./
```

Finally, if I run `BOARD=iCESugar VERBOSE=1 make MinimalBoot` (which I am thinking is the correct one), I get the following error:

```
make \
  DESIGN=MinimalBoot \
  DESIGN_SRC=../../../rtl/templates/processor/neorv32_ProcessorTop_MinimalBoot.vhd \
  iCESugar
make[1]: Entering directory '/home/jeremy/Desktop/neorv32/setups/examples'
make \
  BITSTREAM=neorv32_iCESugar_MinimalBoot.bit \
  run
make[2]: Entering directory '/home/jeremy/Desktop/neorv32/setups/examples'
make -C ../osflow/iCESugar/ \
  BOARD_SRC=../../examples/neorv32_iCESugar_BoardTop_MinimalBoot.vhd \
  TOP=neorv32_iCESugar_BoardTop_MinimalBoot \
  ID=MinimalBoot \
  clean neorv32_iCESugar_MinimalBoot.bit
make[3]: Entering directory '/home/jeremy/Desktop/neorv32/setups/osflow/iCESugar'
rm -f *.{asc,bit,cf,dfu,history,json,o} *-report.txt
yosys  \
  -p \
  "ghdl --std=08 --no-formal neorv32_iCESugar_BoardTop_MinimalBoot; \
  synth_ice40 \
  -top neorv32_iCESugar_BoardTop_MinimalBoot \
  -dsp \
  -json neorv32_iCESugar_MinimalBoot.json"  2>&1 | tee yosys-report.txt

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.9+4081 (open-tool-forge build) (git sha1 862e84eb, gcc 9.3.0-17ubuntu1~20.04 -Os)

-- Running command `ghdl --std=08 --no-formal neorv32_iCESugar_BoardTop_MinimalBoot;   synth_ice40   -top neorv32_iCESugar_BoardTop_MinimalBoot   -dsp   -json neorv32_iCESugar_MinimalBoot.json' --

1. Executing GHDL.
warning: ieee library directory '/__w/1/s/_packages/build_linux_x86_64/fpga-toolchain/lib/ghdl/ieee/v08/' not found
error: cannot find "std" library
ERROR: vhdl import failed.
nextpnr-ice40 \
  --up5k --package sg48 --ignore-loops --timing-allow-fail \
  --pcf iCESugar.pcf \
  --json neorv32_iCESugar_MinimalBoot.json \
  --asc neorv32_iCESugar_MinimalBoot.asc 2>&1 | tee nextpnr-report.txt
ERROR: Failed to open JSON file 'neorv32_iCESugar_MinimalBoot.json'.
0 warnings, 1 error
icepack neorv32_iCESugar_MinimalBoot.asc neorv32_iCESugar_MinimalBoot.bit
Error: Failed to open input file.
make[3]: *** [../PnR_Bit.mk:9: neorv32_iCESugar_MinimalBoot.bit] Error 1
make[3]: Leaving directory '/home/jeremy/Desktop/neorv32/setups/osflow/iCESugar'
make[2]: *** [Makefile:16: run] Error 2
make[2]: Leaving directory '/home/jeremy/Desktop/neorv32/setups/examples'
make[1]: *** [Makefile:35: iCESugar] Error 2
make[1]: Leaving directory '/home/jeremy/Desktop/neorv32/setups/examples'
make: *** [Makefile:51: MinimalBoot] Error 2

```

Though I believe this last one might be due to an issue with my toolchain install? I have just downloaded the toolchain from here: `https://github.com/YosysHQ/fpga-toolchain` (version `nightly-20210703`), extracted it to /opt/fpga-toolchain and added it to my path.

If someone could point me in the right direction, it would be much appreciated.

jeremyherbert, 2021-07-03:
Ok, fixed the issue with the last one (see [https://github.com/YosysHQ/fpga-toolchain#using-ghdl](https://github.com/YosysHQ/fpga-toolchain#using-ghdl)), but am now seeing an error which is not related to the toolchain:

```
make \
  DESIGN=MinimalBoot \
  DESIGN_SRC=../../../rtl/templates/processor/neorv32_ProcessorTop_MinimalBoot.vhd \
  iCESugar
make[1]: Entering directory '/home/jeremy/Desktop/neorv32/setups/examples'
make \
  BITSTREAM=neorv32_iCESugar_MinimalBoot.bit \
  run
make[2]: Entering directory '/home/jeremy/Desktop/neorv32/setups/examples'
make -C ../osflow/iCESugar/ \
  BOARD_SRC=../../examples/neorv32_iCESugar_BoardTop_MinimalBoot.vhd \
  TOP=neorv32_iCESugar_BoardTop_MinimalBoot \
  ID=MinimalBoot \
  clean neorv32_iCESugar_MinimalBoot.bit
make[3]: Entering directory '/home/jeremy/Desktop/neorv32/setups/osflow/iCESugar'
rm -f *.{asc,bit,cf,dfu,history,json,o} *-report.txt
yosys  \
  -p \
  "ghdl --std=08 --no-formal neorv32_iCESugar_BoardTop_MinimalBoot; \
  synth_ice40 \
  -top neorv32_iCESugar_BoardTop_MinimalBoot \
  -dsp \
  -json neorv32_iCESugar_MinimalBoot.json"  2>&1 | tee yosys-report.txt

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.9+4081 (open-tool-forge build) (git sha1 862e84eb, gcc 9.3.0-17ubuntu1~20.04 -Os)

-- Running command `ghdl --std=08 --no-formal neorv32_iCESugar_BoardTop_MinimalBoot;   synth_ice40   -top neorv32_iCESugar_BoardTop_MinimalBoot   -dsp   -json neorv32_iCESugar_MinimalBoot.json' --

1. Executing GHDL.
error: cannot find entity or configuration neorv32_icesugar_boardtop_minimalboot
ERROR: vhdl import failed.
nextpnr-ice40 \
  --up5k --package sg48 --ignore-loops --timing-allow-fail \
  --pcf iCESugar.pcf \
  --json neorv32_iCESugar_MinimalBoot.json \
  --asc neorv32_iCESugar_MinimalBoot.asc 2>&1 | tee nextpnr-report.txt
ERROR: Failed to open JSON file 'neorv32_iCESugar_MinimalBoot.json'.
0 warnings, 1 error
icepack neorv32_iCESugar_MinimalBoot.asc neorv32_iCESugar_MinimalBoot.bit
Error: Failed to open input file.
make[3]: *** [../PnR_Bit.mk:9: neorv32_iCESugar_MinimalBoot.bit] Error 1
make[3]: Leaving directory '/home/jeremy/Desktop/neorv32/setups/osflow/iCESugar'
make[2]: *** [Makefile:16: run] Error 2
make[2]: Leaving directory '/home/jeremy/Desktop/neorv32/setups/examples'
make[1]: *** [Makefile:35: iCESugar] Error 2
make[1]: Leaving directory '/home/jeremy/Desktop/neorv32/setups/examples'
make: *** [Makefile:51: MinimalBoot] Error 2
```

stnolting, 2021-07-03:
This might be just a path issue... I will check that.

Maybe @umarcor can directly spot the problem :wink:

umarcor, 2021-07-03:
@jeremyherbert try `make -C setups/examples/ BOARD=iCESugar MinimalBoot` or:

```sh
cd setups/examples/
make BOARD=iCESugar MinimalBoot
```

umarcor, 2021-07-03:
@jeremyherbert, first off, I apologise for the undocumented state of these makefiles, and for my previous so short answer. Your issue was related to the toolchain, indeed, and you fixed it properly. BTW, the Fomu toolchain is based on fpga-toolchain, so you will get the same issue. Now, the issue is the make target you are using. You got it correct in the last try before fixing the GHDL_PREFIX issue: `make BOARD=iCESugar MinimalBoot`.

Find a more detailed explanation about the makefiles in #96.

jeremyherbert, 2021-07-03:
Thanks for your help! I tried to run your make commands, but I still received the same error as above:

```
...
1. Executing GHDL.
error: cannot find entity or configuration neorv32_icesugar_boardtop_minimalboot
ERROR: vhdl import failed.
...
```

However, with a bit of poking around, it seems that some stale `.cf` files built with the wrong make commands were still in the build directory, and they were causing the GHDL build to break. Once I removed these files everything works great!

closed 2021-07-03 by jeremyherbert