stnolting/neorv32 #105: Example for using an external DDR memory

umarcor, 2021-07-07:
Coming from #98.

> https://github.com/stnolting/neorv32/pull/98#issuecomment-874138098
>
> The OrangeCrab has an external DDR with 128 MB. Hence, the mid-term goal should be to append a controller to the external bus of the NEORV32 and have the IMEM and/or DMEM assigned there.
>
> Note that we can have mixed-language designs. Therefore, we might use the same strategy as in [antonblanchard/microwatt](https://github.com/antonblanchard/microwatt): they use [litedram](https://github.com/antonblanchard/microwatt/tree/master/litedram) (see [enjoy-digital/litedram](https://github.com/enjoy-digital/litedram)).

> https://github.com/stnolting/neorv32/pull/98#issuecomment-874149137
>
> That looks very good! I'm very happy that the memory controller supports Wishbone so there is no need for extra bridging logic.
I will check out the documentation. That would be also a nice add-on for the current Quartus and Vivado examples.

jeremyherbert, 2021-07-07:
I am happy to help test this when the time comes.

stnolting, 2021-11-29:
We now have two links to projects that have SDRAM support in [setups](https://github.com/stnolting/neorv32/tree/master/setups):

* https://github.com/emb4fun/neorv32-examples (Intel) by @emb4fun 
* https://github.com/zipotron/neorv32-complex-setups (Lattice, work-in-progress) by @zipotron

So I think this can be closed?! The litedram/litex approach mentioned by @umarcor should be further discuseed in #115.

umarcor, 2021-11-29:
@stnolting, agree.

closed 2021-11-29 by umarcor