stnolting/neorv32 #108: question: Is the prebuilt build support rv32IMAC extension or just rv32imc?

Saqlinahamad, 2021-07-08:


stnolting, 2021-07-08:
Do you mean the **prebuilt gcc toolchain** from https://github.com/stnolting/riscv-gcc-prebuilt or the **"prebuilt" FPGA processor setups** from https://github.com/stnolting/neorv32/actions/workflows/Implementation.yml?

1. The prebuilt gcc toolchain supports all ISA extensions that were officially ratified in the specific GCC / binutils version. For example, https://github.com/stnolting/riscv-gcc-prebuilt/releases/tag/rv32i-2.0.0 supports all the ISA extensions in you have mentioned (i, m, a, c).
2. The prebuilt FPGA setups use different configuration options of the processor/CPU. The [example setups](https://github.com/stnolting/neorv32/tree/master/setups/examples) that are used in the "implementation" GitHub workflow instantiate processor wrappers from [`rtl/templates/processor`](https://github.com/stnolting/neorv32/tree/master/rtl/templates). Check out the according wrapper VHDL file there to see which CPU ISA extensions are enabled and which are not.

Saqlinahamad, 2021-07-09:
I meant prebuilt gcc toolchain

closed 2021-07-09 by Saqlinahamad