stnolting/neorv32 #121: SIMD/Vector extension

davidbrochart, 2021-07-12:
Could NEORV32 support SIMD or vector extension in the future, or is it not possible/not on the roadmap?

stnolting, 2021-07-12:
Basically, nothing speaks against that. The CPU uses a "co-processor" concept for adding more sophisticated ISA extensions. Right now, the `M` extension (integer multiply and divide) and the `Zfinx` extension (32-bit floating point) are implemented as co-processors. So it would be no problem to add another CP to implement vector operations.

However, the RISC-V vector extension is a really capable but also a huge (in terms of hardware) extension. So I am not sure if this is something that should be implemented within the core, or if vector operations are better off in a custom accelerator (like the processor's "custom functions unit").

Anyway, let's wait until the RISC-V vector extension is frozen (or at least in ratification process). Then we can check which sub-extensions are planned. Maybe we could only implement some of them and not the full-scale vector processing system.

davidbrochart, 2021-07-12:
Thanks for the clarification, I was not aware the Vector extension was not ratified.
What about SIMD? Although from what I heard, it looks like an "inferior" solution compared to the V extension.

stnolting, 2021-07-12:
"Vector" and "SIMD" are synonyms here. Basically, there are two concepts of vector processors. Both concepts use a **S**ingle **I**nstruction to process **M**ultiple **D**ata "words":
* vertical vector processor: each subword is in an individual register and all the subwords are processed in a iterative/serial way
* horizontal vector processor: several subwords are packed in a _single_ register and all subwords are processed in parallel

I am not too familiar with the RISC-V vector extension yet, but I think the [RISC-V `V` extension](https://github.com/riscv/riscv-v-spec) implements the vertical concept. There is another extension being in development: the [RISC-V `P` extension](https://github.com/riscv/riscv-p-spec), which uses the horizontal concept (also not ratified yet).

Both concepts could be implemented into the NEORV32 but I have not really studied much of the specs yet :smile:
For now I think that both extensions are oversized and there are no plans (yet?) to implement them. However, some sub-extensions or at least some specific instructions might be nice to have.

davidbrochart, 2021-07-12:
Thanks again!
I think SiFive has something in the pipe: https://www.sifive.com/cores/intelligence-x280

umarcor, 2021-07-12:
@davidbrochart note that NEORV32 fits in SiFive's E series (ARM's Cortex-M and R) rather than higher performance and 64 bit designs. NEORV32 might be enhanced in the future for supporting applications with higher computation requirements. However, I believe it is not in the scope for now. It's rather a uC for orchestration and minimal computation.

Meanwhile, you might want to have a look at other open source RISC-V cores written in VHDL which are comparable to SiFive's S and U series: https://riscv.org/exchange/cores-socs/. For instance, NOEL-V by Cobham Gaisler (https://www.gaisler.com/index.php/products/processors/noel-v). On the SoC side (https://riscv.org/exchange/cores-socs/), regardless of the language, PULP is doing a very interesting work. See https://pulp-platform.org/ and https://github.com/pulp-platform/ara.

davidbrochart, 2021-07-12:
Thanks @umarcor, I have come across the projects you are mentioning. I will look at them in more details.

stnolting, 2021-07-12:
> I think SiFive has something in the pipe: https://www.sifive.com/cores/intelligence-x280

Nice one!
SiFive is really pushing RISC-V! They already boosted the ratification process of many other ISA extensions. :+1:

> @davidbrochart note that NEORV32 fits in SiFive's E series (ARM's Cortex-M and R) rather than higher performance and 64 bit designs. NEORV32 might be enhanced in the future for supporting [...]

That's right! -> https://stnolting.github.io/neorv32/#_rationale :wink:

stnolting, 2021-07-24:
I think this can be closed now :wink:
Feel free to open a new issue or [discussion](https://github.com/stnolting/neorv32/discussions) if you have further questions or comments. :+1:

closed 2021-07-24 by stnolting