stnolting/neorv32 #132: Cannot fit on XC6SLX9

Alirio926, 2021-07-24:
Hello!
First of all, thanks for sharing this great project.
I'm new to vhdl, tested the project with an altera ep4ce6 and it worked, then I decided to put it into real operation on a hardware with XC6SLX9 and to my surprise ISE Design couldn't fit. Some problem with async ram seems. Any idea what it might be?

I created a project on ISE, added the top level test from the template folder, set the pins and that's it.

```VHDL
ERROR:Pack:2413 - The number of logical LUTRAM blocks exceeds the capacity for the target device.
ERROR:Pack:2744 - This design requires at least 384 SLICEMs for LUTRAMs. The target device has only 360 SLICEM sites. Please modify the
   logical design source to reduce the number of LUTRAMs. 
   As an alternative, please select a device with more SLICEM sites.
ERROR:Map:237 - The design is too large to fit the device.  Please check the Design Summary section to see which resource requirement for
   your design exceeds the resources available in the device. Note that the number of slices reported may not be reflected accurately as
   their packing might not have been completed.`

`
WARNING:HDLCompiler:746 - "E:\VHDL\neorv32\rtl\core\neorv32_package.vhd" Line 1002: Range is empty (null range)
WARNING:HDLCompiler:746 - "E:\VHDL\neorv32\rtl\core\neorv32_package.vhd" Line 1012: Range is empty (null range)
WARNING:HDLCompiler:746 - "E:\VHDL\neorv32\rtl\core\neorv32_top.vhd" Line 196: Range is empty (null range)
WARNING:HDLCompiler:746 - "E:\VHDL\neorv32\rtl\core\neorv32_top.vhd" Line 210: Range is empty (null range)
WARNING:HDLCompiler:746 - "E:\VHDL\neorv32\rtl\core\neorv32_cpu_control.vhd" Line 261: Range is empty (null range)
WARNING:HDLCompiler:746 - "E:\VHDL\neorv32\rtl\core\neorv32_cpu_control.vhd" Line 262: Range is empty (null range)
WARNING:HDLCompiler:746 - "E:\VHDL\neorv32\rtl\core\neorv32_cpu_bus.vhd" Line 149: Range is empty (null range)
WARNING:HDLCompiler:746 - "E:\VHDL\neorv32\rtl\core\neorv32_cpu_bus.vhd" Line 155: Range is empty (null range)
WARNING:HDLCompiler:746 - "E:\VHDL\neorv32\rtl\core\neorv32_cpu_bus.vhd" Line 156: Range is empty (null range)
WARNING:HDLCompiler:746 - "E:\VHDL\neorv32\rtl\core\neorv32_cpu_bus.vhd" Line 157: Range is empty (null range)
WARNING:HDLCompiler:746 - "E:\VHDL\neorv32\rtl\core\neorv32_cpu_bus.vhd" Line 158: Range is empty (null range)
WARNING:HDLCompiler:746 - "E:\VHDL\neorv32\rtl\core\neorv32_cpu_bus.vhd" Line 159: Range is empty (null range)
WARNING:HDLCompiler:746 - "E:\VHDL\neorv32\rtl\core\neorv32_cpu_bus.vhd" Line 459: Range is empty (null range)
WARNING:HDLCompiler:746 - "E:\VHDL\neorv32\rtl\core\neorv32_cpu_bus.vhd" Line 471: Range is empty (null range)
WARNING:HDLCompiler:746 - "E:\VHDL\neorv32\rtl\core\neorv32_cpu_bus.vhd" Line 484: Range is empty (null range)
WARNING:HDLCompiler:220 - "E:\VHDL\neorv32\rtl\core\neorv32_top.vhd" Line 1186: Assignment ignored
WARNING:Xst:647 - Input <wb_dat_i> is never used. This port will be preserved and left unconnected if it belongs to a top-level block or it belongs to a sub-block and the hierarchy of this sub-block is preserved.
WARNING:Xst:647 - Input <slink_tx_rdy_i> is never used. This port will be preserved and left unconnected if it belongs to a top-level block or it belongs to a sub-block and the hierarchy of this sub-block is preserved.
WARNING:Xst:647 - Input <slink_rx_dat_i<0>> is never used. This port will be preserved and left unconnected if it belongs to a top-level block or it belongs to a sub-block and the hierarchy of this sub-block is preserved.
WARNING:Xst:647 - Input <slink_rx_dat_i<1>> is never used. This port will be preserved and left unconnected if it belongs to a top-level block or it belongs to a sub-block and the hierarchy of this sub-block is preserved.
WARNING:Xst:647 - Input <slink_rx_dat_i<2>> is never used. This port will be preserved and left unconnected if it belongs to a top-level block or it belongs to a sub-block and the hierarchy of this sub-block is preserved.
WARNING:Xst:647 - Input <slink_rx_dat_i<3>> is never used. This port will be preserved and left unconnected if it belongs to a top-level block or it belongs to a sub-block and the hierarchy of this sub-block is preserved.
WARNING:Xst:647 - Input <slink_rx_dat_i<4>> is never used. This port will be preserved and left unconnected if it belongs to a top-level block or it belongs to a sub-block and the hierarchy of this sub-block is preserved.
WARNING:Xst:647 - Input <slink_rx_dat_i<5>> is never used. This port will be preserved and left unconnected if it belongs to a top-level block or it belongs to a sub-block and the hierarchy of this sub-block is preserved.
WARNING:Xst:647 - Input <slink_rx_dat_i<6>> is never used. This port will be preserved and left unconnected if it belongs to a top-level block or it belongs to a sub-block and the hierarchy of this sub-block is preserved.
WARNING:Xst:647 - Input <slink_rx_dat_i<7>> is never used. This port will be preserved and left unconnected if it belongs to a top-level block or it belongs to a sub-block and the hierarchy of this sub-block is preserved.
WARNING:Xst:647 - Input <slink_rx_val_i> is never used. This port will be preserved and left unconnected if it belongs to a top-level block or it belongs to a sub-block and the hierarchy of this sub-block is preserved.
WARNING:Xst:647 - Input <cfs_in_i> is never used. This port will be preserved and left unconnected if it belongs to a top-level block or it belongs to a sub-block and the hierarchy of this sub-block is preserved.
WARNING:Xst:647 - Input <mtime_i> is never used. This port will be preserved and left unconnected if it belongs to a top-level block or it belongs to a sub-block and the hierarchy of this sub-block is preserved.
WARNING:Xst:647 - Input <jtag_trst_i> is never used. This port will be preserved and left unconnected if it belongs to a top-level block or it belongs to a sub-block and the hierarchy of this sub-block is preserved.
WARNING:Xst:647 - Input <jtag_tck_i> is never used. This port will be preserved and left unconnected if it belongs to a top-level block or it belongs to a sub-block and the hierarchy of this sub-block is preserved.
WARNING:Xst:647 - Input <jtag_tms_i> is never used. This port will be preserved and left unconnected if it belongs to a top-level block or it belongs to a sub-block and the hierarchy of this sub-block is preserved.
WARNING:Xst:647 - Input <wb_ack_i> is never used. This port will be preserved and left unconnected if it belongs to a top-level block or it belongs to a sub-block and the hierarchy of this sub-block is preserved.
WARNING:Xst:647 - Input <wb_err_i> is never used. This port will be preserved and left unconnected if it belongs to a top-level block or it belongs to a sub-block and the hierarchy of this sub-block is preserved.
WARNING:Xst:647 - Input <uart1_rxd_i> is never used. This port will be preserved and left unconnected if it belongs to a top-level block or it belongs to a sub-block and the hierarchy of this sub-block is preserved.
WARNING:Xst:647 - Input <uart1_cts_i> is never used. This port will be preserved and left unconnected if it belongs to a top-level block or it belongs to a sub-block and the hierarchy of this sub-block is preserved.
WARNING:Xst:647 - Input <spi_sdi_i> is never used. This port will be preserved and left unconnected if it belongs to a top-level block or it belongs to a sub-block and the hierarchy of this sub-block is preserved.
WARNING:Xst:647 - Input <mtime_irq_i> is never used. This port will be preserved and left unconnected if it belongs to a top-level block or it belongs to a sub-block and the hierarchy of this sub-block is preserved.
WARNING:Xst:2563 - Inout <twi_sda_io> is never assigned. Tied to value Z.
WARNING:Xst:2563 - Inout <twi_scl_io> is never assigned. Tied to value Z.
    WARNING:Xst:2404 -  FFs/Latches <clk_gen_en<7:2>> (without init value) have a constant value of 0 in block <neorv32_top>.
WARNING:Xst:647 - Input <alu_add_i<0:0>> is never used. This port will be preserved and left unconnected if it belongs to a top-level block or it belongs to a sub-block and the hierarchy of this sub-block is preserved.
WARNING:Xst:647 - Input <fpu_flags_i> is never used. This port will be preserved and left unconnected if it belongs to a top-level block or it belongs to a sub-block and the hierarchy of this sub-block is preserved.
[... cut ...]
WARNING:Xst:2677 - Node <neorv32_top_inst/neorv32_cpu_inst/neorv32_cpu_control_inst/ctrl_16> of sequential type is unconnected in block <neorv32_ProcessorTop_Test>.
WARNING:Xst:2677 - Node <neorv32_top_inst/neorv32_cpu_inst/neorv32_cpu_control_inst/ctrl_19> of sequential type is unconnected in block <neorv32_ProcessorTop_Test>.
WARNING:Xst:2677 - Node <neorv32_top_inst/neorv32_uart0_inst_true.neorv32_uart0_inst/uart_rts_o> of sequential type is unconnected in block <neorv32_ProcessorTop_Test>.
WARNING:Xst:1336 -  (*) More than 100% of Device resources are used`

**Environment:**
 - Windows 10
 - ISE 14.7

stnolting, 2021-07-25:
Hey there!

The test-setup ([`rtl/templates/processor/neorv32_ProcessorTop_Test.vhd`](https://github.com/stnolting/neorv32/blob/master/rtl/templates/processor/neorv32_ProcessorTop_Test.vhd)) implements 16kB for the IMEM, 8kB for the DMEM, 4kB for the boot ROM and 256 bytes for the CPU register file. That makes a total memory consumption of ~226 kbit. According to the Spartan-6 LX9 datasheet the FPGA provides 576kbit block RAM - so that _should_ be sufficient 🤔

Seems like more than the available block RAMs are used and ISE tries to provide more memory by using up all distributed RAMs.
Can you post some cut-out of the synthesis/mapping reports? Maybe something like "resource utilization by entitiy" (it has been a while since I have worked with ISE...)? The reports should show which component consumes all the block RAMs.

However, you could try reducing the instruction (IMEM) and data (DMEM) memory sizes to make it fit on your FPGA

* Reduce IMEM size (for example to 8*1024 bytes):

https://github.com/stnolting/neorv32/blob/4830a5bbd01d2106a9c5e57c5022ef197b910dda/rtl/templates/processor/neorv32_ProcessorTop_Test.vhd#L99

* And/or reduce DMEM size (for example to 4*1024 bytes):

https://github.com/stnolting/neorv32/blob/4830a5bbd01d2106a9c5e57c5022ef197b910dda/rtl/templates/processor/neorv32_ProcessorTop_Test.vhd#L102

I would start with decreasing DMEM size. Seems like only a few kbits are missing to successfully map the design. Furthermore, most of the example applications do not require 8kB DMEM anyway :smile:

:warning: If you modify the DMEM size you need to adapt the linker script and recompile your software application:

https://github.com/stnolting/neorv32/blob/4830a5bbd01d2106a9c5e57c5022ef197b910dda/sw/common/neorv32.ld#L62

You can find more information regarding this in the user guide: https://stnolting.github.io/neorv32/ug/#_general_software_framework_setup

Btw, the warnings you mentioned are "normal" and are not critical at all :wink:

Alirio926, 2021-07-25:
Thankyou again for fast resp ^^.
The sintetize work who:
```vhdl
MEM_INT_IMEM_SIZE           => 4*1024
MEM_INT_DMEM_SIZE           => 4*1024,
```
[image]
[image]

stnolting, 2021-07-26:
This is interesting!

[image]

The boot ROM was mapped to two block RAMs, but the IMEM and DMEM were not. Even though they have the same size now (all 4kB). :thinking:

There are some things that you could try to make ISE map those memories to block RAM, too:

1. As I said I have not worked with ISE in years, but I think there was an option to configure a "threshold" that defines the minimum RAM size that will get mapped to block RAM. DMEM and IMEM are built from 4 byte-wide memories each. So each of these byte RAMs is just 1kB big in your current setups. ISE might think this is too small for "wasting" block RAM resources... Maybe you can check out the synthesis HDL options: `Synthesis -> Process Properties -> HDL options` (copied form here: https://stackoverflow.com/questions/4671203/where-to-force-xilinx-ise-to-use-block-rams).

2. Alternatively, you could try adding attributes to the VHDL memory sources. For instance, you could try this for `rtl/core/neorv32_dmem.vhd` (I hope the syntax is correct):

```vhdl
  attribute ram_style of mem_ram_b0_rd : signal is "block"; 
  attribute ram_style of mem_ram_b1_rd : signal is "block"; 
  attribute ram_style of mem_ram_b2_rd : signal is "block"; 
  attribute ram_style of mem_ram_b3_rd : signal is "block"; 
```

3. If all this does not work, there is a problem with ISE identifying the memory VHDL code. We had some discussion about that issue (#35, ..., #101). I still hope to have a generalized description that can be understood by _any_ synthesis tool 😅 Anyway, the final option would be either modifying the memory VHDL source code or using memory IPs generated by ISE.

Alirio926, 2021-07-26:
Tried option one and two, no luck.

stnolting, 2021-07-26:
I am sorry top hear that.
Can you post/upload the synthesis report? I am really curious where ISE is failing to map those memories.. 🤔

Alirio926, 2021-07-26:
Follow Detailed Synthesis Report
```vhdl
Release 14.7 - xst P.20131013 (nt64)
Copyright (c) 1995-2013 Xilinx, Inc.  All rights reserved.
--> Parameter TMPDIR set to xst/projnav.tmp

Total REAL time to Xst completion: 0.00 secs
Total CPU time to Xst completion: 0.11 secs
 
--> Parameter xsthdpdir set to xst

Total REAL time to Xst completion: 0.00 secs
Total CPU time to Xst completion: 0.12 secs
 
--> Reading design: neorv32_ProcessorTop_Test.prj

TABLE OF CONTENTS
  1) Synthesis Options Summary
  2) HDL Parsing
  3) HDL Elaboration
  4) HDL Synthesis
       4.1) HDL Synthesis Report
  5) Advanced HDL Synthesis
       5.1) Advanced HDL Synthesis Report
  6) Low Level Synthesis
  7) Partition Report
  8) Design Summary
       8.1) Primitive and Black Box Usage
       8.2) Device utilization summary
       8.3) Partition Resource Summary
       8.4) Timing Report
            8.4.1) Clock Information
            8.4.2) Asynchronous Control Signals Information
            8.4.3) Timing Summary
            8.4.4) Timing Details
            8.4.5) Cross Clock Domains Report

=========================================================================
*                      Synthesis Options Summary                        *
=========================================================================
---- Source Parameters
Input File Name                    : "neorv32_ProcessorTop_Test.prj"
Ignore Synthesis Constraint File   : NO

---- Target Parameters
Output File Name                   : "neorv32_ProcessorTop_Test"
Output Format                      : NGC
Target Device                      : xc6slx9-3-tqg144

---- Source Options
Top Module Name                    : neorv32_ProcessorTop_Test
Automatic FSM Extraction           : YES
FSM Encoding Algorithm             : Auto
Safe Implementation                : No
FSM Style                          : LUT
RAM Extraction                     : Yes
RAM Style                          : Block
ROM Extraction                     : Yes
Shift Register Extraction          : YES
ROM Style                          : Auto
Resource Sharing                   : YES
Asynchronous To Synchronous        : NO
Shift Register Minimum Size        : 2
Use DSP Block                      : Auto
Automatic Register Balancing       : No

---- Target Options
LUT Combining                      : Auto
Reduce Control Sets                : Auto
Add IO Buffers                     : YES
[... 2097 lines cut ...]
Cross Clock Domains Report:
--------------------------

Clock to Setup on destination clock clk_i
---------------+---------+---------+---------+---------+
               | Src:Rise| Src:Fall| Src:Rise| Src:Fall|
Source Clock   |Dest:Rise|Dest:Rise|Dest:Fall|Dest:Fall|
---------------+---------+---------+---------+---------+
clk_i          |    8.526|         |         |         |
---------------+---------+---------+---------+---------+

=========================================================================

Total REAL time to Xst completion: 35.00 secs
Total CPU time to Xst completion: 34.93 secs
 
--> 

Total memory usage is 4623924 kilobytes

Number of errors   :    0 (   0 filtered)
Number of warnings :  385 (   0 filtered)
Number of infos    :   57 (   0 filtered)

```

stnolting, 2021-07-27:
That's interesting.

> Synthesizing (advanced) Unit <neorv32_dmem>.
> INFO:Xst:3218 - HDL ADVISOR - The RAM <Mram_mem_ram_b0> will be implemented on LUTs either because you have described an asynchronous read or because of [...]

Seems like ISE has problems to map the BRAM's read- and write-enable signals. This is caused by the memory VHDL description, which implicitly eliminates the read-during-write behavior question (as there cannot be any read-during-write; it's either read or write).

The Spartan-6 BRAM primitives are not really different to the Series-7 ones. Vivado has no problems identifying the current DMEM description, so it seems to be an ISE issue. However, you could try using a different description for the DMEM (this is related to #101). For example, you could try the attached version of the DMEM:

* [neorv32_dmem.vhd.TXT](https://github.com/stnolting/neorv32/files/6885352/neorv32_dmem.vhd.TXT)

Remove the `.TXT` suffix and replace the original DMEM file `rtl/core/neorv32_dmem.vhd` by this one. Then try another synthesis. The DMEM should be mapped to BRAM now (hopefully). If this works, then the same code modifications can be done for the IMEM.

umarcor, 2021-07-31:
@Alirio926, this is just a friendly suggestion: you can change the extension of your logs/reports to `.txt` and then "drag & drop" the file into you comment in these issues. That will "upload" the file and show it as a link (see @stnolting's previous message). It makes easier for other users to grab them, without scrolling down all the issue.

Alirio926, 2021-08-03:
Hello, sorry abou delay to test.
Nice @umarcor noted.
Follow the new Syntetize detailled Report. 
[Syntetize report dmem8k + imem8k.txt](https://github.com/stnolting/neorv32/files/6925935/Syntetize.report.dmem8k.%2B.imem8k.txt)

On imem.vhd i push the code above to out process. same dmem, it's ok?
```vhd
-- read data --
  rdata <= mem_b3_rd & mem_b2_rd & mem_b1_rd & mem_b0_rd;
```
[neorv32_imem .vhd.txt](https://github.com/stnolting/neorv32/files/6925941/neorv32_imem.vhd.txt)

Compiled successfully, at night I will test it on the hardware.

stnolting, 2021-08-03:
> Hello, sorry abou delay to test.

No worries :wink:

> Follow the new Syntetize detailled Report.
Syntetize report dmem8k + imem8k.txt

According to your report, both IMEM and DMEM are mapped to block-RAM :+1:
But I am not sure here - which memory VHDL description for IMEM & DMEM did you use?

> neorv32_imem .vhd.txt

Cut-out from your file:

```vhdl
  -- read data --
  rdata <= mem_b3_rd & mem_b2_rd & mem_b1_rd & mem_b0_rd;
  -- output gate --
  data_o <= rdata when (rden = '1') else (others => '0');
```

:warning: This will work if the IMEM is implemented as RAM (top generic `INT_BOOTLOADER_EN => true`). But it will **not** work if the IMEM is implemented as ROM (top generic `INT_BOOTLOADER_EN => false`) as the read-data from the ROM-process ("imem_rom") is missing.

closed 2021-09-02 by Alirio926