stnolting/neorv32 #133: Some Problems using a Vivado-Block Design

AWenzel83, 2021-07-29:
Hello,

after quite some time I was finally using an up to date version of this core again, and the probems with the axi4lite-System template from [issue #8](https://github.com/stnolting/neorv32/issues/8) are back :
- IO_CFS_CONFIG needs a default value (like x"00000000")
- the default values of XIRQ_TRIGGER_TYPE and XIRQ_TRIGGER_POLARITY need to be changed from a generic initialization with "others" to a fixed one like above

and btw:
when did you remove the "upload"-target from the makefile? That was quite convenient (But not as important now, with a working debugger and the possibility to upload via JTAG)

stnolting, 2021-07-30:
Oh crap!
Seems like I keep forgetting about this Vivado issue over and over again... 🙈 
Anyway, thanks for the hint! I will fix that (again).

> when did you remove the "upload"-target from the makefile? That was quite convenient 

The bootloader required additional checks in each "state" so that it jumps back to "get executable via UART" when a special char is received. Unfortunately, that required more program space than I thought and I would like to keep the default bootloader below 4kB. That is why this feature has been removed

However, I just noticed the _raw_ upload script is still available: [`sw/image_gen/uart_upload.sh`](https://github.com/stnolting/neorv32/blob/master/sw/image_gen/uart_upload.sh)
The upload command needs to be changed from `#` to `u` and then it should still work - but only in the first 8 seconds after processor reset. I will also update that.

stnolting, 2021-08-12:
Fixed!

Please open a new issue if you encounter further packaging problems. Hopefully, this was the last time that I forget about those _fixed-size_ generic defaults... :smile:

closed 2021-08-12 by stnolting