stnolting/neorv32 #135: Missing documentation and Makefiles for iCE40 and ECP5 Lattice FPGAs

zipotron, 2021-08-04:
Hello everybody, first, thanks for this amazing work!
Well, I have at home the board AlhambraII (iCE40) and the ULX3S (ECP5), I am using Yosys with the GHDL plugin. I wanted to synthesize using Yosys and NextPNR for those boards, but couldn't find a start point to do it.
Please, if anyone support me I would like to do those Makefiles.
Thanks!

stnolting, 2021-08-05:
Hey there!

> first, thanks for this amazing work!

Thank you very much :)

> Well, I have at home the board AlhambraII (iCE40) and the ULX3S (ECP5), I am using Yosys with the GHDL plugin. I wanted to synthesize using Yosys and NextPNR for those boards, but couldn't find a start point to do it.
Please, if anyone support me I would like to do those Makefiles.

The documentation of the osflow setups is currently work-in-progress (see #96 and #131).
I am not familiar with the AlhambraII or the ULX3S boards, but the makefiles for iCE40 and ECP5 FPGAs are already there:
* [`setups/osflow/boards`](https://github.com/stnolting/neorv32/tree/master/setups/osflow/boards)

You will just need to adapt the pin-mappings:
* [`setups/osflow/constraints`](https://github.com/stnolting/neorv32/tree/master/setups/osflow/constraints)

Then you can run this:
```
neorv32$ make -C setups/examples BOARD=<fpga_board> <soc>
```

For example for the ECP5-based OrangeCrab board using a minimal NEORV32 SoC you can do:
```
neorv32$ make -C setups/examples BOARD=OrangeCrab MinimalBoot
```

zipotron, 2021-08-08:
Hello! Thanks for the information, is perfect start point for try to make it run in those board! I will let you know the results!

umarcor, 2021-08-08:
@zipotron, can you please confirm that you are using Alhambra II (HX4K) and not IceZum Alhambra (HX1K)? If so, I suggest you create a PR where:

- Add the constraints file to `setups/osflow/constraints`.
- Edit https://github.com/stnolting/neorv32/blob/master/setups/osflow/boards/index.mk and add an `ifeq` block for defining `CONSTRAINTS`, `PNRFLAGS` (device and package) and `IMPL ?= neorv32_$(BOARD)_II_$(ID)`.

We can then iterate on that.

zipotron, 2021-08-08:
@umarcor Confirmed! I have the Alhambra II (HX4K), I will start on that ASAP! (In two or three days I guess, depend on my work load)

zipotron, 2021-08-09:
PR done, I need help with the ULX3S Makefiles, I don't know why try to take iCE40 VHDL files instead ECP5.
AlhambraII is well done, but fails synthesis, seams something related with the optimization, I can work around with a bit of support

closed 2021-09-01 by zipotron

gojimmypi, 2022-12-03:
Looks like the setups have been moved here:

https://github.com/stnolting/neorv32-setups/tree/main/osflow/boards

stnolting, 2022-12-03:
Right, we sourced-out the FPGA implementations to a new repository.