stnolting/neorv32 #145: Radiant setup fails to build with Radiant 2.2.1.239.2

turbinenreiter, 2021-09-07:
**Describe the bug**

Building the project fails at the last step ("Export Files"), all prior steps work.

The build output:

```
************************************************************
**  Gate-Level Simulation File                            **
************************************************************

backanno "neorv32_upduino_v3_impl_1.udb"  -o "neorv32_upduino_v3_impl_1_vo.vo"      -sp "High-Performance_1.2V"  -w -neg
backanno: version Radiant Software (64-bit) 2.2.1.239.2

Loading neorv32_upduino_v3_impl_1.udb ...
Loading device for application GENERIC from file 'itpa08.nph' in environment: /home/plam/lscc/radiant/2.2/ispfpga.
Package Status:                     Preliminary    Version 1.5.
Performance Hardware Data Status:   Advanced       Version 1.0.

Design:  neorv32_upduino_v3_top
Family:  iCE40UP
Device:  iCE40UP5K
Package: SG48
Performance Grade:   High-Performance_1.2V
WARNING - Unable to find the instance/port 'flash_sdi_i' in the constraint 'ldc_set_location -site {17} [get_ports flash_sdi_i]'

WARNING - In the constraint 'ldc_set_location -site {17} [get_ports flash_sdi_i]', the locate object is not specified

Writing a verilog netlist based on the neorv32_upduino_v3_impl_1 design file.

Writing Verilog netlist to file neorv32_upduino_v3_impl_1_vo.vo
Writing SDF timing to file neorv32_upduino_v3_impl_1_vo.sdf
malloc_consolidate(): unaligned fastbin chunk detected
Done: error code -1
```

**To Reproduce**

Follow the build instructions described here: https://github.com/stnolting/neorv32/tree/master/setups/radiant/UPduino_v3
while using Radiant 2.2.1.239.2

**Expected behavior**

I expected the build to succeed and create output files

**Screenshots**

[image]

**Environment:**

 - Fedora 34
 - Radiant 2.2.1.239.2

**Hardware:**

- UPduino 3.0

**Additional context**

turbinenreiter, 2021-09-07:
I just realized that this error message looks more like one of the tools within Radiant is broken, and given the fact that I run it on Fedora 34 instead of Ubuntu 16.04(!), which is Radiants supported Linux distro, I guess the issue is not to be found in this project. I got the osflow toolchain working in the meantime. I think this can be closed.

stnolting, 2021-09-07:
Hey there!

I do not have Radiant version 2.2 so I cannot reproduce your setup.
The default Radiant setup from `setups/radiant` works with Radiant 2.1 and I just tested it using the latest Radiant 3.0 version - successfully. 

Seems like your design has some timing errors (5). Have you been prompted to execute the last step (export) even though there are timing errors? Maybe this crashed something., because the physical constraint warning for the `flash_sdi_i` does not really make sense here... 🤔 

If you do not want to update to Radiant 3.0, you could try the more sophisticated _Sinplify Pro_ synthesis engine instead of the default Lattice Synthesis Engine (LSE). From my experience Sinplify Pro provides faster (but larger) design results. Maybe this could eliminate the timing error issue.

To switch to Sinplify Pro:
* in the file list on the right right click on the implementation (`imp_1 (Lattice LSE`)
* in the drop-down menu select "properties"
* in the project properties click on "Lattice LSE" in the "Synthesis Tool: Lattice LSE" line
* select Sinplify Pro in the drop-down menu and close the dialog 
* rerun everything by clicking on "Run All" (green arrow)

stnolting, 2021-09-07:
> I just realized that this error message looks more like one of the tools within Radiant is broken, and given the fact that I run it on Fedora 34 instead of Ubuntu 16.04(!), which is Radiants [...]

Good point!

However, I suggest to try Sinplify before updating Radiant or any OS-related thing. :wink:

turbinenreiter, 2021-09-07:
I tried Sinplify and also got the same error message. I think the issue is incompatibility between my host OS and Radiant. Thanks for the help!

umarcor, 2021-09-07:
@turbinenreiter I suggest checking whether package `redhat-lsb` is installed. Some years ago, that was required for running some vendor tools on Fedora.

turbinenreiter, 2021-09-08:
> redhat-lsb

I installed it, same issue.

stnolting, 2021-09-08:
I think there is an issue in [`rtl/processor_templates/neorv32_ProcessorTop_UP5KDemo.vhd`](https://github.com/stnolting/neorv32/blob/master/rtl/processor_templates/neorv32_ProcessorTop_UP5KDemo.vhd), which is currently used by the Radiant example setup: the `flash_sdi_i` signal is defined in the entity but not actually used in the design. This _should not_ be a problem for Radiant, but maybe this is the cause of your export problem. I will fix that.

stnolting, 2021-09-08:
Should be fixed now (_fingers crossed_) :wink:

turbinenreiter, 2021-09-09:
> malloc_consolidate(): unaligned fastbin chunk detected

Still same issue. I will try this on a Radiant-support OS later to confirm that this is an OS issue rather than an neorv32 issue :D

turbinenreiter, 2021-11-02:
Can be closed, it worked on other OS, so the problem is on my (unsupported) OS.

closed 2021-11-02 by stnolting