stnolting/neorv32 #148: Windows workflow keeps failing

stnolting, 2021-09-10:
The FPGA implementation part of the Windows workflow keeps failing (cut-out from https://github.com/stnolting/neorv32/runs/3562298603?check_suite_focus=true#step:5:27).

```
yosys -m ghdl.so \
  -p \
  "ghdl --std=08 --no-formal neorv32_UPduino_BoardTop_MinimalBoot; \
  synth_ice40 \
  -top neorv32_UPduino_BoardTop_MinimalBoot -dsp \
  -json neorv32_UPduino_v3_MinimalBoot.json"  2>&1 | tee yosys-report.txt
ERROR: 
  This version of Yosys cannot load plugins at runtime.
  Some plugins may have been included at build time.
  Use option `-H' to see the available built-in and plugin commands.
```

Is this a package problem?!
@umarcor could you have a look? :upside_down_face:

umarcor, 2021-09-10:
I was just pulling to start rebasing and bringing myself up to date :wink:
As you saw, I've been quite busy in august, but I think I can allocate some hours this weekend to revisit the pending issues/workflows/etc.

stnolting, 2021-09-10:
> As you saw, I've been quite busy in august

Take your time! There is really no need to hurry. This issue is not super urgent+critical.

> but I think I can allocate some hours this weekend to revisit the pending issues/workflows/etc.

Thank you so much! 😊 :+1:

umarcor, 2021-09-10:
This is the reason: https://github.com/stnolting/neorv32/blame/79058c17e1c1b10d1573c8712d754bd22300f0f6/setups/osflow/boards/index.mk#L1

`GHDL_PLUGIN_MODULE` needs to be set when the toolchain has ghdl-yosys-plugin built as a module. In case the plugin is built into yosys, that must not be set. When adding support to Alhambra and ULX3S, that envvar was unconditionally set.

More precisely, on MSYS2, the plugin is built into yosys, because yosys does not support loading modules dynamically on Windows. Hence, I assume that @zipotron is either a GNU/Linux user, or he was using containers.

stnolting, 2021-09-10:
Ah I see. I was not thinking about that at all... 🙈 
Thank you so much for the quick response!

closed 2021-09-10 by stnolting

umarcor, 2021-09-10:
My pleasure 😃

PR #149 (2021-09-10): [setup/osflow/boards/index.mk] do not set GHDL_PLUGIN_MODULE unconditionally
Close #148.