stnolting/neorv32 #169: SDRAM controller for the ULX3S

zipotron, 2021-10-02:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

zipotron, 2021-10-02:
Hello again! I am studying how to enable the 32MB of SDRAM in the ULX3S, and I found in other projects interesting codes in VHDL that access to the ULX3S SDRAM (https://github.com/nullobject/sdram-fpga), but those are not Wishbone compatible (or at least in not mentioned). I found in the file "rtl/processor_templates/neorv32_ProcessorTop_MinimalBoot.vhd" that we have "MEM_EXT_EN" that suppose to be the connection with that SDRAM.
I would like to implement the controller to link the wishbone external memory with the VHDL ULX3S SDRAM. If its OK I start to do it and hopefully if my work allows me do a PR soon, but I will need a bit support. Do you like the idea?

umarcor, 2021-10-03:
See #105.

stnolting, 2021-10-03:
> and I found in other projects interesting codes in VHDL that access to the ULX3S SDRAM (https://github.com/nullobject/sdram-fpga), but those are not Wishbone compatible (or at least in not mentioned)

That looks interesting! I am also searching for a small SDRAM controller to tinker with one of my board's memories. Maybe I will give it a try. From a first look, the interface seems to be pretty straightforward, so it shouldn't be a big deal to connect it to the processor's external bus interface.

> I would like to implement the controller to link the wishbone external memory with the VHDL ULX3S SDRAM. If its OK I start to do it and hopefully if my work allows me do a PR soon, but I will need a bit support. Do you like the idea?

:+1:

We can help with the processor's bus interface, but for questions regarding the SDRAM controller you might be be better off contacting @nullobject.

zipotron, 2021-10-04:
Thanks @umarcor !
@stnolting  Yes, even better, I am trying to contact the Radiona team, as soon as I have something on what is possible to start working I will do PR. As usually, it will came as soon as my work allows me.

zipotron, 2021-10-17:
Hello again @stnolting and @umarcor! I got some improvement regarding the SDRAM. I got in touch with the ULX3S guys, first I got a Verilog 8-bit controller from one of their project (I select that because came with really interesting test support) and translate to VHDL. Because I had some doubts about how to make it works, I ask them, and after we make it work the owner of the project offer me to adapt the test to work with https://github.com/nullobject/sdram-fpga, that is 32-bit controller and will be better choice for integrate in NeoRV32! On the way they had to fix some defects detected when we pass the test to the Nullobject controller, but now seams working!
Just ULX3S team advice to me that we have to make double check, there is a wired trick done in sdram.vhd line 441 that may cause troubles, you can see our conversations here https://gitter.im/ulx3s/Lobby
I would like to ask you to have a look of the code before to start adapting to the Wishbone intervace and integrate it, the code is here: https://github.com/zipotron/sdram-fpga
Looking forward opinions and suggestions!

umarcor, 2021-10-17:
@zipotron, note that you can use Verilog sources together with NEORV32, through GHDL + Yosys. See example MixedLanguage for FOMU (https://github.com/stnolting/neorv32/tree/master/setups/osflow/board_tops) which is based on https://github.com/im-tomu/fomu-workshop/tree/master/hdl/mixed/blink. See https://github.com/stnolting/neorv32/blob/master/setups/osflow/Makefile#L92-L97. You should be able to create a WithSDRAM target by copying that snippet.

Nevertheless, I believe it is interesting to have the controller available in VHDL as well. I recommend channel https://gitter.im/vhdl/General for questions related to the language.

zipotron, 2021-10-18:
Hi @umarcor , yes, I know, thanks for the info but as you can see, the exposed code have the controller in VHDL and the test in Verilog, and this is the part of the job I did. I put an effort in provide VHDL controller for follow the style, and for other reason more egoist, I am good Verilog coder, but just beginner in VHDL, and this kind of task give me the excuse for study it.
Thanks also for the channel, is going a be useful for future, but my doubt now is functional, not regarding the language.

umarcor, 2021-10-18:
> as you can see, the exposed code have the controller in VHDL and the test in Verilog, and this is the part of the job I did. I put an effort in provide VHDL controller for follow the style, and for [...]

Oh, I misunderstood. I thought the whole controller was in Verilog and you were converting it to VHDL. So, do you want to keep doing mixed-language simulation (keep the test in Verilog) or do you want to provide a complete solution in VHDL? In the latter case, authors of open source verification frameworks (https://larsasplund.github.io/github-facts/) such as OSVVM, VUnit or cocotb do participate in https://gitter.im/vhdl/General. So, you can head there not only for questions related to the language, but to ask about the functionality you want to achieve and what are the alternatives.

BTW, you can alt+click on a gitter message, and you'll get a direct link to it, instead of a link to the channel.

zipotron, 2021-10-19:
Last thing, well, still early for ask this, but, I would like to have something for test the external RAM, like small program as a bootloader that play with leds if the memory is writable/readable...Anyone have already anything?

zipotron, 2021-10-31:
@stnolting I am testing!

> This is really really bad! sweat_smile
The bootloader encountered an unrecoverable exception - in this case it was trying to access the SPI module (at address 0xffffffa8) but there is no SPI module implemented in the ULX3S example setup. I think you might have executed the l command, right?

No, didn't press any key to get that, and I am still getting the same in the master branch:

> Loading...
[ERR 0x00000005 0xffff0d58 0xffffffa8]

> ERROR_1

Even I tried to increase the memory to 32kb, the same, and if I override the image "neorv32_bootloader_image.vhd" the error still 

> ERROR_3

I checkout the branch "ulx3s_add_wishbone" and I couldn't synthesize using "make -C setups/osflow BOARD=ULX3S MinimalBoot"
Now I am free, and online, we can try fixes, and I can test!!

stnolting, 2021-10-31:
Let's stay on the master branch at first.

> No, didn't press any key to get that, and I am still getting the same in the master branch:

Does this appear right after resetting the processor?

> ERROR_1

Can you post the whole bootloader output? Especially the hardware information it prints at the beginning.

stnolting, 2021-10-31:
> I checkout the branch "ulx3s_add_wishbone" and I couldn't synthesize using "make -C setups/osflow BOARD=ULX3S MinimalBoot"

Sorry, I have overseen that.
The PR uses a different _design_ now. `make -C setups/osflow BOARD=ULX3S Default` should word.

zipotron, 2021-10-31:
> Does this appear right after resetting the processor?

Just after the "hart" led stop blinking

> Can you post the whole bootloader output? Especially the hardware information it prints at the beginning.

That is the whole output!!! Just this 3 lines:

    Loading...
    [ERR 0x00000005 0xffff0d58 0xffffffa8]

    ERROR_1

stnolting, 2021-10-31:
Ah ok, I see.

This behavior is normal. After reset, the bootloader waits 8s for a user input. If there is no user input the bootloader starts the autoboot sequence and tries to fetch an executable via SPI from an external flash. Since there is no SPI module implemented in your design, you get this error: `[ERR 0x00000005 0xffff0d58 0xffffffa8]`

Can you upload an execute programs using the bootloader?
-> https://stnolting.github.io/neorv32/ug/#_uploading_and_starting_of_a_binary_executable_image_via_uart

To test this, please compile the [`sw/example/blink_led`](https://github.com/stnolting/neorv32/tree/master/sw/example/blink_led) example program using `make clean_all exe`.

Connect to your FPGA board via UART, reset the processor and wait for the bootloader to show up. Hit any key to skip the autoboot sequence, then press `u` and upload the `neorv32_exe.bin` file from `sw/example/blink_led`. After the upload is complete, you can execute the program by pressing `e`.

zipotron, 2021-10-31:
> This behavior is normal. After reset, the bootloader waits 8s for a user input. If there is no user input the bootloader starts the autoboot sequence and tries to fetch an executable via SPI from an [...]

Ouch! I am so sorry, I was expecting the menu pop-up after the led stop blinking! Yes, was working! I just try to push "H" key before the led stop blinking and is working! really sorry for make you work around for no reason, my fault. I just compile the Blink_led app and I am going a restart in Windows for uploading it! BTW, do you know any way to upload app from Linux?

stnolting, 2021-10-31:
:+1:

Can you upload and execute the `hex_viewer` program? That would be the next step targeting SDRAM support.

> BTW, do you know any way to upload app from Linux?

I only use graphical programs for Windows. On Linux I am just using the shell to interact with the UART's tty device. Not pretty, but handy :wink:

zipotron, 2021-10-31:
Blink_led app works!

> Can you upload and execute the hex_viewer program? That would be the next step targeting SDRAM support.

Sure, I am going a try to do in Linux using the shell... OK! Works!
I think we have all necessary to start thinking in the SDRAM support, I will close this thread and we can start a new one when I push the first draft of the PR.
Thanks!

closed 2021-10-31 by zipotron

stnolting, 2021-11-01:
Btw, we should continue the discussion regarding the Wishbone interface in #189

Mecrisp, 2022-03-04:
@zipotron I am from the dark side, and I wrote a basic SDRAM controller for ULX3S in January. It is neither wishbone nor VHDL, but it is intended for teaching and well commented. You may use it for guidance: https://github.com/BrunoLevy/learn-fpga/blob/master/FemtoRV/RTL/SDRAM/muchtoremember.v

zipotron, 2022-03-10:
@Mecrisp Hi! And thanks for came from the dark side and share your work! I had a look and seams really nice work. But unfortunately for my actual task don't help too much... I had already a a Verilog SDRAM controller (Is not mine but I tested and is working), and I wanted to re implement it in VHDL ( many reasons, like integrate with NeoRV32 without mixing VHDL and verilog, but the main reason, It was a personal change). And I did it but never works... I suspect that could be a GHDL bug regarding "inout" ports, but couldnt verify yet. And last weeks I had some troubles at work, and lets be honest, I lose a lot of motivation.
@stnolting suggested me to use Lattice Diamond for investigate, but I am waiting for the account validation for get the free license already two weeks...
Just in case you have curiosity, we were discussing it in this thread: https://github.com/emb4fun/neorv32-examples/issues/1#issuecomment-1053680617
Maybe I should go back to Verilog (In which I have more experience and knowledge) and then definitely use your controller!
Thanks a lot for care and help!!!