stnolting/neorv32 #171: Make Check Fails (riscv32-unknown-elf-gcc is not found)

AbdulrahmanKH7, 2021-10-05:
Hi,
Firstly thank you very much for this tremendous effort and I wish you all the best.
I am trying to install the prebuilt toolchain you made and when I test it in one of the provided examples (blinky) it gives me the following error:
---------------- Check: NEORV32_HOME folder ----------------
NEORV32_HOME: ../../..
---------------- Check: riscv32-unknown-elf-gcc ----------------
make: riscv32-unknown-elf-gcc: No such file or directory
make: *** [../../common/common.mk:207: check] Error 127

I searched and found that maybe it is a 32bit file and my Ubuntu (on a Windows VM) is 64bit so I installed the packages required to solve this:
architecture i386
libc6:i386 libncurses5:i386 libstdc++6:i386

 Also I installed the build-essential and gcc-multilib because maybe some tool is missing (Linker for example). 

Of course I used this command first: $ export PATH=$PATH:/opt/riscv/bin

and still the problem is not solved yet. I would appreciate any help. Thanks again.

stnolting, 2021-10-05:
Hey there!
Thank you very much for the nice words! :)

Regarding your issue:
I think there might be a problem with your toolchain installation. Please try running `riscv32-unknown-elf-gcc -v` in your console. If this fails, the binaries of the toolchain cannot be found. You should see `riscv32-unknown-elf-gcc` (and all the other binaries) when executing `ls /opt/riscv/bin`.

AbdulrahmanKH7, 2021-10-05:
Yes Isee them but I need to add this ~ before the path: ls ~/opt/riscv/bin

AbdulrahmanKH7, 2021-10-05:
I think I understand the problem now; I created an opt folder while there is one in the computer directory. I will try again with this folder and see.

stnolting, 2021-10-05:
> I need to add this ~ before the path: ls ~/opt/riscv/bin

I'm no Linux guru, but doesn't that mean `opt/riscv/bin` is in your home folder (`~` = `$HOME`)?

You could try modifying your PATH variable again: `export PATH=$PATH:/opt/riscv/bin`
Or copy `~/opt/riscv/bin` to `/opt/riscv/bin`.

stnolting, 2021-10-05:
> I think I understand the problem now; I created an opt folder while there is one in the computer directory. I will try again with this folder and see.

Seems like we had perfect timing with our responses! :smile: :+1:

AbdulrahmanKH7, 2021-10-05:
Yes we had =D
I moved the folder to `/opt/riscv/bin` but still the same problem..

stnolting, 2021-10-05:
Do you now see `riscv32-unknown-elf-gcc` when running `ls /opt/riscv/bin` (no tilde)?

If you cannot run `riscv32-unknown-elf-gcc -v` your PATH variable might be "corrupted". Try again in a new, clean shell (`export ...` and `riscv32-unknown-elf-gcc -v`).

AbdulrahmanKH7, 2021-10-05:
This is what I receive when I run `riscv32-unknown-elf-gcc -v`

Using built-in specs.
COLLECT_GCC=riscv32-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/opt/riscv/bin/../libexec/gcc/riscv32-unknown-elf/10.2.0/lto-wrapper
Target: riscv32-unknown-elf
Configured with: /home/stnolting/tmp/riscv-gnu-toolchain/riscv-gcc/configure --target=riscv32-unknown-elf --prefix=/tmp/riscv --disable-shared --disable-threads --enable-languages=c,c++ --with-system-zlib --enable-tls --with-newlib --with-sysroot=/tmp/riscv/riscv32-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././riscv-gcc --disable-multilib --with-abi=ilp32 --with-arch=rv32i --with-tune=rocket 'CFLAGS_FOR_TARGET=-Os   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os   -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 10.2.0 (GCC)

stnolting, 2021-10-05:
That look good! :+1:
Now `neorv32/sw/example/blink_led$ make check` should work as well.

AbdulrahmanKH7, 2021-10-05:
Still not working, maybe because my example path is:
`/neorv32/sw/example/blink_led$`
with a slash at the beginning?

stnolting, 2021-10-05:
You just need to navigate (using `cd`) into that folder and run `make check` _there_.

AbdulrahmanKH7, 2021-10-05:
It worked! Thanks a lot, sorry I am new to Linux so I needed sometime =D.
Kindest regards!

closed 2021-10-05 by AbdulrahmanKH7

stnolting, 2021-10-05:
Good to hear! :+1:

> Thanks a lot, sorry I am new to Linux so I needed sometime =D.

No worries! Linux (and especially VMs) can be tricky sometimes. :wink: