stnolting/neorv32 #193: makefile upload target missing or deprecated?

balux33, 2021-11-01:
Hi!
Previously i used the make upload command to upload program via uart and bootloader 
and generally worked well.
Newer versions no longer include the upload target.
Is it just left out or has it become obsolete?
if so, what is the new way to upload via uart?

stnolting, 2021-11-01:
Hey!
Yes, that feature was deprecated because it was never really reliable.

The standard way to upload executables via UART is to use the bootloader console in a terminal program. Right now, there is no default option to upload executables directly from the shell. However, the underlying script is still available [`sw/image_gen/uart_upload.sh`](https://github.com/stnolting/neorv32/blob/master/sw/image_gen/uart_upload.sh) but I have not tested that with never versions.

Recently, I stumbled across this: https://github.com/islandcontroller/neorv32load I have not tried that myself but it looks handy!

stnolting, 2021-11-17:
@balux33 
If you have no further questions this can be closed, right?

closed 2021-11-28 by stnolting