stnolting/neorv32 #197: Instances of uninferred RAM logic

ownway88, 2021-11-03:
I tried to implement the neorv32 core to a cyclone II device EP2C8Q208C8. The tool I use is Quartus II 13.0 64-bit.
I followed the well documented user guide step by step to build a simple project. But I ran into errors during Analysis & Synthesis.

Info (276014): Found 8 instances of uninferred RAM logic
	Info (276007): RAM logic "neorv32_top:neorv32_top_inst|neorv32_imem:\neorv32_int_imem_inst_true:neorv32_int_imem_inst|mem_ram_b1" is uninferred due to asynchronous read logic
	Info (276007): RAM logic "neorv32_top:neorv32_top_inst|neorv32_dmem:\neorv32_int_dmem_inst_true:neorv32_int_dmem_inst|mem_ram_b1" is uninferred due to asynchronous read logic
	Info (276007): RAM logic "neorv32_top:neorv32_top_inst|neorv32_imem:\neorv32_int_imem_inst_true:neorv32_int_imem_inst|mem_ram_b0" is uninferred due to asynchronous read logic
	Info (276007): RAM logic "neorv32_top:neorv32_top_inst|neorv32_dmem:\neorv32_int_dmem_inst_true:neorv32_int_dmem_inst|mem_ram_b0" is uninferred due to asynchronous read logic
	Info (276007): RAM logic "neorv32_top:neorv32_top_inst|neorv32_imem:\neorv32_int_imem_inst_true:neorv32_int_imem_inst|mem_ram_b3" is uninferred due to asynchronous read logic
	Info (276007): RAM logic "neorv32_top:neorv32_top_inst|neorv32_dmem:\neorv32_int_dmem_inst_true:neorv32_int_dmem_inst|mem_ram_b3" is uninferred due to asynchronous read logic
	Info (276007): RAM logic "neorv32_top:neorv32_top_inst|neorv32_imem:\neorv32_int_imem_inst_true:neorv32_int_imem_inst|mem_ram_b2" is uninferred due to asynchronous read logic
	Info (276007): RAM logic "neorv32_top:neorv32_top_inst|neorv32_dmem:\neorv32_int_dmem_inst_true:neorv32_int_dmem_inst|mem_ram_b2" is uninferred due to asynchronous read logic

Error (276003): Cannot convert all sets of registers into RAM megafunctions when creating nodes. The resulting number of registers remaining in design exceeds the number of registers in the device or the number specified by the assignment max_number_of_registers_from_uninferred_rams. This can cause longer compilation time or result in insufficient memory to complete Analysis and Synthesis

I doubt it is a resource limitation from this device, I tried to reduced the default size of IMEM and DMEM anyway, it is not helpful, still the same error -8 instances of uninferred RAM logic.  Can someone give me some hints? Thanks!!!

stnolting, 2021-11-03:
I do remember this problem from working with old Quartus versions at college... 🙈 
According to the Altera forum this seems to be a tool issue. However, you could try using the "recommended coding style for RAMs" for the Cyclone II - maybe the old version of Quartus can identity that and actually infer block RAM.

* 💾 [neorv32_dmem.default.vhd.TXT](https://github.com/stnolting/neorv32/files/7468623/neorv32_dmem.default.vhd.TXT)

Download the file, remove the `.TXT` suffix and replace the default `rtl/core/mem/neorv32_dmem.default.vhd` file by this one. This modified DMEM version uses a register for buffering the address for read-accesses. The same modifications could be done for the IMEM and the bootloader ROM without effecting functionality.

ownway88, 2021-11-03:
Yes, :) 
Unfortunately the last version of Quartus which support Cyclone II is 13.0..
I implemented the suggested changes to DMEM and IMEM (not the boot rom, that file seems to be OK), now the compilation is done successfully! Thank you!

umarcor, 2021-11-03:
@stnolting, what about adding that file to a subdir in `setups/quartus`, along with a README?

stnolting, 2021-11-03:
@umarcor 
Good idea. But what about putting them as `neorv32_*mem.cyclone2.vhd` or `neorv32_*mem.altera.vhd` in `rtl/core/mem`?

umarcor, 2021-11-03:
@stnolting that sounds good to me! You are the one not convinced about adding multiple descriptions in `rtl/core/mem` 😆 . I pushed for the creation of that subdir precisely for use cases such as this one.

stnolting, 2021-11-03:
Seems like reality convinced me once again :smile:
How should we name those new *MEM files? `altera` or `cyclone2` might be misleading since there are no primitives used. It is just a different HDL style. Maybe `alternate`?

umarcor, 2021-11-03:
@stnolting I think it is good to make it explicit that this is for some older version of quartus. According to @ownway88, there is some "last version of Quartus supporting Cyclone II". Therefore, I think `neorv32_*mem.cyclone2.vhd` is correct. If we find this description to be required for some other tool/device, we can later rename it.

closed 2021-11-04 by stnolting