stnolting/neorv32 #206: ❌ Implementation workflow failing

stnolting, 2021-11-13:
Currently, all tasks of the implementation workflow (triggered by PRs) keep failing with the following error message:

```
synth_name: cannot handle IIR_KIND_SELECTED_ELEMENT (../../rtl/core/neorv32_fifo.vhd:127:43)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (v1.0.0-859-g1f3b1857) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-linux-gnu
/github/workspace/setups/osflow/
Command line:

Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
******************************************************************
```

I am not sure if this is an issue with a new GHDL / GHDL-yosys version. The code line from the error message looks like this:
https://github.com/stnolting/neorv32/blob/7ae64d7d95973fbf77ed0a5cd602aa3105219103/rtl/core/neorv32_fifo.vhd#L127

`fifo.r_pnt` and `fifo.w_pnt` might be zero here so there is a comparison of two zero-range vectors. This _should_ be no VHDL / code problem, right? Quartus, Vivado and Radiant do not complain about this. :thinking:

@umarcor could have a look?

umarcor, 2021-11-13:
Since `rtl/core/neorv32_fifo.vhd` was not modified in the last month, this is a legit bug/regression in GHDL. Regardless of the code being correct, GHDL should not crash.
@stnolting, can you please open an issue in https://github.com/ghdl/ghdl? As a reproducer, just tell that current `master` of NEORV32 is failing with latest GHDL since a few hours ago. Any of the failing CI commands is a valid reproducer (e.g `make -C setups/osflow BOARD=UPduino MinimalBoot`).

stnolting, 2021-11-13:
-> https://github.com/ghdl/ghdl/issues/1911

stnolting, 2021-11-14:
@umarcor thanks for fixing :)

closed 2021-11-14 by stnolting