stnolting/neorv32 #218: FPGA Board template for supported FPGAs on osflow

lab-mathias-claussen, 2021-11-23:
Hi,

as the NEORV32 supports gernerically the iCE40upK5 and other FPGAs it would be great to have a template for adding a new, currently not supported board. So adding support for all the baord arround there shoudl be fairly easy and would allow for easy contrbution. 

#217 is one of the examples where most of the work is copy and past. 

Best Regards

stnolting, 2021-11-23:
Interesting idea! But I am not exactly sure how to do this... :thinking: Do you propose a specific structure?

lab-mathias-claussen, 2021-11-24:
Hi,

I woudl add in a first step a template board like, Template_ICE40_UP5K with all requiered files and a bare minimum of vhdl and constrains (No actual pin mapping). Keep in this template boardnames with $Board , $Config, to allow later even more automation for it. 

This is not the best idea for sure.

stnolting, 2021-11-27:
I like that idea. But basically, all the already available osflow setups could be used as templates for adding/porting a new setup.

I think it might be better (and maybe easier to maintain :wink:) to add a "porting todo-list" to the osflow README [`setups/osflow/README.md`](https://github.com/stnolting/neorv32/blob/master/setups/osflow/README.md). We could use your port of the iCEBreaker board from #217 to illustrate the required steps. :thinking:
But I'm open for discussion.

closed 2021-12-14 by stnolting

umarcor, 2022-01-09:
First off, I'm sorry about taking so long to reply here. This was a crazy month!

My proposal is to submodule [hdl/constraints](https://github.com/hdl/constraints). The purpose of hdl/constraints is precisely to avoid duplicating PCF/LPF/XDC files in each RTL/(V)HDL project. The fact that all constraints files in this repo are located in `setups/osflow/constraints/` and separated from other (make)files is meant to make it easier to replace the directory with a submodule. See also hdl/constraints#5.
Submoduling hdl/constraints would reduce the addition of new boards to the creation of a BoardTop file, plus editing two makefiles.

However, I am unsure about submoduling hdl/constraints in this repo. Maybe it's time to move subdir `setups` to `neorv32-setups`, and have both neorv32 and hdl/constraints submoduled there. @stnolting wdyt?

umarcor, 2022-01-09:
See https://github.com/umarcor/neorv32-setups/tree/constraints-submodule.

stnolting, 2022-01-11:
> However, I am unsure about submoduling hdl/constraints in this repo. Maybe it's time to move subdir setups to neorv32-setups, and have both neorv32 and hdl/constraints submoduled there. @stnolting wdyt?

I think this is a _very good_ idea!