stnolting/neorv32 #250: why doesn't make pdf work?

salmansheikh, 2022-01-06:
I installed asciidoctor and ruby-asciidoctor-pdf in Ubuntu 20.04 (WSL).
Then it complained about asciidoctor-diagram. 
Then I installed it using a gem install asciidoctor-diagram

Then when I typed make pdf I got all kind of errors:

 make pdf
[ -f revnumber.txt ] && REVNUMBER='-a revnumber='"$(cat revnumber.txt)" || unset REVNUMBER; \
asciidoctor-pdf $REVNUMBER \
  -a pdf-theme=neorv32-theme.yml \
  -r asciidoctor-diagram \
  datasheet/main.adoc \
  --out-file public/pdf/NEORV32.pdf
Traceback (most recent call last):
        6: from /usr/bin/asciidoctor-pdf:25:in `<main>'
        5: from /usr/lib/ruby/vendor_ruby/asciidoctor/cli/options.rb:274:in `parse!'
        4: from /usr/lib/ruby/vendor_ruby/asciidoctor/cli/options.rb:274:in `each'
        3: from /usr/lib/ruby/vendor_ruby/asciidoctor/cli/options.rb:276:in `block in parse!'
        2: from /usr/lib/ruby/vendor_ruby/polyglot.rb:65:in `require'
        1: from /usr/lib/ruby/2.7.0/rubygems/core_ext/kernel_require.rb:92:in `require'
/usr/lib/ruby/2.7.0/rubygems/core_ext/kernel_require.rb:92:in `require': cannot load such file -- asciidoctor-diagram (LoadError)
        20: from /usr/bin/asciidoctor-pdf:25:in `<main>'
        19: from /usr/lib/ruby/vendor_ruby/asciidoctor/cli/options.rb:274:in `parse!'
        18: from /usr/lib/ruby/vendor_ruby/asciidoctor/cli/options.rb:274:in `each'
        17: from /usr/lib/ruby/vendor_ruby/asciidoctor/cli/options.rb:276:in `block in parse!'
        16: from /usr/lib/ruby/vendor_ruby/polyglot.rb:65:in `require'
        15: from /usr/lib/ruby/2.7.0/rubygems/core_ext/kernel_require.rb:156:in `require'
        14: from /usr/lib/ruby/2.7.0/rubygems/core_ext/kernel_require.rb:168:in `rescue in require'
        13: from /usr/lib/ruby/2.7.0/rubygems/core_ext/kernel_require.rb:168:in `require'
        12: from /var/lib/gems/2.7.0/gems/asciidoctor-diagram-2.2.1/lib/asciidoctor-diagram.rb:1:in `<top (required)>'
        11: from /var/lib/gems/2.7.0/gems/asciidoctor-diagram-2.2.1/lib/asciidoctor-diagram.rb:1:in `require_relative'
        10: from /var/lib/gems/2.7.0/gems/asciidoctor-diagram-2.2.1/lib/asciidoctor-diagram/a2s.rb:2:in `<top (required)>'
         9: from /var/lib/gems/2.7.0/gems/asciidoctor-diagram-2.2.1/lib/asciidoctor-diagram/a2s.rb:2:in `require_relative'
         8: from /var/lib/gems/2.7.0/gems/asciidoctor-diagram-2.2.1/lib/asciidoctor-diagram/a2s/extension.rb:1:in `<top (required)>'
         7: from /var/lib/gems/2.7.0/gems/asciidoctor-diagram-2.2.1/lib/asciidoctor-diagram/a2s/extension.rb:1:in `require_relative'
         6: from /var/lib/gems/2.7.0/gems/asciidoctor-diagram-2.2.1/lib/asciidoctor-diagram/a2s/converter.rb:2:in `<top (required)>'
         5: from /var/lib/gems/2.7.0/gems/asciidoctor-diagram-2.2.1/lib/asciidoctor-diagram/a2s/converter.rb:2:in `require_relative'
         4: from /var/lib/gems/2.7.0/gems/asciidoctor-diagram-2.2.1/lib/asciidoctor-diagram/util/cli_generator.rb:2:in `<top (required)>'
         3: from /usr/lib/ruby/vendor_ruby/polyglot.rb:65:in `require'
         2: from /usr/lib/ruby/2.7.0/rubygems/core_ext/kernel_require.rb:92:in `require'
         1: from /usr/lib/ruby/2.7.0/rubygems/core_ext/kernel_require.rb:92:in `require'
/usr/lib/ruby/2.7.0/tempfile.rb:121:in `<top (required)>': superclass mismatch for class Tempfile (TypeError)

stnolting, 2022-01-06:
I am not sure where this errors comes from, but it seems like there is a problem with your installation.
Maybe you could try a clean re-install.

If want to have the project's [data sheet](https://stnolting.github.io/neorv32/) as a PDF you can download the automatically build PDFs from the "[nightly](https://github.com/stnolting/neorv32/releases/tag/nightly)" release:
[💾 NEORV32-nightly.pdf](https://github.com/stnolting/neorv32/releases/download/nightly/NEORV32-nightly.pdf)

stnolting, 2022-01-07:
Do the other targets work with your installation? Like `make html`?

salmansheikh, 2022-01-07:
Yeah, make html worked. Thanks.

umarcor, 2022-01-07:
As @stnolting said, there seems to be some issue with the asciidoctor-diagram installation on WSL. Alternatively, you can `make -C docs container` to run asciidoctor on a container.

stnolting, 2022-02-03:
Can this be closed?

salmansheikh, 2022-02-03:
yes. Close it.

closed 2022-02-03 by salmansheikh