stnolting/neorv32 #257: User Guide - 1.3. Installation / Wrong export Statement

befedo, 2022-01-23:
**Describe the bug**
The user guide chapter 1.3. installation has a wrong export statement for the PATH variable.

**To Reproduce**
```{bash}
$ export PATH:$PATH:/opt/riscv/bin
bash: export: `PATH:/usr/local/bin:/usr/bin:/usr/local/sbin:/opt/riscv/bin': not a valid identifier
```

**Expected behavior**
```{bash}
$ export PATH=$PATH:/opt/riscv/bin && echo ${PATH}
/usr/local/bin:/usr/bin:/usr/local/sbin:/opt/riscv/bin
```
> Replace first `:` with `=`.

**Environment:**
```{bash}
$ bash --version
bash --version
GNU bash, version 5.1.16(1)-release (x86_64-pc-linux-gnu)
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

This is free software; you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

Can anyone confirm this or is something in my setup wrong?

Best Regards,
Befedo

umarcor, 2022-01-23:
I can confirm that the fix looks good. Do you want to submit a PR? The file to be fixed is https://github.com/stnolting/neorv32/blob/master/docs/userguide/sw_toolchain_setup.adoc#13-installation.

befedo, 2022-01-23:
Hello @umarcor,

I've added the pull request #258 if it can be merged we can close this issue.

Thanks for your time and best regards,
Befedo

stnolting, 2022-01-24:
Thanks for fixing!

PR #258 (2022-01-24): [DOC] User Guide - 1.3. Installation
fix(doc): Closes #257 and changes `export PATH:$PATH:/opt/riscv/bin` to
`export PATH=$PATH:/opt/riscv/bin`.
commit 62625799 (2022-01-24): [DOC] User Guide - 1.3. Installation

fix(doc): Closes #257 and changes `export PATH:$PATH:/opt/riscv/bin` to
`export PATH=$PATH:/opt/riscv/bin`.
closed 2022-01-24 by stnolting