stnolting/neorv32 #266: [idea} add option to convert processor setup to Verilog

stnolting, 2022-01-29:
There are some (commercial) tools out there that have problems handling VHDL source files. Maybe it would be a nice feature to provide some kind of script that allows to convert a certain processor setup (processor top entity, [processor templates](https://github.com/stnolting/neorv32/tree/master/rtl/processor_templates), [integration wrappers](https://github.com/stnolting/neorv32/tree/master/rtl/system_integration), ...) into a Verilog file/netlist. Having a NEORV32 version that can be simulated using [Verilator](https://www.veripool.org/verilator/) would be really awesome! 🚀 

GHDL + yosys-plugin is capable of converting VHDL files into Verilog files (http://ghdl.github.io/ghdl/using/Synthesis.html#cmdoption-ghdl-out):
```
yosys -m ghdl -p 'ghdl filename.vhdl -e unit_name; write_verilog filename.v'
```

I am not sure about the "quality" of the resulting Verilog source files nor do I know if they can deal with certain features (bi-direction/tri-state signals?). Maybe anyone has some experience here?

umarcor, 2022-01-29:
`ghdl synth` allows converting (synthesisable) VHDL code to Verilog, without requiring Yosys. The difference is that ghdl synth produces an unoptimised netlist, while yosys can run multiple optimisation stages. Option [--out=verilog](http://ghdl.github.io/ghdl/using/Synthesis.html#cmdoption-ghdl-out) , which you referenced, corresponds to ghdl synth, not to ghdl-yosys-plugin. Similarly, `--out=vhdl` or `--out=raw-vhdl` allow generating VHDL 19993 netlists that vendor tools with poor VHDL support can handle.

Verilator is mostly limited to synthesisable sources, hence there should not be much difference between ghdl synth or ghdl-yosys-plugin.

stnolting, 2022-02-05:
I need to check that out.

Can you say something about the "code quality" of the verilog code generated by GHDL or yosys?
I think it would be nice to have a verilog-only setup in neorv32-setups.

umarcor, 2022-02-05:
Implementation should be as easy as adding the following target to [neorv32-setups: osflow/synthesis.mk](https://github.com/stnolting/neorv32-setups/blob/main/osflow/synthesis.mk):

```makefile
neorv32.v: neorv32-obj08.cf ${DESIGN_SRC} ${BOARD_SRC}
	ghdl synth $(GHDL_FLAGS) --out=verilog $(TOP)
```

> Can you say something about the "code quality" of the verilog code generated by GHDL or yosys?

The difference is the purpose of each tool, not the quality.

- GHDL is a VHDL toolkit and it's focused on understanding the VHDL language, and allowing to use it (for simulation, cosimulation, synthesis, linting, documentation, etc.). None of those are "the best", but they are all ready-to-use and are good enough (comparable to the performance/quality of vendor tools).
- Yosys is a synthesis toolkit and it's focused on transformations/modifications of a design which is synthesisable and described in one specific language/syntax/IR/API format. The optimised/transformed synthesisable designs can be used for formal analysis, and dealing with solvers is also in the scope of Yosys (through SymbiYosys).
- Verilator is a transpilation toolkit to convert the synthesisable subset of Verilog into C++ and optimise it for multi-core and fast simulation.
  - Recently, work has been done by Antmicro and others in order to add non-synthesisable code support to Verilator, in their way to having open source support for using System Verilog and UVM.
- Surelog is a System Verilog toolkit focused on understanding the language. The scope is comparable to GHDL's, but the difference is that GHDL is 20y old, and Surelog is relatively recent. Hence, Surelog does not have built-in simulation or synthesis support; it was designed since the beginning to be plugged into existing tools, such as Yosys or Verilator.

Therefore, the following should be equivalent:

- GHDL -> ghdl-yosys-plugin -> Yosys -> Verilator
- GHDL synth -> Yosys -> Verilator

I'm unsure about the following (would be an interesting experiment to have some experimental results):

- GHDL -> ghdl-yosys-plugin -> Yosys -> Verilator
- GHDL synth -> Verilator

You can also do:

- GHDL synth -> VHDL -> Vivado
- GHDL -> ghdl-yosys-plugin -> Yosys -> Vivado (see [FPGA Interchange format to enable interoperable FPGA tooling](https://antmicro.com/blog/2021/09/fpga-interchange-format/))

Disclaimer: I work part-time for Antmicro since this week.

stnolting, 2022-02-10:
> Implementation should be as easy as adding the following target to [neorv32-setups: osflow/synthesis.mk](https://github.com/stnolting/neorv32-setups/blob/main/osflow/synthesis.mk):
> [...]
> I'm unsure about the following (would be an interesting experiment to have some experimental results):

Do you think it's worth a PR?

> The difference is the purpose of each tool, not the quality.

Thanks for the overview!

> Disclaimer: I work part-time for Antmicro since this week.

Awesome! Great to hear! 👍

umarcor, 2022-02-10:
> Looking at #264 do you think it's worth a PR?

Yes. Microwatt uses this approach already. Ideally, conversion from VHDL would done by LiteX or some other generalised wrapper/helper. However, at the moment, adding that target is the most straightforward solution. I'm in doubt about the top level, though. Since LiteX does have board information, maybe a dummy top needs to be used.

stnolting, 2022-08-26:
I am tinkering with the NEORV32 as plain Verilog netlist. I have tested the version using GHDL `synth` as well as the GHDL/GHDL-Yosys approach. :tada:

The "straightforward" GHDL/Yosys approach does not work at all because of some combinatorial loops inside the CPU core (?!?!?). It seems like some internal registers have been turned into a latch-based setup... However, when using some of Yosys's optimization options the resulting netlist can be successfully synthesized (tested with Vivado) - with the UART producing some weird outputs sometimes...

The approach using GHDL's `synth` option (almost) worked out of the box :rocket: I have successfully synthesized the design (Vivado) and also simulated it using Vivado's ISIM and [Icarus Verilog](http://iverilog.icarus.com/). It appears that the processor's internal hierarchy remains intact and many of the "most important" signals also keep their names - so inspecting certain signals in simulation works quite well. However, I had to do some minor rtl edits to make the simulation run - I will update the code base accordingly.

Interestingly, the hardware utilization (Vivado) is way smaller for the plain-GHDL approach than it is for the Yosys (+optimizations) one.

Anyway, I would like to share the _NEORV32-wrapper-to-verilog-netlist_ setup in some way, but I am not sure how... :thinking:
* add that to the [neorv32-setups](https://github.com/stnolting/neorv32-setups) repository?
* add it to the [`rtl/system_integration`](https://github.com/stnolting/neorv32/tree/main/rtl/system_integration) folder of this repository?
* or maybe even create a new repository like "neorv32-verilog" for that?

Furthermore, I think it would be nice to have a GitHub Workflow that does the to-Verilog conversion (using GHDL) and that also runs a simple "Hello World" simulation using Icarus Verilog.

Any suggestions anyone?

stnolting, 2022-08-28:
For the records 😉

* [neorv32-verilog](https://github.com/stnolting/neorv32-verilog) repository
* [icarus-verilog-prebuilt](https://github.com/stnolting/icarus-verilog-prebuilt) repository

closed 2022-08-28 by stnolting