stnolting/neorv32 #269: [CFU] does anyone know or actual CFU-Playground applications?

stnolting, 2022-02-03:
@google provides some kind of framework with their [CFU-Playground](https://github.com/google/CFU-Playground) to add custom instructions to a RISC-V design. There is no real description of the interface, latency, et.c in their repo (at least if have found none) but it seems like @grayresearch has drafted some general specifications in https://github.com/grayresearch/CFU.

Does anyone know of _real_ CFU implementations? The one(s) in the CFU-Playground repository are just exemplary. I would like to check those CFUs for interoperability with the recent NEORV32 `Zfxcfu` ISA extension update (#264).

umarcor, 2022-02-03:
@stnolting did you see https://cfu-playground.readthedocs.io/en/latest/projects/mnv2_first.html ?

@tcal-x @mithro @kgugala are you aware of other open source CFUs?

stnolting, 2022-02-04:
That looks great! I will have a closer look.
But still I cannot find the actual HDL files!? 🤔

umarcor, 2022-02-05:
> That looks great! I will have a closer look. But still I cannot find the actual HDL files!? 🤔

I believe it's Migen, so you need to generate/export the HDL (Verilog). However, you can do it once, and then deal with the Verilog as if it was handwritten.

closed 2022-12-20 by stnolting