stnolting/neorv32 #270: Odd Information Written on UART During Execution

AbdulrahmanKH7, 2022-02-03:
Hello,

I hope you are well.

Could you please help with the picture below?

[image]

Do you have an idea what does this means? What is the source of execution failure?

Thank you and Best Regards
Abdulrahman

stnolting, 2022-02-03:
These are reports generated by the "debug trap handlers" of the NEORV32 runtime environment (I just realized that I really need to add more information to the data sheet regarding this... :wink:).

* "Instruction access fault" means there was no ACK generated when reading instruction data from the address given by `PC`.
* "Load address misaligned" is an exception that occurs when you try (for example) to read a 32-bit data word from an unaligned address like `0x8000_0002`.
* "Store address misaligned" refers to the same problem but for write operations.

Can you provide the code or some fragment that caused all those exceptions?

AbdulrahmanKH7, 2022-02-03:
Sure, it is a modified version of the newhope code available online. I attached for you the modified version.
You need only to call this function at the main file you create:
test_keys();
[newhope_modified.zip](https://github.com/stnolting/neorv32/files/7994019/newhope_modified.zip)
Many thanks.

AbdulrahmanKH7, 2022-02-03:
Edit:
Better use this main easier for you:
#include <stdint.h>
#include <neorv32.h>
#include "newhope/api.h"

#define printf neorv32_uart_printf

/**********************************************************************//**
 * @name User configuration
 **************************************************************************/
/**@{*/
/** UART BAUD rate */
#define BAUD_RATE 19200
/** Use the custom ASM version for blinking the LEDs defined (= uncommented) */
//#define USE_ASM_VERSION
/**@}*/

int main() {

 neorv32_uart_setup(BAUD_RATE, PARITY_NONE, FLOW_CONTROL_NONE);
   // capture all exceptions and give debug info via UART
  // this is not required, but keeps us safe
  neorv32_rte_setup();

  // check available hardware extensions and compare with compiler flags
  neorv32_rte_check_isa(0); // silent = 0 -> show message if isa mismatch

printf("START TEST\n");

test_keys();

  return 0;
}

stnolting, 2022-02-03:
> `printf("START TEST\n");`

This might be the problem. The default `printf` from `stdio` is not supported "out of the box". It depends on your `stdio` library how this functions behaves in terms of low-level hardware access. So you would need to provide some NEORV32-specific functions to tell `printf` how to actually write chars to the UART module.

However, `printf` is quite heavy. Hence, I highly recommend using the NEORV32-flavored version `neorv32_uart0_printf`:

https://github.com/stnolting/neorv32/blob/75c22a83228abb04b1eea1c3dc48b509f9dfd3b3/sw/lib/source/neorv32_uart.c#L317-L381

If you need all the features of `printf` then you could use `ee_printf` from [`sw/example/coremark/ee_printf.c`](https://github.com/stnolting/neorv32/blob/master/sw/example/coremark/ee_printf.c).

AbdulrahmanKH7, 2022-02-03:
I am using the neo printf but with #define, see the beginning of the main code given:
#define printf neorv32_uart_printf

stnolting, 2022-02-03:
Oh right, I just saw that 😅 sorry...

What hardware version are you using?

Maybe `test_keys();` is where all the exceptions are generated. What does this function do?

stnolting, 2022-02-03:
Btw, you should `not` include `stdio.h`. Maybe this is also raising some exceptions because of some library parts might be linked.

stnolting, 2022-02-03:
> `#define printf neorv32_uart_printf`

This macro definition is incomplete as it lacks the argument(s). I think it should look like this:

```c
#define printf(...) neorv32_uart0_printf(__VA_ARGS__)
```

AbdulrahmanKH7, 2022-02-03:
Surely test_keys(); is the reason as it calls all the other code functions, I tried not using #define and replacing each printf with neoprintf, also removed the stdio.h but still no progress.

AbdulrahmanKH7, 2022-02-03:
I think hw version is 1.6.1.7 from the change log file

stnolting, 2022-02-03:
Could you provide more of your code?

AbdulrahmanKH7, 2022-02-03:
I provided full code above. Ok I will try that.

stnolting, 2022-02-03:
> I provided full code above.

I meant the sources of the `test_keys();` function. I think that might be the cause of the problem.

AbdulrahmanKH7, 2022-02-03:
Everything is there in the attached zip file (2nd reply)

AbdulrahmanKH7, 2022-02-04:
No I don't have. Yes there are warnings and the toolchain is the one you built.
I will look at the warnings and try the new HW on Sunday.

stnolting, 2022-02-04:
What kind of warnings do you get? Maybe this can help to figure out what is going wrong.

You could also add print statements before and after each function call to see where things are going wrong. If you have openOCD and a JTAG adapter you could also use the processor's [on-chip debugger](https://stnolting.github.io/neorv32/ug/#_debugging_using_the_on_chip_debugger) to step through your code.

stnolting, 2022-02-04:
For the records:

I have reworked the data sheet's section about the the runtime environment and the trap messages generated by the RTE:
* :books: [NEORV32 Runtime Environment (RTE)](https://stnolting.github.io/neorv32/#_neorv32_runtime_environment)

AbdulrahmanKH7, 2022-02-06:
I tried the new HW but the issue is still not solved. I will try to troubleshoot the problem in detail. Thanks for your help. I will close the thread.

closed 2022-02-06 by AbdulrahmanKH7

stnolting, 2022-02-07:
Do not hesitate to open a new issue if need any kind of help.