stnolting/neorv32 #284: CSRs and trap while in interrupt handler

Mecrisp, 2022-03-05:
Moin Stefan,

nice to meet you again in RISC-V territories!

We were busy on the Verilog side, too:
https://github.com/BrunoLevy/learn-fpga/tree/master/FemtoRV/RTL/PROCESSOR

But while tackling full atomic support (hey, you cheated!), I have difficulties with understanding the RISC-V specification in case an illegal instruction/ECALL/EBREAK happens while the processor is busy servicing an interrupt. I failed to find a nice, clean way for trapping while executing an interrupt handler according to the specification, as there seem to be no CSRs for a two-layer-exception.

So when entering an interrupt or exception, PC is saved in MEPC, current opcode is saved in MTVAL, MCAUSE is updated to the source of the exception, and entry point in MTVEC is put into PC. This is fine if there is either an interrupt or an exception, but I miss a second set of CSRs for a second exception layer.

Could you please give me hints or a pointer on how to resolve this conflict in a specification-conformant way? For a freestyle approach, I would just add more CSRs on my own.

Best wishes for your project,
Matthias

stnolting, 2022-03-05:
Hey Matthias!
Nice to see you again! :)

> We were busy on the Verilog side, too:
https://github.com/BrunoLevy/learn-fpga/tree/master/FemtoRV/RTL/PROCESSOR

I saw that. Great project! :+1:

> But while tackling full atomic support (hey, you cheated!),

Well, somehow... 😅 load-reservate (`lr`) and store-conditional (`sc`) are sufficient to provide atomic access in this single-core setup. For the NEORV32 these instructions are not evaluated by the memory (even though that is supported by the hardware), but by the CPU itself to make sure that lr-sc combinations fail if there is access from another "thread".

Furthermore, the atomic memory operations are hard to implement because they actually execute three operations in one: load from memory, perform some ALU operation, store to memory.

> I have difficulties with understanding the RISC-V specification in case an illegal instruction/ECALL/EBREAK happens while the processor is busy servicing an interrupt.

Short answer: You can't.

Long answer:
Complex RISC-V cores provide additional less-privileged operation modes like "user" or even "supervisor". You can delegate handling of interrupts and exception (using special CSRs like `*ideleg`) to those modes so they are handled - for example - **in** user mode. User mode has it's own set of `epc`, `cause`, `tvec` CSRs to handle them. Thus, you always have machine mode as fall-back for handling traps that occurred in a user-mode trap handler.

> So when entering an interrupt or exception, PC is saved in MEPC, current opcode is saved in MTVAL, MCAUSE is updated to the source of the exception, and entry point in MTVEC is put into PC. This is [...]

Interrupts are automatically disabled when entering the trap handler as the hardware automatically clears `*status.MIE`. Hence, there is no risk of "nested" interrupts. However, synchronous exception like unaligned memory accesses could still occur - it is the task of the programmer to avoid these case as there is no chance to recover from them.

Regarding interrupts: There is a proposal for a `Smrnmi` ISA extension (_resumable non-maskable interrupts_) that adds another set of `cause`, `epc` and `tvec` CSRs to allow resumable handling of (non-maskable) interrupts even _inside_ interrupt-handlers.

Mecrisp, 2022-03-05:
Many thanks for the clarifications and pointers - the resumable NMI proposal describes exactly what I was looking for!

https://lists.riscv.org/g/tech-privileged/message/421

I have already implemented all the AMO opcodes, these might come handy when IO access is used together with heavy interrupt load. This is the current state, if you like to have a peek into a possible atomic opcodes implementation for reference:

http://mecrisp.sourceforge.net/2021_09_05-femtorv32_individua_with_ebreak_and_traps.v

I had to add two additional states and a second ALU multiplexer for reaching the goal, and regular interrupts are already disabled while in NMI in my implementation. Now onwards to add the resumable NMI!

stnolting, 2022-03-10:
> Many thanks for the clarifications and pointers - the resumable NMI proposal describes exactly what I was looking for!

`rnmi` is getting "official" :wink:
-> https://github.com/riscv/riscv-isa-manual/pull/766

> I have already implemented all the AMO opcodes [...] I had to add two additional states and a second ALU multiplexer for reaching the goal

Can you say something about the "hardware complexity" of those AMO operations? When they are handled _inside_ the CPU one would need something like a multi-cycle/multi-operation instruction (load, ALU, store) handling, right? I was thinking about implementing those AMO operations as part of the memory/bus system (so _outside_ of the CPU) like in https://github.com/pulp-platform/axi_riscv_atomics.

Mecrisp, 2022-03-12:
AMO in memory comes handy when DMA access might interfere. But I have not tried; I can only give advice based on my own AMO-in-CPU implementation. Today, I polished the atomics without NMI for release, and I see an increase of 322 LCs in a design on Lattice UP5K when going from RV32IMC to RV32IMAC processor with otherwise similiar functionality and both without RNMI which I postponed after closely examining the specs.

You may compare these files to see what it takes to add AMO support:

https://github.com/BrunoLevy/learn-fpga/blob/master/FemtoRV/RTL/PROCESSOR/femtorv32_gracilis.v
https://github.com/BrunoLevy/learn-fpga/blob/d7a37ec7a5f1166a55f956b4d1637f2008c06b51/FemtoRV/RTL/PROCESSOR/femtorv32_individua.v

I found that the costly parts of the ALU can be reused, I just had to add more multiplexers. For read-modify-write support in the state machine, I reused the normal "lw" infrastructure and added two additional states that writeback the modified value and wait for memory busy signaling in order for this special write to finish.

stnolting, 2022-05-19:
I think this can be closed, right? ;)

closed 2022-05-21 by stnolting