stnolting/neorv32 #325: Instruction 'ecall' becomes 'nop' when external IRQ is active at the same time

GideonZ, 2022-05-21:
**Describe the bug**
When the external IRQ is active at the moment that the CPU needs to execute an 'ecall' instruction, the IRQ is taken, but the ecall trap is not. NeoRV32 simply skips the ecall instruction. It is suspected that this causes FreeRTOS to crash when preemption is off. 

**To Reproduce**
I isolated the problem in FreeRTOS by hypothesis and then wrote a VHDL testcase that runs a test program. The test program is written in Risc-V assembly and executes this sequence 50 times by means of a macro: { ```addi a0, 1```; ```ecall```, ```addi a1, 1``` }. A0 and A1 are initialized to 0 at the start. The trap handler does the usual read of ```mcause``` to differentiate a trap from an external IRQ. When it is a trap, it increments A2; if it is an IRQ it does not. When the program ends, A0, A1 and A2 are written to an I/O register for easy viewing. Without interrupts, all three registers contain 50 at the end of the test run. When IRQ is set to '1' during this sequence, A0 and A1 are still 50 upon completion, but A2 is smaller. The conclusion is that the instruction before ```ecall``` is executed, the instruction after ```ecall``` is executed as well, but the trap itself is not.

**Expected behavior**
It is expected that the CPU executes the ecall instruction after returning from the IRQ handler, or prioritizes the ecall instruction over the interrupt, but NEVER treat it as 'nop'!

**Screenshots**
I can create some VHDL simulation screenshots, but I think the description suffices. I can provide the assembly program.

**Build tool environment:**
Irrelevant

**Hardware:**
 - Implemented CPU extensions: Zicsr for interrupts, instruction cache
 - Peripherals: None
 - Maybe important to note: program runs from external memory (through wishbone + memory model that allows loading binaries upon simulation start.)

stnolting, 2022-05-21:
Oh dear, that sounds like a severe bug in the CPU's exception logic... I will have a look at this. Thank you very much for the bug report! :+1:

GideonZ, 2022-05-21:
test program: [https://github.com/GideonZ/neorv32/blob/main/mysim/main.S](https://github.com/GideonZ/neorv32/blob/main/mysim/main.S)

stnolting, 2022-05-22:
I can confirm this bug. I am using the default test bench to trigger the MEI (machine external interrupt) before making an environment call. I use a variable number of NOPs here plus some delay registers in the MEI signal of the testbench to "align" the trap triggers of MEI and ECALL.

```c
  neorv32_cpu_store_unsigned_word(0xFF000000UL, (1 << CSR_MIE_MEIE)); // set MEI high
//asm volatile ("nop"); // test delay MEI -> ECALL
//asm volatile ("nop");
//asm volatile ("nop");
  asm volatile ("ecall");
```

So far, I have not tested your test program yet.

I know where the problem is, but this is kinda hard to fix. 😅 I managed to rework the CPU's trap controller and so far this bug seems to be fixed. However, the trap controller is a tricky thing as it is also crucial for the on-chip debugger to operate properly. Right now I am doing some tests (also on real hardware). If everything looks fine I will do a PR.

Thanks again for finding this nasty bug! :wink:

GideonZ, 2022-05-22:
Oh... If you have already reworked the trap controller, then I don't need to debug the debugger anymore. You may have seen my pull request. The program runs, but the debugger fails to trap on a breakpoint with that PR. It still halts and downloads the code just fine though. So I thought I'd give it another hour or so to fix it, but if you have fixed it already, I better focus on other things and await your PR.

stnolting, 2022-05-22:
Sorry, I just saw your PR right now. Great work by the way! I have not tested your Pr yet, but will also have a look at that.

I have fixed the exception issue (I think) and the debugger also works again - except for single stepping... But I am working on that :wink:

stnolting, 2022-05-22:
I have opened a PR with my fix in #327. So far everything looks good, but I will run some more tests. Maybe you can test the changes from that PR as well :wink:

closed 2022-05-29 by stnolting