stnolting/neorv32 #328: Booting Linux on NEORV32

mayankmahant, 2022-05-27:
Dear
I would like to know that is there any way boot linux on Neorv32

stnolting, 2022-05-28:
Hey @mayankmahant!

You cannot run desktop operating systems like Linux on the NEORV32, because it does not provide hardware features that are required for hosting those operating system (like a memory management unit, supervisor mode, double floating-point support, ...).

stnolting, 2022-06-04:
I think this can be closed.
Feel free to open a new issue or a new [discussion](https://github.com/stnolting/neorv32/discussions) if you have further question. If you like you can also have a chat at out [gitter channel](https://gitter.im/neorv32/community).

closed 2022-06-04 by stnolting

mayankmahant, 2022-06-13:
Thanks for your feedback sir