stnolting/neorv32 #330: SPI Outputs not initializied with Reset

akaeba, 2022-06-01:
Hi Stephan,

if the Reset is applied to the neorv32 the outputs of the spi are staying uninitilized. The [SPI](https://github.com/stnolting/neorv32/blob/main/rtl/core/neorv32_spi.vhd) uses also only a clock. To avoid unpredictable behaviour needs at least the _CSn_ outputs an reset defined output value. This avoid accidentally activation of the driven SPI component. Better would to assign also CLK and MOSI. In my use case are 180us necessary to assign an valid value.

[image]

[image]

If you like i can submit a patch. What do you think?

Greetings,
Andreas

stnolting, 2022-06-02:
Hey Andreas,

the SPI chip-select lines _are_ initilized, but it takes quite some time since they are configured by a _software reset_ (`crt0` writing zero to all IO device's registers) rater than by an actual hardware reset.

I thought this would not cause any troubles with SPI devices. All outgoing SPI signals might be undefined, but for sure there are NO clock pulses generated on the SPI clock line so any SPI peripheral should not perform any internal operations. We also have the same situation during power-up of the FPGA where all FPGA pins might be in tr-istate mode until the FPGA is configured.

However, I am open for a discussion. Maybe it is about time to add a _real_ hardware reset to all IO/peripheral devices. 🤔

andkae, 2022-06-03:
For the SPI looks also good, now is the Chipselect de-asserted with the reset:

[image]

Thanks for providing the fix!

stnolting, 2022-06-03:
Awesome! The MOSI signal seems to clear correctly now, too

Same question here: you have used the modified version from #334, didn't you?

andkae, 2022-06-03:
yes i used #334

closed 2022-06-03 by stnolting