stnolting/neorv32 #332: neorv32_twi_start_trans does not work with prsc>1

andkae, 2022-06-02:
Hi Stephan,

at the moment i do some tests with the [TWI](https://github.com/stnolting/neorv32/blob/main/rtl/core/neorv32_twi.vhd). My belonging C code looks like follows:

```c
/** Includes **/
/* Standard libs */
#include <string.h>     // memset

/* Processor */
#include <neorv32.h>
#include <neorv32_twi.h>

/**
 *  main
 */
int main()
{
    /** Variables **/

    /* init UART */
    neorv32_uart0_setup(115200, PARITY_NONE, FLOW_CONTROL_NONE);

    /* entry message */
    neorv32_uart0_print("Interfaces I2C Core of NEORV32\n");

    /* setup I2C */
    neorv32_twi_disable();
        // neorv32_twi_setup(uint8_t prsc)
    neorv32_twi_setup(4);
    neorv32_twi_mack_enable();
    neorv32_twi_enable();

    /* write to I2C */
    neorv32_twi_start_trans(0x55);
    neorv32_twi_generate_stop();

    while ( 1 ) { } // inifinite loop
    return 0;
}
```

For asserting the transfer i use the NEORV32 function neorv32_twi_start_trans. For small prescaler values like 1 works this very well.

PRSC=1:
[image]

For PRSC=4 is only the starbit sended and then happends nothing
[image]

I did some investigation, the issue seems like follows, if a higher PRSC value is used, then is the FSM in the startbit state _arbiter=101_ and the write of ths address - in my case 0x55 - is skipped:
[image]

One possible solution could be to poll the busy flag, this would lead to an change in neorv32_twi_start_trans() and the IRQ driven I2C would not work anymore.

What are your thoughts?

BR,
Andreas

stnolting, 2022-06-02:
Hey Andreas,

you need to generate a start conditions first before you can do actual TWI transfers. So you need to call `neorv32_twi_generate_start` first:

```c
    /* write to I2C */
    neorv32_twi_generate_start(); // generate START conditions before triggering actual transfers
    neorv32_twi_start_trans(0x55);
    neorv32_twi_generate_stop();
```

I just had a quick look at the [TWI documentation](https://stnolting.github.io/neorv32/#_two_wire_serial_interface_controller_twi) - and it seems like the documentation is not really clear here. I need to fix that! :wink:

stnolting, 2022-06-02:
I just realized I messed up `neorv32_twi_start_trans()` and `neorv32_twi_trans()` - my fault! 😅
I will have a look at this!

stnolting, 2022-06-02:
Your fix is great workaround for now, but I think a dedicated hardware reset for all IO/peripheral devices would be a better permanent solution. Otherwise we might get trouble with this again in other places. I think I'll start working on that reset issue on another branch then you can have a look, too, and we can make sure to eliminate all those hardware-vs-simulation issue.

andkae, 2022-06-03:
Hi Stephan, it works! Thanks for the really fast providing the fix

```c
    /* write to I2C */
    neorv32_twi_start_trans(0x55);
    neorv32_twi_generate_stop();
```

the belonging waveform with global reset
[image]

BR,
Andreas

stnolting, 2022-06-03:
Great to hear! You have used the modified version from #334, right?

andkae, 2022-06-03:
yes #334 was used

closed 2022-06-03 by stnolting