stnolting/neorv32 #335: SPI in Highspeed does not Honor Spi Mode 0

andkae, 2022-06-03:
Hi Stephan,

at the moment i do some tests with the [SPI](https://github.com/stnolting/neorv32/blob/main/rtl/core/neorv32_spi.vhd) If i use following code:

```c
/* Processor */
#include <neorv32.h>
#include <neorv32_spi.h>

/**
 *  main
 */
int main()
{
    /** Variables **/

    /* init UART */
    neorv32_uart0_setup(115200, PARITY_NONE, FLOW_CONTROL_NONE);

    /* entry message */
    neorv32_uart0_print("Interfaces SPI Core of NEORV32\n");

    /* setup SPI */
    neorv32_spi_disable();
        // neorv32_spi_setup(uint8_t prsc, uint8_t clk_phase, uint8_t clk_polarity, uint8_t data_size)
    neorv32_spi_setup(0, 0, 0, 0);  // spi mode 0, 8bit
    neorv32_spi_highspeed_enable(); // running at half of the processor clock
    neorv32_spi_enable();

    /* perform SPI write */
    neorv32_spi_cs_en(0);   // lines are LOW when activated.
    neorv32_spi_trans(0xaa);
    neorv32_spi_cs_dis(0);  // lines are HIGH when deactivated.

    /* stop programm counter */
    while ( 1 ) { } // inifinite loop
    return 0;
}
```
According:
https://upload.wikimedia.org/wikipedia/commons/thumb/6/6b/SPI_timing_diagram2.svg/495px-SPI_timing_diagram2.svg.png
Is in Spi Mode 0 on the first clock edge the data value latched.

In Wavefrom the data is on the first edge changed, which could lead to a fail on the SPI:
[image]

What is your opinion?

BR,
Andreas

andkae, 2022-06-03:
If i  not enable the High speed mode, then is at least the MOSI updates one clock cycle to early

```c
// neorv32_spi_highspeed_enable()
```
[image]

One solution could be perhaps to delay the enable one cycle for the bit shift.

Another question, if i understand the spi mode 0 right then is on the first clock edge the MISO captured and on the second clock edge the MOSI updated. In this implementation:

https://github.com/stnolting/neorv32/blob/e7b9c0554e0d43f9e96a7c50861534d5913fbdf3/rtl/core/neorv32_spi.vhd#L258

leads every latch to an update, that means the SPI mode 0 can't realized. Do i see it right?

Thanks for you help

andkae, 2022-06-03:
PRSC=1

same one clock cycle to early:
[image]

stnolting, 2022-06-03:
You are right, there is a phase offset problem between clock and data_in/data_out. I guess we have the same problem in the XIP's SPI interface...

I will have a closer look at this. I'm feeling like this can be fixed easily, but we'll see. I'll open a PR as soon as I have a fix.

stnolting, 2022-06-04:
With the fix from #335 the SPI transactions now look much better. I have tested all four SPI modes and also the high-speed mode. Clock and data now seem to be aligned correctly.

This is the transmission from your example (clock mode 0, high-speed mode enabled, transmitting `0xaa`):

[image]

closed 2022-06-04 by stnolting

andkae, 2022-06-05:
Hi Stephan, i can also confirm that in the highspeed mode it now works corectly:

[image]

Thanks for providing the fix!

BR,
Andreas

stnolting, 2022-06-05:
> i can also confirm that the highspeed mode now works corectly:

🎉
Thanks for the feedback!