stnolting/neorv32 #339: TWI Accidenttally bit left shift after byte transmission

andkae, 2022-06-08:
Hi Stephan,

at the moment i try to read some bytes from an I2C slave. Thereofre i use an properitary I2C macro and the [TWI](https://github.com/stnolting/neorv32/blob/main/rtl/core/neorv32_twi.vhd). My belonging testcode looks following:

```c
/* Processor */
#include <neorv32.h>
#include <neorv32_twi.h>

/**
 *  main
 */
int main()
{
    /** Variables **/
    const uint8_t   uint8I2cAdr7 = 15;
    uint8_t         uint8I2cRdBuf[10];

    /* init UART */
    neorv32_uart0_setup(115200, PARITY_NONE, FLOW_CONTROL_NONE);

    /* entry message */
    neorv32_uart0_print("I2C\n");

    /* setup I2C */
    neorv32_twi_disable();
        // neorv32_twi_setup(uint8_t prsc)
    neorv32_twi_setup(3);
    neorv32_twi_mack_enable();
    neorv32_twi_enable();

    /* Read from CHC Dummy */
    neorv32_twi_start_trans((uint8I2cAdr7 << 1) | 0);   // write access
    neorv32_twi_trans(0x23);                            // read pointer
    neorv32_twi_start_trans((uint8I2cAdr7 << 1) | 1);   // read access
    for ( uint8_t i = 0; i < (sizeof(uint8I2cRdBuf)/sizeof(uint8I2cRdBuf[0])); i++ ) {
        neorv32_twi_trans(0xFF);    //  dominant bits '0', set by slave
        uint8I2cRdBuf[i] = neorv32_twi_get_data();
    }
    neorv32_twi_generate_stop();

    /* print to console */
    for ( uint8_t i = 0; i < (sizeof(uint8I2cRdBuf)/sizeof(uint8I2cRdBuf[0])); i++ ) {
        neorv32_uart0_printf("%x;", uint8I2cRdBuf[i]);
    }

    /* stop programm counter */
    while ( 1 ) { }
    return 0;
}
```

In my example has the first read byte the value 0xF9 _I2C_DATA_.
[image]

When the processor reads the value is the SFR shifted one possition and i recieve 0xF2:
[image]

In my opinion is this caused by:
https://github.com/stnolting/neorv32/blob/af12cde967ed2facb1c5a43fd2fc09b700381980/rtl/core/neorv32_twi.vhd#L259

Whats your opinion?

Thanks for the support.

BR,
Andreas

stnolting, 2022-06-08:
Hey Andreas!

I will have a closer look a this. I have tested several IC2 devices on _real hardware_ using the the interactive [demo_twi](https://github.com/stnolting/neorv32/tree/main/sw/example/demo_twi) program and I thought the module was doing fine 😅

---------------------

Some questions regarding your setup:

* What kind of IC2 device are you trying to access?

* This will activate the host-generated ACK bit. Is this really required (all the time) by your device?
```c
neorv32_twi_mack_enable();
```

> In my opinion is this caused by:

Yes and no. `rtx_sreg` is a 9-bit shift register. The upper most 8 bits will contain the received data while the LSB contains the
ACK sampled from the device. When reading the sampled data byte, the module only outputs the 8 most significant bits:
https://github.com/stnolting/neorv32/blob/af12cde967ed2facb1c5a43fd2fc09b700381980/rtl/core/neorv32_twi.vhd#L154

andkae, 2022-06-08:
Hi Stephan,

> This will activate the host-generated ACK bit. Is this really required (all the time) by your device?

If i disable this, same result
```c
//neorv32_twi_mack_enable();
```

andkae, 2022-06-08:
In addition the twi_clk_phase:

[image]

> I will have a closer look a this.

Thanks very well :)

andkae, 2022-06-08:
Okay, got it. The Problem is that from the Inventra macro the set I2C speed needs to match with the I2C speed of the TWI:

[image]

In my case with 60MHz clock is only 240kHz or 1.5MHz possible. This mismatches with the defined I2C modes:
* 100kHz
* 400kHz
* 1Mhz

The shorting of the clock phase causes this issue:
[image]

stnolting, 2022-06-08:
> In addition the twi_clk_phase:

This signal does not look good... Seems like there is a problem with the clock stretching? Can you check the `twi_clk_halt` signal? Maybe your TWI device is trying to slow down the transmission.

Anyway, maybe there is a synchronization problem when the module initiates a new operation (START, STOP, transmission). I'll check that.

stnolting, 2022-06-08:
> Okay, got it. The Problem is that from the Inventra macro the set I2C speed needs to match with the I2C speed of the TWI:

Oh okay, so it's just a configuration problem?!

andkae, 2022-06-08:
> twi_clk_halt

[image]

andkae, 2022-06-08:
> Oh okay, so it's just a configuration problem?!

There are two views:
1) The I2C Macro is configured for 400kHz
2) The TWI runs at 60MHz only at 240kHz or 1.5MHz

The effect comes of the mismatch of both set-upped frequencies

andkae, 2022-06-08:
What do you think are the changes to add a fine graned post divider into the TWI to match better the I2C standard frequencies?

stnolting, 2022-06-08:
> What do you think are the changes to add a fine graned post divider into the TWI to match better the I2C standard frequencies?

Sure, this would be possible by modifying the TWI's clock generator. Right now, we have the standard clock prescalers available:

```c
enum NEORV32_CLOCK_PRSC_enum {
  CLK_PRSC_2    = 0, /**< CPU_CLK (from clk_i top signal) / 2 */
  CLK_PRSC_4    = 1, /**< CPU_CLK (from clk_i top signal) / 4 */
  CLK_PRSC_8    = 2, /**< CPU_CLK (from clk_i top signal) / 8 */
  CLK_PRSC_64   = 3, /**< CPU_CLK (from clk_i top signal) / 64 */
  CLK_PRSC_128  = 4, /**< CPU_CLK (from clk_i top signal) / 128 */
  CLK_PRSC_1024 = 5, /**< CPU_CLK (from clk_i top signal) / 1024 */
  CLK_PRSC_2048 = 6, /**< CPU_CLK (from clk_i top signal) / 2048 */
  CLK_PRSC_4096 = 7  /**< CPU_CLK (from clk_i top signal) / 4096 */
};
```
Internally, the TWI module adds another `div4` scaling factor to this.

How much more "precision" / fine-grained divider options do you thing might be suitable?

stnolting, 2022-06-08:
> Only a combinatoric spike, no problem.

Right, but unpretty nevertheless :wink:

>  In the mean time i think we can close this issue. Sorry for the "wrong alarm" The topic with the divider we can add as disccusion.

No problem! I found a problem with synchronization due to this issue! Fortunately, this has not occurred so far.

andkae, 2022-06-08:
> I do not know your TWI device module, but when just looking at the I2C specs there should be no problem if the master is running on a slower clock than the actual device..?!

Yes, my TWI shorts the clock high phase, and this leads to a double shift in the neorv32_twi.

[image]

andkae, 2022-06-08:
> No problem! I found a problem with synchronization due to this issue! Fortunately, this has not occurred so far.
Oh nice :)

stnolting, 2022-06-08:
> Yes, my TWI shorts the clock high phase, and this leads to a double shift in the neorv32_twi.

Hmm, strange... This should halt the whole sampling logic...

andkae, 2022-06-09:
I checked with #340, now is in the read byte case the ACK of the byte read by the master missing:

[image]

stnolting, 2022-06-09:
What do you mean by "missing"?

The TWI module samples the ACK/NACK in the 9th clock cycle. In your waveform the accessed device does not pull SDA low in that cycle so this is interpreted as NACK.

andkae, 2022-06-09:
> What do you mean by "missing"?

In the Waveform, I do an byte read from an Slave (Direction: Slave -> Master) There i had expected that the Master confirms the Byte Receive with an ACK. Anyways, i need to check the I2C spec.

stnolting, 2022-06-09:
The NEORV32 TWI module will never do the ACK (actively setting SDA low in the 9th clock cycle) unless explicitly enabled ("master-ACK) by

```c
neorv32_twi_mack_enable();
```

stnolting, 2022-06-09:
This gives a nice overview of ACK/NACK: https://www.ti.com/lit/pdf/slva704

stnolting, 2022-06-10:
I think I'm done with reworking the TWI module. 🤔 The bus interaction now looks way better and it even ensures the bus staying claimed until a STOP condition is generated.

closed 2022-06-10 by stnolting