stnolting/neorv32 #346: Cannot Debug with Segger Studio

AbdulrahmanKH7, 2022-06-12:
Hello!

I used the same FTDI board you used, and this tutorial:
https://www.emb4fun.com/riscv/ses4rv/index.html

and downloaded the Segger J-Link software

The error displayed is:
Failed to open DLL

Please help, it would be much easier to upload and debug the code this way.
Many thanks
Abdulrahman

stnolting, 2022-06-12:
Hey there!

I am no expert here, but I am not sure if Segger Studio actually supports other interface adapters then their J-Links. Maybe @emb4fun (the author of the tutorial you have linked) knows more about that :wink:

So you could get a J-Link (like the cheaper EDU version) or you can use OpenOcd + GDB.

AbdulrahmanKH7, 2022-06-12:
Alright, I am not sure if I should wait for his response here or if there is an issues forum on his page because I could not find an issues forum here:

https://github.com/emb4fun/neorv32-examples

Thanks!

stnolting, 2022-06-12:
Maybe he'll see that I've linked him.

> https://github.com/emb4fun/neorv32-examples

You could also try opening an issue over there

Anyway, I am pretty sure you are going to need a J-Link adapter when you want to use Embedded Studio. OpenOCD, gcc/GDB and eclipse could be an open-source alternative.

emb4fun, 2022-06-13:
Yes, you really need a J-Link if you want to use Embedded Studio.

AbdulrahmanKH7, 2022-06-13:
So the FT2232H-56Q Mini Module won't work in this case?? @emb4fun
Thanks

stnolting, 2022-06-28:
Seems like there is no other option than using a J-Link probe.

stnolting, 2022-07-01:
I think this can be closed. Feel free to open another issue if you have further questions.

closed 2022-07-01 by stnolting