stnolting/neorv32 #350: Processor wrapper for LiteX integration

stnolting, 2022-06-17:
Finally (😅) the CPU core seems to be in a tidied-up and "optimized" state so I would like to come back to the LiteX framework. @enjoy-digital already did some awesome work and provided a first CPU-only NEORV32 integration (-> #115).

As discussed in #115 I would like to extend this initial setup. Instead of just integration the NEORV32 CPU core I would like to provide a "core complex" for the integration into LiteX. This core complex includes the CPU (obviously) and also some "important" processor modules. I am not talking about peripheral modules like UART and SPI - these can be implemented using the LiteX system. I am thinking about a core complex that implement CPU-close modules like the instruction cache, the RISC-V machine system timer and the on-chip debugger. For that I am proposing a new system wrapper in [`rtl/processor_templates`](https://github.com/stnolting/neorv32/tree/main/rtl/processor_templates) (maybe `neorv32_litex.vhd`) for implementing the LiteX NEORV32 "core complex".

However, there are a lot of issues I would like to discuss first.

## Interface

The initial LiteX port by @enjoy-digital uses a wrapper for converting the CPU's native interface into a Wishbone-ish interface. The NEORV32 already provides a Wishbone bridge, which could be used as interface to the LiteX framework.

This is a _single_ interface that is used for fetching instructions and for reading/writing data to a _unified_ address space. Bus traffic can be reduced by using the NEORV32 instruction cache.

Questions:
* Which Wishbone protocol? "Pipelined" or "classic"? In case of the NEORV32 the only difference here is the handling of the STB signal (see figure six in [Data Sheet - Processor External Memory interface](https://stnolting.github.io/neorv32/#_processor_external_memory_interface_wishbone_axi4_lite)).
* What about _maximal_ access latency? Is there an upper bound (let's say 2000 clock cycles) after that a pending bus access will be acknowledged (for sure!)?

## Configuration

According to https://github.com/enjoy-digital/litex/wiki/CPUs#soft-cpu-variants LiteX provides several abstract configuration options like "lite", "standard" and "full". To simplify the configuration of the NEORV32 core complex I propose to use as few generics as possible. One could be a simple integer to pass "lite, standard, ..." to the complex. Based on this integer the complex defines all the NEORV32 generic (like cache configuration, ISA extensions, CPU-close modules like the timer, etc.).

The question is how to map those configuration options. "Full" is pretty obvious, but what about "standard" and "lite"?

## Address Space

The CPU-close modules of the core complex (like the machine timer) are implemented as memory-mapped devices. All these memory occupy an address range starting at `0xFFFF0000`. This 64kB-wide address space would NOT be available for the LiteX system. Question here: is this a problem? And what about the boot/reset address of the CPU? Is this flexible or can be hardwire this to some fixed value?

Btw, a user could still "fine tune" all the available configuration options by modifying the core complex VHDL file.

## More to come...

I just realized this might be way to much information for a single issue... 😅 Maybe we can "outsource" some things into the [discussions](https://github.com/stnolting/neorv32/discussions) section.

enjoy-digital, 2022-06-20:
Hi @stnolting,

great to see you are interested to improve the initial work. Here are some first answers:

> ## Interface

LiteX supports both Wishbone "Pipelined" and "Classic" and from the description of your implementation, the implementation seems similar in LiteX. Some peripherals are able to pipeline accesses (ex SRAM), the others will just hand the accesses in a minimum 2 cycles. The SoC implements a timeout that will ack the access for the CPU if no peripherals responds. This timeout if configurable and defaults to 1e6 cycles.

Note: If this can be useful for NeoRV32: LiteX also supports AXI-Full/AXI-Lite buses and can automatically do the bridge between standards/data-width.

> ## Configuration

We used `lite`/`standard`/`full`  on some CPUs, but this is not mandatory and we have used other names some recent CPUs (ex FemtoRV: https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/femtorv/core.py#L16-L24). We can now also expose/pass command line parameters from/to the CPU (ex VexRiscv-SMP: https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/vexriscv_smp/core.py#L57-L77), so there are different possibilities/way to do the configuration.

We could just used the variants that make sense for you (and with names that you could also define).

> ## Address Space

This is flexible: The CPU wrapper can enforce part of the address mapping to force the SoC to use it. For the reset address, this is also flexible: LiteX can configure it on the CPU or use a fixed value provided by the CPU.

> ## VHDL/Verilog:

This was not a question, but just for info, we are also trying to simplify VHDL -> Verilog conversion in LiteX and do it automatically when the toolchain does not support VHDL (ex for simulation with Verilator or with open-source toolchains). This should benefits NeoRV32 since LiteX should be able in the future to transparently convert the VHDL to Verilog without having to implement this manually in the CPU wrappers as has been done here: https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/neorv32/core.py#L178-L192. This should also simplify creation of variants based on different VHDL files.

stnolting, 2022-06-20:
Thank you so much for the detailed answers!

> ## Interface

I think we should start with simple Wishbone - maybe in the "pipelined" version as I am planning to provide _real_ pipelined/burst transfers in the future... AXI also sounds great, but I think for now this would be unnecessarily complex and might also increase latency (my AXI bridge is not optimized and just supports the minimal AXI-Lite protocol).

A timeout of 1e6 cycles is quite a lot 😅 I think it would be good to enabled the processor's "bus timeout feature" (with a timeout value less than 1e6) to raise an exception if there is no response. This is just me, but this feature helped me a lot when debugging more complex system setups with lots of memory-mapped peripherals.

> ## Configuration

It is great to have so much flexibility here - but at the same time this makes it more complicated to think about all the different flavors :wink: At first we should agree on a maximal set of features for the "core complex" and then we can define 2 or more less-complex versions.

> ## Address Space

This is great, I was not aware of this. Having a fixed boot address of all-zero would be the simplest solution I think.

> ## VHDL/Verilog

Yeah, I have seen that - really great work! :+1: I think a lot of VHDL-based setups can benefit from that.

## Setup Questions

I am still thinking about how to provide the NEORV32 core complex in terms of _this_ repository. Having a single top VHDL file would be fine, right? Do we need to have some additional "auxiliary" files there (e.g. for telling LiteX about boot address, bus timeouts, etc.) or will this be in [`enjoy-digital/litex/tree/master/litex/soc/cores/cpu/neorv32`](https://github.com/enjoy-digital/litex/tree/master/litex/soc/cores/cpu/neorv32)?

Btw, I just had a discussion with some colleagues about LiteX and we were arguing how to pronounce it correctly... 😅 Is it "Lite-X" or "Li-Tex"?

stnolting, 2022-06-22:
> LiteX is pronounced Lite-X :)

Thanks for solving this mystery 😅

I have started writing a core wrapper providing four configurations: `minimal`, `lite`, `standard` and `full`. The details of each config can be discussed when I have a PR available. However, I am wondering what the most convenient passing of the configuration might be: a simple integer that selects 0..3? Or would you prefer an actual string to be passed to the wrapper?

What about configuration options for clock speed (which is not actually required here) and for the on-chip debugger? A simple integer for the clock and a boolean parameter for the OCD?

I have seen some cores use something like `standard+debug` - this can be resolved into unique VHDL generics, right?

Another question: I would like to keep things simple at first. However I would like to add a single interrupt input (which is the RISC-V "machine external interrupt"). I have seen other cores do also provide a single interrupt. Does it trigger on rising/falling edge or on low/high level?

enjoy-digital, 2022-06-27:
Hi @stnolting,

sorry for the delay, good progress! For the parameters, we have lots of flexibility in the wrappers with the use of Python, so the parameter format is really up to you and what is the easiest for you. I'm fine with your suggestions. We basically just need to know how to configure the core and how to do the variant -> configuration mapping in the wrapper.

For the debug, we only have this on VexRiscv and this could also specific configuration to do on the instance.

For the interrupt, the other CPUs that don't have a specific interrupt handler are using high level on the interrupt signal to indicate an interrupt.

enjoy-digital, 2022-06-29:
The minimal and lite variants seems to be working fine with: https://github.com/enjoy-digital/litex/commit/1f1d9f5c0e0b89a6cf60cc1c5be555e0bd49ecc6 :):

`litex_sim --cpu-type=neorv32`

````
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Jun 29 2022 15:49:14
 BIOS CRC passed (68a3867e)

 LiteX git sha1: 1f1d9f5c

--=============== SoC ==================--
CPU:		NEORV32 @ 1MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128KiB
SRAM:		8KiB

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
No boot medium found

--============= Console ================--

litex> 
````

Standard and Full variants do not seem to be working, but I haven't investigated yet. This could be related to the icache and should not be complicated to sort out together.

Unfortunately, GHDL/Yosys flow does not seem to be able to keep parametrization on Verilog converted from VHDL, so LiteX will have to directly modify the default values of the CONFIG/DEBUG generics in neorv32_litex_core_complex to be able to configure the core complex.

stnolting, 2022-06-30:
Wow, that was fast! Great to see the wrapper is working! :)

> Standard and Full variants do not seem to be working, but I haven't investigated yet. This could be related to the icache and should not be complicated to sort out together.

I have tested the `standard` and `full` variants - so at least the core itself seems to be fine. However, I do not think this problem is caused by the bus accesses since the cache also uses simple Wishbone transfers (no bursts). I think there might be a synchronization problem. I do not claim to understand the complete boot process of LiteX but maybe the instruction cache contains outdated data at some point.

I saw the prototypes for cache flush functions in `litex/soc/cores/cpu/neorv32/system.h`. Maybe we need to add the instruction cache flush mechanism here:

```c
static inline void flush_cpu_icache(void){ __asm__ volatile ("fence.i"); };
static inline void flush_cpu_dcache(void){ __asm__ volatile ("fence");   }; /* No data cache yet */
```

Btw, the data cache flush function is just a placeholder yet.

> Unfortunately, GHDL/Yosys flow does not seem to be able to keep parametrization on Verilog converted from VHDL, so LiteX will have to directly modify the default values of the CONFIG/DEBUG generics in neorv32_litex_core_complex to be able to configure the core complex.

Too bad but I know these _flaws_... How can we circumvent this? Several "instances" for each config combination? Or maybe some different parametrization style to "trick" yosys?

-----------

In a next step it might be nice to add toolchain/gcc support for the configurations above "minimal":

`litex/soc/cores/cpu/neorv32/core.py`:
```python
    "minimal":          "-march=rv32i     -mabi=ilp32",
    "lite":             "-march=rv32imc   -mabi=ilp32",
    "standard":         "-march=rv32imc   -mabi=ilp32",
    "full":             "-march=rv32imc   -mabi=ilp32",
```

But I can also do that via a PR later on 😅

enjoy-digital, 2022-07-05:
Hi @stnolting,

thanks for the info. I'll probably first look at allowing the variants selection and we could then try to understand together the issue when enabling the cache. For the parametrization, I'll probably cheat a little bit: make the LiteX build modify the VHDL wrapper's initial values before converting it to verilog. This is not very clean but will avoid creating another wrapper around the...wrapper.

enjoy-digital, 2022-07-05:
I imagine that parameters are first fixed/resolved before doing the VHDL -> Verilog conversion because propagating parameters through conversion would probably be really challenging. So I haven't' really looked at the code but can guess it's the reason parameters with non-default values are not allowed at the top level.

Using constants would indeed make things easier for replacement with LiteX before the Verilog conversion, if you can do that in the wrapper, I'll implement the replacement.

stnolting, 2022-07-15:
I have removed the generics and now two constants are used to select the configuration:
```vhdl
-- configuration select ----------------------------------------------------------------------------
constant CONFIG : natural := 0;     -- configuration select (0=minimal, 1=lite, 2=standard, 3=full)
constant DEBUG  : boolean := false; -- enable on-chip debugger, valid for all configurations
-- -------------------------------------------------------------------------------------------------
```

I have also found (and fixed!) a bug in the Wishbone bus interface (https://github.com/stnolting/neorv32/pull/367). I'll need to do some more testing but I think the problem with the configurations using the i-cache should be resolved :wink:

stnolting, 2022-07-18:
Yeay! Awesome! 👍😄

I'll do a PR [enjoy-digital/litex](https://github.com/enjoy-digital/litex) to add cache functions and interrupt management and I will link the LiteX setup in the NEORV32 documentation.

I think we can close this now. Thanks again for all your work and help!

enjoy-digital, 2022-07-19:
Great that you've been able to reproduce the results! More that happy to collaborate on this and eventually go further if needed.

stnolting, 2022-07-19:
Btw, I have never worked with GitHub wikis so far, but can I add NEORV32 to https://github.com/enjoy-digital/litex/wiki/CPUs?

enjoy-digital, 2022-07-19:
Yes, you should be able to edit the wiki directly, please let me know if you have troubles.

stnolting, 2022-07-19:
Works without problems: https://github.com/enjoy-digital/litex/wiki/CPUs#risc-v---neorv32 :+1:

closed 2022-07-19 by stnolting

kalyze, 2023-09-13:
Hi @stnolting,

We are currently using LiteX to implement a SoC, and we'd like to use last version of your softcore implementation. 
It seems that the current version of the litex_core wrapper does not include the latest core upgrades. Do you plan to update the litex wrapper with the current core version?

Thanks!

stnolting, 2023-09-13:
Hey @kalyze!

Seems like I have not updated those system integration wrappers in a while... 🙈
Which specific configuration option(s) are you missing?

kalyze, 2023-09-13:
By now we have the core running but we have some problems in getting the jtag debugger working.
After debugger connection we can halt and resume program execution. When the execution stops, gdb tells that the core stopped on address 0x00000000 but our program starts on address 0x40000000. Breakpoints are also ignored.

I already modified neorv32/core.py in litex library to enable debugger peripheral and assign jtag signals to some gpios.
I tryed to update the core to a newer version, hoping to fix this problem but ghdl failed in generating verilog. I saw your integration is considerably changed since the last litex system integration. 

Have your ever had similar problem with the debugger module before? Do you now how can I fix them?

Thanks!

stnolting, 2023-09-15:
Hey @kalyze!

The debugger itself works quite flawlessly so far :wink: However, I have never tested it inside a LiteX-setup.

What configuration of the NEORV32 core complex are you using? Have you tested a setup without caches?

kalyze, 2023-09-21:
Hi @stnolting,

by now I've only tested the core configuration you give in the LiteX wrapper. I only modified it to connect jtag signals.
After some tests, I'm still in trouble on getting the breakpoints working. I'm also not able execute program step-by-step.

I haven't tested a setup without caches yet. How can I test it?

stnolting, 2023-09-21:
To disable the caches for all configurations set all entries in these two lines to `false`:
https://github.com/stnolting/neorv32/blob/c7590c9cc86b1646926f57d0ab0ab610e0471af8/rtl/system_integration/neorv32_litex_core_complex.vhd#L153
https://github.com/stnolting/neorv32/blob/c7590c9cc86b1646926f57d0ab0ab610e0471af8/rtl/system_integration/neorv32_litex_core_complex.vhd#L157

Which version of NEORV32 are you using?

Can you post the openOCD messages when connecting to the core and when manually halting/resuming the CPU?

kalyze, 2023-09-25:
Hi @stnolting,
we finally got a working NEORV32-LiteX configuration with the debugger! 

We updated the NEORV32 core to version 1.8.9, fixed memory map in _core.py_ and now we have the debugger working correctly.

If you are interested, here you can find our modifications:
https://github.com/protech-engineering/litex/blob/protech/litex/soc/cores/cpu/neorv32/core.py

If you don't find anything wrong, we can open a pull request to update official LiteX repository.

Thanks for the support!

stnolting, 2023-09-25:
> we finally got a working NEORV32-LiteX configuration with the debugger!

Great to hear! 👍

> We updated the NEORV32 core to version 1.8.9, fixed memory map in core.py and now we have the debugger working correctly.

What exactly did you fix? I think the VHDL file list was outdated, but what about this memory map?

> If you don't find anything wrong, we can open a pull request to update official LiteX repository.

That would be awesome! 😃 

Thank you very much for updating/fixing!

enjoy-digital, 2023-09-26:
Hi @kalyze,

thanks for fixing the LiteX integration! Happy to review/test/merge a PR in LiteX if you submit it.

kalyze, 2023-09-26:
> What exactly did you fix? I think the VHDL file list was outdated, but what about this memory map?

Hi @stnolting,
in order to make the core working, we had to move _io_regions_ in LiteX _"core.py"_ to a non cached memory region. Then we changed the _"memory_map"_ property to map _csr_ to the same region, to avoid LiteX build errors.

stnolting, 2023-09-26:
Thanks for updating! 👍