stnolting/neorv32 #355: External MIRQ some times missed

andkae, 2022-06-27:
Hello Stephan,

i use an NEORV32 config with an external IRQ input of MIRQ

```vhdl
IO_MTIME_EN => false, -- implement machine system timer
mtime_irq_i     => std_ulogic(mtime_irq),  -- machine timer interrupt, available if IO_MTIME_EN = false
```

The belonging c code looks like follows:
```c
/* Processor */
#include <neorv32.h>

void mtime_irq_handler(void) 
{
	neorv32_uart0_putc('.'); // send tick symbol via UART
}

int main()
{
    /* init UART */
    neorv32_uart0_setup(115200, PARITY_NONE, FLOW_CONTROL_NONE);
	
	/* enable error capture */
	neorv32_rte_setup();
	
	/* install MTIME interrupt handler to RTE */
	neorv32_rte_exception_install(RTE_TRAP_MTI, mtime_irq_handler);
	
	/* enable MTIME IRQ */
	neorv32_cpu_irq_enable(CSR_MIE_MTIE); 		// enable MTIME interrupt
	
	/* enable global interrupt */
	neorv32_cpu_eint();	
	
	/* end */
	while (1) {
		__asm("nop");
	}
}
```

The MIRQ is only asserted one clock cycle. The Waveform looks like follows:
[image]
There are three cursors, on the two first leads the IRQ to no UART interaction and the third starts an UART transfer. My expections were three UART transfers. The MIRQ is missing wenn der Signal _mtime_irq_ toggles but there is no action on the UART.

[image]

Thanks for your support!

BR,
Andreas

stnolting, 2022-06-27:
Hey Andreas!

Seems like there is a "bug" in the documentation. Section [[2.5.1. RISC-V Standard Interrupts](https://stnolting.github.io/neorv32/#_risc_v_standard_interrupts) just say something about the FIRQs, but there should be:

> The RISC-V standard interrupts are **level-triggered and high-active**. Once set the signal has to stay high until the interrupt request is explicitly acknowledged (e.g. writing to a memory-mapped register). The RISC-V standard interrupts can **NOT** be acknowledged by writing zero to the according `mip` CSR bit.

I'll fix that.

So, your external MTI source has to keep the interrupt signal high until the program explicitly acknowledges the request.

andkae, 2022-06-27:
> I'll fix that.

Thanks for your help :)

andkae, 2022-06-28:
Yes :)

commit 0dd8a737 (2022-06-27): [docs] "mip" CSR: minor fixes

#355
closed 2022-06-28 by stnolting