stnolting/neorv32 #358: Make clean does not honor custom includes

andkae, 2022-07-05:
Hi Stephan,

when i write a custom makefile which uses the statements:

```bash
APP_SRC ?= $(wildcard ./*.c) $(wildcard ./*.s) $(wildcard ./*.cpp) $(wildcard ./*.S) $(wildcard ./include/myC/*.c)
```
then does the `make clean` not honor this paths. See the Main makefile for reference:

https://github.com/stnolting/neorv32/blob/f321f5d7ba0e772667b9632b66904ac3a19196b9/sw/common/common.mk#L368

How we can deal with that?

Thanks for your help.

BR,
Andreas

stnolting, 2022-07-05:
Hey Andreas!

Do _include_ the NEORV32 main makefile in your makefile? Then `make clean_all` should also remove all build artifacts.

For example have a look at [`sw/example/demo_freeRTOS/makefile`](https://github.com/stnolting/neorv32/blob/main/sw/example/demo_freeRTOS/makefile):

https://github.com/stnolting/neorv32/blob/f321f5d7ba0e772667b9632b66904ac3a19196b9/sw/example/demo_freeRTOS/makefile#L140

stnolting, 2022-07-05:
Yeah, maybe the description is misleading...

Maybe we should change the help text from
```
clean        - clean up project"
clean_all    - clean up project, core libraries and image generator"
```

to
```
clean        - clean up project home folder"
clean_all    - clean up whole project, core libraries and image generator"
```
? 🤔 

Any ideas?