stnolting/neorv32 #376: [idea] extend SLINK interrupts

stnolting, 2022-07-28:
This idea was started by @betocool-prog in the NEORV32 [gitter channel](https://gitter.im/neorv32/community):

> If it's size configurable and channel configurable, yes. But we'd need to add an additional status register, maybe, to read out which interrupt it was. I'm thinking:
>    At least one data word => transition from empty to non empty
>    Half or more full => from (size/2 - 1) to size/2
>    Full => No space available... equivalent to overflow?
>    Empty => transition from at least one to 0 elements.
>  My strategy as C programmer would be to read Size/2 samples every time I get a half-full interrupt, or if not using interrupt, reading the half-full status bit(s).

The SLINK module provides **just** two interrupt lines: one for signaling TX-related events and another one for signaling RX-related events. I agree that a more fine-grained interrupt cause configuration is a good idea (especially the half-full situation).

Do weed need to define those interrupt conditions individually (per-channel) or can we have a global configuration (one for all channels) just separated for RX and TX?

betocool-prog, 2022-07-28:
It's possible to set one interrupt for the RX line and one interrupt for the TX line for the CPU and still get the full feature of events by OR'ing the flags internally inside the SLINK module.

The RX/TX interrupt signals only once, whenever any of the above conditions happen. Each condition is a flag on a interrupt status register, and as part of the ISR, the programmer should act accordingly. In some weird cases, it's possible that all interrupt flags are on, for example, if the buffer is never read. That also gives the programmer the possibility of ignoring certain interrupts.

The interrupts should/could be cleared by writing to the flag itself or by writing to an interrupt clear register. The first option I think is a bit more complicated to write, the second uses a few more resources.

stnolting, 2022-07-28:
The NEORV32 internal interrupt are a bit "odd" I think... Whenever a certain interrupt condition is met, the according interrupt signal fires (1 cycle high). Then the interrupt becomes pending inside the CPU. The according interrupt handler then has to clear this "pending interrupt flag" in the CPU so it can trigger again.

Back to the SLINK. How about this: per-channel interrupt configuration plus a status register to check the FIFO level of each channel.

The modified SLINK register map would look something like this:
```
CTRL:
- enable/reset[1]      (r/w)
- rx_log2_fifo_size[4] (r/-)
- tx_log2_fifo_size[4] (r/-)

IRQ:
- rx_irq_mode[16] (r/w, 2-bit for each RX channel)
  - 00: off
  - 01: off (unused)
  - 10: trigger IRQ when FIFO(i) goes from "empty" to "not empty"
  - 11: trigger IRQ when FIFO(i) becomes "at least half full"
- tx_irq_mode[16] (r/w, 2-bit for each RX channel)
  - 00: off
  - 01: off (unused)
  - 10: trigger IRQ when FIFO(i) goes from "not empty" to "empty"
  - 11: trigger IRQ when FIFO(i) becomes "at least half full"

RX_STATUS:
- rx_empty[8] (r/-)
- rx_half[8]  (r/-)
- rx_full[8]  (r/-)
- rx_last[8]  (r/-)

TX_STATUS:
- tx_empty[8] (r/-)
- tx_half[8]  (r/-)
- tx_full[8]  (r/-)
- tx_last[8]  (r/w)

DATA[8]:
- LINK[8][32] (r/w)
```

betocool-prog, 2022-07-28:
I don't think the NEORV32 is odd in that sense, I've seen lots of other cases where you need to explicitly clear an interrupt before you can trigger it again.

What if instead of 16 IRQs (8 RX  + 8 TX) it's 2 IRQs total for SLINK with:
```
CTRL:
- enable/reset[1]      (r/w)
- rx_log2_fifo_size[4] (r/-)
- tx_log2_fifo_size[4] (r/-)

IRQ:
- rx_irq_mode[1] (r/w, 2-bit for each RX channel)
  - Channel[0..7] 
    - 00: off
    - 01: off (unused)
    - 10: trigger IRQ when FIFO(i) goes from "empty" to "not empty"
    - 11: trigger IRQ when FIFO(i) becomes "at least half full"
- tx_irq_mode[1] (r/w, 2-bit for each RX channel)
  Channel[0..7]  
    - 00: off
    - 01: off (unused)
    - 10: trigger IRQ when FIFO(i) goes from "not empty" to "empty"
    - 11: trigger IRQ when FIFO(i) becomes "at least half full"

RX_STATUS:
- rx_empty[8] (r/-)
- rx_half[8]  (r/-)
- rx_full[8]  (r/-)
- rx_last[8]  (r/-)

TX_STATUS:
- tx_empty[8] (r/-)
- tx_half[8]  (r/-)
- tx_full[8]  (r/-)
- tx_last[8]  (r/w)

DATA[8]:
- LINK[8][32] (r/w)
```
This would decrease the number of IRQ lines to the CPU, but then the programmer needs to handle them.

I haven't looked at the SLINK module in detail yet, so I don't know how complex the changes would be.

stnolting, 2022-07-28:
> This would decrease the number of IRQ lines to the CPU, but then the programmer needs to handle them.

There are just two IRQ lines from SLINK to the CPU (one for RX, one for TX). For example, the RX interrupt can be triggered by _any_ RX channel that matches the according interrupt configuration.

betocool-prog, 2022-07-29:
Oooops! My bad! That means that we were thinking the same thing about the register and IRQ structure.

I will try to combine my project this weekend with looking into the SLINK more in detail. Maybe something comes out of that. There's rain forecast, so.... my only interruption should be "Back To The Future III".

stnolting, 2022-07-29:
Let me summarize the most important points of the SLINK modifications I have in mind:

* Two new register `RX_STATUS` and `TX_STATUS` to expose all the FIFO's status signals (empty, at least half full, full).
* The FIFO status of each FIFO can optionally trigger the RX interrupt (for RX links) or the TX interrupt (for TX links)
* RX interrupt options:
  * FIFO _becomes_ empty
  * FIFO _becomes_ at least half full
* TX interrupt options:
  * FIFO _becomes_ empty
  * FIFO _becomes_ less than half full

If several RX/TX interrupt conditions occur at the same time the interrupt handler can check the `RX_STATUS` and `TX_STATUS` status registers to check which links actually triggered the interrupt(s). Having a "half full" interrupt condition should also allow to implement double buffering (as you have suggested).

I think I'll experiment with the SLINK module a bit and do a PR if there's something to show :wink:

> my only interruption should be "Back To The Future III

Great Scott! I love this one! :+1:

betocool-prog, 2022-07-31:
I cloned the slink_irq_rework locally. I'm having a look at the vhd module as well as doing some changes to the .c example to trigger and check the interrupts.

stnolting, 2022-07-31:
Great! Thanks for your help!

betocool-prog, 2022-07-31:
I updated my "[slink_irq_rework](https://github.com/betocool-prog/neorv32/tree/slink_irq_rework)" branch.

I got it to work in the end after reconfiguring the RX interrupt. I didn't test the TX interrupt much, but the data transfer works.
Summary: 
- main.c:
  - Added new IRQ tests at the end of the original tests
  - Modified the RX IRQ configuration
  - It's quite verbose! You should be able to play with a few values of blocks and loops.
- neorv32_test_setup_slink.vhd
  - I added this file to the slink_demo in case you wanted to try it out on your setup. I could not find an SLINK_demo .vhd project somewhere else. I'm happy for it to be removed at the final merge.
  - The FIFO is 256 words deep.
  - It uses the regular bootloader.

I haven't looked at the documentation yet... it took me a little while to get it working right! Let me know if you want me to look into anything else.

stnolting, 2022-07-31:
> I got it to work in the end after reconfiguring the RX interrupt. I didn't test the TX interrupt much, but the data transfer works.

Good to hear! I also made some more tests and I think everything works as expected. so I think we can merge this. 

> neorv32_test_setup_slink.vhd 

As far as I can see this is just used to implement a local SLINK echo, right? I am using the default NEORV32 testbench for that.

Thanks for your help and work!

closed 2022-07-31 by stnolting