stnolting/neorv32 #412: Verification status

Blebowski, 2022-09-13:
Hello,

we were looking at NEORV32 some time ago in a company where I work as possible candidate
for using it as RISCV CPU. In the end, we picked IBEX CPU from lowRISC instead. The main
reason was verification status.

Does NEORV32 have any verification TB (other than simple launching example SW) ? Has it been
verified compared to some instruction/cycle accurate RISCV model? Has it been simulated in
some commerical toolchain? What are the future directions of NEORV in terms of verification?

I open up this issue just out of curiosity about future directions of NEORV :)

stnolting, 2022-09-13:
Hey there!

> we picked IBEX CPU from lowRISC

I also like lowRISC. Ibex is a really great core! :+1:

> Does NEORV32 have any verification TB (other than simple launching example SW) ?

We do have a self-checking testbench in `sim`. It runs the [processor check](https://github.com/stnolting/neorv32/tree/main/sw/example/processor_check) program that outputs a specific string only if all tests have passed. But there isn't something like a formal verification of the CPU / the whole processor - just because I have no idea how to do that 😅

> Has it been verified compared to some instruction/cycle accurate RISCV model?

We are using the official [RISC-V architecture test framework](https://github.com/riscv-non-isa/riscv-arch-test) (the older 2.x version) to check the core's code base once a weak for compatibility (with the golden reference provided by RISC-V - obtained from SPIKE). See https://github.com/stnolting/neorv32-verif

> Has it been simulated in some commerical toolchain?

So, you ask if the core _can be simulated_ using commercial tools?

I am using Xilinx ISIM and Modelsim and I know that some others are also using Modelsim.

> What are the future directions of NEORV in terms of verification?

Good question! 😄

To be honest, I am not sure. Formal verification would be interesting, but at least for the moment this is out of scope (for me).

Currently, I am thinking about porting the (new) RISC-V architecture test framework, which is based on [RISCOF](https://riscof.readthedocs.io/en/stable/). It is a little bit more complex than the older 2.x framework, but much more capable. Basically, you create a YAML file of the core with all it's registers, features, IO stuff and everything and throw it into a RISC-V golden-reference simulator (SPIKE/Sail) together with test programs for all different instructions and functions. Then you execute the same on the device under test and compare the results.

Blebowski, 2022-09-13:
Hi, thanks for the reply.

For the methodology of verification, I was refering to something like:
 
Testbench with randomly generated program, and DUT state compared (e.g. post each instruction) with model state (e.g. model such as spike). IMHO it is important to have at least some constrained random generated program, since it can catch nice corner-cases. Also, having "code-coverage" (statement, branch, toggle) evaluated and at high numbers is important. Functional coverage (e.g. on all instruction types, and all operands used, or combinations of all valid operands, etc.) is also important.

Out of this, a lot can be done in open-source world (unfortunately, not all). GHDL supports PSL, which has functional coverage, so you can do that quite easily. As for the simulation framework, it does not really matter. Professional world is using UVM, but OSVVM will do too. As long as you use a good model (such as spike), it is just a matter of generating valid program, calling the model from simulation, and comparing with DUT (and catching error if not).

As for the code-coverage, there is no OS simulator which has decent code coverage implemented, I am doing some slow progress in this regard in: https://github.com/nickg/nvc/issues/474
I estimate that by the end of this year, I will have some first usable implementation available.

I completely did not mean formal verification. I am aware of Symbiosys available in OS domain, but I doubt about its complete usability (due to lack of complete SV language support), I don't know how well does it cope with VHDL and extraction of formal properties out of PSL. Sure that commercial tools like Jasper are out of scope of this project.

stnolting, 2022-09-20:
Sorry for the late reply...

I do not have a real background in verification, so setting up a self-checking testbench was the best I could come up with. Recently, I have been tinkering with the [RISCOF](https://riscof.readthedocs.io/en/stable/index.html) RISC-V Compatibility Framework to check the processor for RISC-V ISA compatibility.

> Testbench with randomly generated program, and DUT state compared [...]

The framework uses the test cases from [riscv-arch-test](https://github.com/riscv-non-isa/riscv-arch-test) and runs them on the DUT (the NEORV) and a "golden" reference like Sail or Spike. The framework does not check the DUT/REF state after each single operation/instruction and does not check all the possible architecture states (yet?). However, from a plain functional point of view, this provides a very confident "verification" approach (IMHO).

My port of the framework is available in [neorv32-riscof](https://github.com/stnolting/neorv32-riscof), but it is still work-in-progress.

> GHDL supports PSL, which has functional coverage, so you can do that quite easily. [...]

Code coverage would be great to check the quality of the RTL code. However, I am not very confident with this topic. I found some nice projects that use GHDL to do a simple coverage analysis, but I am not sure how "valuable" this data is.

> I am doing some slow progress in this regard in: https://github.com/nickg/nvc/issues/474 [...]

That sound interesting. I think I'll need to have a closer look at nvc.

mikaelsky, 2022-11-30:
To add a few cents to the verification status conversation.

I currently have the core running in xcelium with some caveats. To get a clean sim going some light touches where needed in a few of source files. E.g. using '-' for don't care values, while theoretically nice, ends up causing large amounts of 'X' propagation as '-' resolves to 'X' with almost any logical operator.
The current GHDL setup is configured to ignore x-prop, which can lead to challenges. Note that VHDL by nature will almost always have some elements of X-prop at time 0, this can be scripted out by injecting an disable x-prop command, run 1, then re-enabling x-prop.

@Blebowski is correct in randomized instruction testing is the next step up in validation of the core. The fact that the core passes the architecture test framework is impressive in itself and an amazing start.
Utilizing UVM is A way to achieve randomized instruction testing. Would maybe suggest exploring a test framework like cocotb. From experience UVM can be made to work for randomized instruction testing but with significant challenges. Cocotb might well be easier to get starting with.

The purpose of randomized instruction testing is to determine whether sequences of instructions, and in particular neighboring instructions, have.. unfortunate interference via side-effects, e.g. status flag updates delays that can impact a: sub; branch conditional sequence. These are not normally covered within a directed test framework like riscof.
Randomized testing also helps finding unfortunate assumptions in data conversion, rounding and bus transactors.

Next level up is to begin to add in randomizing of events surrounding the core, e.g. random interrupt injection during normal code operation. This is the "Here be dragons" part of processor developement.

stnolting, 2022-12-02:
> E.g. using '-' for don't care values, while theoretically nice, ends up causing large amounts of 'X' propagation as '-' resolves to 'X' with almost any logical operator.

This _should_ be he case only right after reset. After some cycles the whole circuit should have settled into defined states.

> @Blebowski is correct in randomized instruction testing is the next step up in validation of the core

Some time ago, I wrote something like an "instruction fuzzer" that basically executes all possible 32-bit instruction in some kind of sandbox (using RISC-V's user mode and physical memory protection). The intention was to find "opcode holes" that did not trigger an illegal instruction exception.

This was the best I could come up with as I am not familiar with formal verification 😅 

Using UVM/cocotb is still beyond my capabilities, but it is definitely on my todo list!

closed 2023-07-16 by stnolting