stnolting/neorv32 #418: Missing reset for counters

LucaZulberti, 2022-09-26:
Hi,

using your core in simulation I noticed that mcycle* csr registers are tied to X due to a missing initialization at reset in this process:

https://github.com/stnolting/neorv32/blob/d540612fde954ca2a54e4117e854cd5c4e816e5e/rtl/core/neorv32_cpu_control.vhd#L2280

Was this done intentionally?

Best regards,
Luca.

stnolting, 2022-09-26:
Hey there!

Yes, this was intentionally and is also annotated in the [Data Sheet / Machine Counter and Timer CSRs](https://stnolting.github.io/neorv32/#_machine_counter_and_timer_csrs).

By not providing a reset for all the counter CSRs the requirements for the reset network are relaxed as there are up to 28+2 counter CSRs (`mcycle`, `minstret`, 28x `hpmcounter*`). Each counter is 64-bit wide resulting in up to 30*64=1920 flip flops potentially requiring a reset.

On most FPGAs the reset "comes for free" but I have also seen routing issues when having a large reset network (especially on the Lattice ice40 architecture).

However, maybe we should add a dedicated reset here (if only for `mcycle` and `minstret`) as these CSRs are the only ones right now that do not have a dedicated hardware reset. 🤔

LucaZulberti, 2022-09-27:
Thank you for your reply! Sorry I didn't read that in the docs.
If there aren't any other implications, I think it can be kept as it is to save resources.
I will add the initialization of these registers in my startup code, thank you!

stnolting, 2022-09-27:
Btw, the default startup code in [`sw/common/crt0.S`](https://github.com/stnolting/neorv32/blob/main/sw/common/crt0.S) already provides an intialization of `mcycle` and `minstret`:
https://github.com/stnolting/neorv32/blob/a978b64097ac68a50a213b58a59bc216ac27dea0/sw/common/crt0.S#L79-L82

LucaZulberti, 2022-09-27:
Thanks, I integrated the core into an existing development environment, but I should have looked better at your :-D

We can close this issue ;-)

closed 2022-09-27 by LucaZulberti