stnolting/neorv32 #422: crt0

Andk56, 2022-10-04:
Good afternoon.
As far as I understood crt0 for bootloader is different from .exe. But. The object file is not changed during compilation. Accordingly, the situation of using crt0, which is not quite suitable for the application, is real, if you do not use make clean_all.
Is it possible to enter crt0.S.o validation for bootloader or for .exe?

stnolting, 2022-10-04:
Hey there!

> As far as I understood crt0 for bootloader is different from .exe.

That is correct. When compiling the bootloader, the `make_bootloader` symbol is defined that omits some initialization parts from `crt0` (mainly calling constructors and destructors). The idea behind this is to save some ROM space.

> Accordingly, the situation of using crt0, which is not quite suitable for the application, is real, if you do not use make clean_all.

That is correct. You should do `clean_all` once when developing a "normal" executable.

> Is it possible to enter crt0.S.o validation for bootloader or for .exe?

I am not sure what you mean. Validation of what? :wink:

Andk56, 2022-10-04:
Hello Stephan!

I meant the correctness of crt0.o for the current program.
That is, now I am simultaneously making a bootloader for my needs and a program. I accidentally found out that crt0 does not behave the way I wanted and found out what I described above.
Actually the question is whether it is possible to check the correspondence of crt0.o for bootloader or exe using make tools. Or make different crt0 names for bootloader and exe.

stnolting, 2022-10-04:
Now I know what you mean :wink:

> the correctness of crt0.o for the current program.

Basically, it should not matter if the bootloader is compiled without the `make_bootloader` being defined - there will be just some more unused code eating up the spare boot ROM.

> I accidentally found out that crt0 does not behave the way I wanted and found out what I described above.

Can you give some more details? What was not working as expected?

> Actually the question is whether it is possible to check the correspondence of crt0.o for bootloader or exe using make tools. 

One option could be to check if the final ELF contains the constructor setup code from `crt0`:

https://github.com/stnolting/neorv32/blob/4d5952af04cc5c52b90c977d8d3067a5070cf5d5/sw/common/crt0.S#L166

You can check for this using the `elf_info` make target:

```bash
$  make elf_info | grep 'crt0_call_constructors'
```

If the ELF was **not** compiled for the bootloader you should get something like this:
```
neorv32/sw/example/blink_led$ make elf_info | grep 'crt0_call_constructors'
000000d8 l       .text  00000000 __crt0_call_constructors
000000e0 l       .text  00000000 __crt0_call_constructors_loop
000000f4 l       .text  00000000 __crt0_call_constructors_loop_end
```

If the ELF **was** compiled for the bootloader `grep` should not print anything.

Andk56, 2022-10-04:
Thank you for your feedback, it looks like I created the problem myself. :)
I needed the bootloader to be activated again after the end of the user program.
I added extra code after the __crt0_main_aftermath_end: label, and didn't notice that it was in the #ifndef make_bootloader conditional compilation.
Looks like I need to move my code below the corresponding #endif.

__crt0_main_aftermath_end:
 la ra, 0xFFFF0000    //
 jalr ra                     // execute bootloader again

#endif      <<<!!!!

Now, if I don't do make clean_all, then my code is simply missing from the crt0 object code.
Thank you for pointing out the make_bootloader key.

The question is closed, it's my mistake.

closed 2022-10-04 by Andk56

stnolting, 2022-10-04:
If you just want to restart the bootloader when your main application returns, you can use the "after main handler":
[📚 Data Sheet: CRT0 - After-Main Handler](https://stnolting.github.io/neorv32/#_after_main_handler)

Basically, you could add the following code to your main program:
```c
/**********************************************************************//**
 * "after-main" handler that is executed after the application's
 * main function returns (called by crt0.S start-up code)
 **************************************************************************/
void __neorv32_crt0_after_main(int32_t return_code) {

  asm volatile ("li ra, %[imm] \n"
                "jr 0(ra)      \n"
                : : [imm] "i" (BOOTLOADER_BASE_ADDRESS));
}
```

Of course, you can also modify the default crt0 code, but this approach does not change the core code base.