stnolting/neorv32 #432: SPI MOSI produces spikes

andkae, 2022-10-19:
Hi Stephan,

at the moment i do some verification measurements on NEORV32. There i saw this on the SPI MOSI line:

[image]

[image]

There you see transistions with 12ns - spikes.  The Shift clock is 25MHz, 40ns. From the functional point of view should this cause no issues, but it produces additional EMC noise on the line. In my opinion is this caused by the tap connection on the SREG, see below:

https://github.com/stnolting/neorv32/blob/d63b7c0c8f4b6f07d561986046aa448181cd4cea/rtl/core/neorv32_spi.vhd#L410-L419

My suggestions is to assign to the output always the MSB of the SREG and the spi data width configuration is honored by the bit pos of output data word load:

 * 8Bit DW: Bit31:Bit24 is loaded
 * 16Bit DW: Bit31:Bit16
 * 32bit DW: load full word

What Do you think?

BR,
Andreas

andkae, 2022-10-19:
This could be the line where you sort in the data right pos aligned:

https://github.com/stnolting/neorv32/blob/d63b7c0c8f4b6f07d561986046aa448181cd4cea/rtl/core/neorv32_spi.vhd#L269

If you like i can prepare an PR.

GideonZ, 2022-10-19:
True, but adding a wide multiplexer is relatively expensive in terms of
logic cells. Another solution is to simply add a register on the output of
SCLK and MOSI if the timing with MISO allows it? Or... Implement the
multiplexer only for MSB, i.e. load it with bit 30, 22, 14 or 6 from the
current shift register, or from bit 31, 23, 15 or 7 from the source word.
(Manual retiming...)

Or, yet another solution is to always use 8 bit mode and solve the
selection all the way at the top, even before a write enters the fifo,
reducing the FIFO to 8 bits only, at the cost of needing 4 system clock
cycles to write a 32 bit value.

As always, an optimized solution is harder to read than a straight forward
one.

stnolting, 2022-10-19:
@andkae 

> There you see transistions with 12ns - spikes. The Shift clock is 25MHz, 40ns. From the functional point of view should this cause no issues, but it produces additional EMC noise on the line.

I cannot reproduce those nasty spikes with my FPGA setup here. Could you share your SPI configuration or some minimal example to reproduce those spikes?

Are those spikes occurring **within** a single SPI transmission or **in between** two consecutive transmissions?

I will have a look at the simulation again. Are you sure those spikes are caused by the processor itself?

@GideonZ 

> True, but adding a wide multiplexer is relatively expensive in terms of
logic cells. Another solution is to simply add a register on the output of
SCLK and MOSI if the timing with MISO allows it?

I agree :wink:

An additional register stage should work for most cases, but if you are using the high speed feature of the SPI module, the SPI clock toggles at half of the processor clock. Another register stage might hurt for those corner cases.

If the spikes are really caused by the SPI core we could just AND-gate the SPI data output signal so it is forced to zero if there is no real transaction in progress.

andkae, 2022-10-20:
Hi Stephan,

after changing:

1)

https://github.com/stnolting/neorv32/blob/7cf7eff7050e7df1fe17666546c87d6cd16c0210/rtl/core/neorv32_spi.vhd#L269

to

```vhdl
with ctrl(ctrl_size1_c downto ctrl_size0_c) select tx_fifo.wdata <=
    data_i(07 downto 00) & x"000000" when "00",
    data_i(15 downto 00) & x"0000"   when "01",
    data_i(23 downto 00) & x"00"     when "10",
    data_i                           when others;
```

2)

https://github.com/stnolting/neorv32/blob/7cf7eff7050e7df1fe17666546c87d6cd16c0210/rtl/core/neorv32_spi.vhd#L410-L419

to

```vhdl
-- transmission size --
data_size: process(ctrl, rtx_engine)
begin
case ctrl(ctrl_size1_c downto ctrl_size0_c) is
  when "00"   => rtx_engine.bytecnt <= "001"; -- 8-bit mode
  when "01"   => rtx_engine.bytecnt <= "010"; -- 16-bit mode
  when "10"   => rtx_engine.bytecnt <= "011"; -- 24-bit mode
  when others => rtx_engine.bytecnt <= "100"; -- 32-bit mode
end case;
end process data_size;
spi_sdo_o <= rtx_engine.sreg(31);
```
I see no more spikes on the lines

[image]

This solution works, the question is with the ressource allocation....

andkae, 2022-10-20:
> Could you share your SPI configuration

This for start.

```c
	// Flow Control
neorv32_spi_init(&g_neorv32_spi);
	// SPI Core
neorv32_spi_disable();
	// neorv32_spi_setup(uint8_t prsc, uint8_t clk_phase, uint8_t clk_polarity, uint8_t data_size)
neorv32_spi_setup(0, 0, 0, 0, 3);  // spi mode 0, 8bit, 11: TX fifo empty
neorv32_spi_enable();
```

andkae, 2022-10-20:
Following code could avoid the big Mux:

```vhdl
-- transmission size --
data_size: process(ctrl, rtx_engine)
begin
case ctrl(ctrl_size1_c downto ctrl_size0_c) is
    when "00"   => rtx_engine.bytecnt <= "001"; spi_sreg_sel_c <= "0001"; -- 8-bit mode
    when "01"   => rtx_engine.bytecnt <= "010"; spi_sreg_sel_c <= "0010"; -- 16-bit mode
    when "10"   => rtx_engine.bytecnt <= "011"; spi_sreg_sel_c <= "0100"; -- 24-bit mode
    when others => rtx_engine.bytecnt <= "100"; spi_sreg_sel_c <= "1000"; -- 32-bit mode
end case;
end process data_size;

process(rstn_i, clk_i)
  begin
    if (rstn_i = '0') then
      spi_sreg_sel <= "1000"
    elsif rising_edge(clk_i) then
      spi_sreg_sel <= spi_sreg_sel_c
    end if;
end;

spi_dos(0) <= spi_sreg_sel(0) and rtx_engine.sreg(07); -- 8-bit mode
spi_dos(1) <= spi_sreg_sel(1) and rtx_engine.sreg(15); -- 16-bit mode
spi_dos(2) <= spi_sreg_sel(2) and rtx_engine.sreg(23); -- 24-bit mode
spi_dos(3) <= spi_sreg_sel(3) and rtx_engine.sreg(31); -- 32-bit mode
spi_sdo_o  <= or_reduce(spi_dos);
```

stnolting, 2022-10-20:
I still do not understand the cause of those spikes... 😅 

Could you share some more of your program code? I think the spikes might be caused by some SPI module register accesses in between actual SPI transmissions.

-------------------------------

This is the original code that derives the SDI output signal:

```vhdl
  -- transmission size --
  data_size: process(ctrl, rtx_engine)
  begin
    case ctrl(ctrl_size1_c downto ctrl_size0_c) is
      when "00"   => rtx_engine.bytecnt <= "001"; spi_sdo_o <= rtx_engine.sreg(07); -- 8-bit mode
      when "01"   => rtx_engine.bytecnt <= "010"; spi_sdo_o <= rtx_engine.sreg(15); -- 16-bit mode
      when "10"   => rtx_engine.bytecnt <= "011"; spi_sdo_o <= rtx_engine.sreg(23); -- 24-bit mode
      when others => rtx_engine.bytecnt <= "100"; spi_sdo_o <= rtx_engine.sreg(31); -- 32-bit mode
    end case;
  end process data_size;
```

The only change to make `spi_sdo_o` emit sort high-pulses is changing the `ctrl(ctrl_size1_c downto ctrl_size0_c)` or altering `rtx_engine.sreg`.

`rtx_engine.sreg` is static in between SPI transmissions, so this can't the be cause.

So I think that your program code might interact with `ctrl(ctrl_size1_c downto ctrl_size0_c)` that causes spikes on `spi_sdo_o` by selecting different bits of `rtx_engine.sreg`. 🤔

andkae, 2022-10-21:
Hi Stephan,

> The only change to make `spi_sdo_o` emit sort high-pulses is changing the `ctrl(ctrl_size1_c downto ctrl_size0_c)` or altering `rtx_engine.sreg`.

Me too :) I look also in my code but there are no CTRL reg accesses, except CSN. Give me some time i strip down my firmware, probably i see then the spikes on my site with this KISS example.

stnolting, 2022-10-21:
> I look also in my code but there are no CTRL reg accesses, except CSN

If just the CS bits are altered, there should be absolutely no impact on the SDO line.

> But altering rtx_engine.sreg should trigger the next transfer cycle?

There is no direct access to `rtx_engine.sreg`.

Application software can write data to `NEORV32_SPI.DATA`, which will put the data (a full 32-bit word) to the TX FIFO:

https://github.com/stnolting/neorv32/blob/7cf7eff7050e7df1fe17666546c87d6cd16c0210/rtl/core/neorv32_spi.vhd#L304-L306

When the TX FIFO is not empty AND the serial transmission engine is idle, a full 32-bit word is fetched from the TX FIFO and written to `rtx_engine.sreg`:

https://github.com/stnolting/neorv32/blob/7cf7eff7050e7df1fe17666546c87d6cd16c0210/rtl/core/neorv32_spi.vhd#L336

https://github.com/stnolting/neorv32/blob/7cf7eff7050e7df1fe17666546c87d6cd16c0210/rtl/core/neorv32_spi.vhd#L352-L359

Then the serial transmission engine will transmit the number of selected bits (8, 16, 24 or 32) starting with the according MSB by doing left shifts.

> And the point what me confuses me, when i run the FW in the Modelsim at RTL there are absolute no spikes.

That's the really strange point 😄
I cannot see any spike in the simulator, too. Are you sharing your FPGA's SPI pins with some other FPGA-internal logic? For example, I multiplex the chip's SPI lines for the SPI module and the XIP module.

andkae, 2022-10-21:
> Are you sharing your FPGA's SPI pins with some other FPGA-internal logic?

No. Directly to the Pins.

I prepare a strip-down FW. But it will take a while, im the next weeks not available.

stnolting, 2022-10-21:
No hurry!
And thanks for evaluating 😉

andkae, 2022-11-07:
Hi Stephan, here the example code:

```c
/** Includes **/
/* Standard libs */
#include <string.h>     // memset

/* NEORV32 Framework */
#include <neorv32.h>
#include "neorv32_spi_irq.h"    // IRQ driven dataflow on SPI 

/** global variables **/
/* SPI Flash */
    // NEORV32 SPI Driver 
t_neorv32_spi       g_neorv32_spi;                                  // common storage element for neorv32 spi

/**
 *  spi_irq_handler
 *    SPI interrupt handler
 */
void spi_irq_handler(void) 
{
    /* service SPI HW */
    neorv32_spi_isr(&g_neorv32_spi);
}

/**
 *  main
 */
int main()
{
    uint8_t     uint8MemBuf[20];

    /* NEORV32 Init */
    neorv32_uart0_setup(115200, PARITY_NONE, FLOW_CONTROL_NONE);    // setup UART
    neorv32_rte_setup();                                            // enable error capture
    
    /* Install RTE Handlers */
    neorv32_rte_exception_install(SPI_RTE_ID, spi_irq_handler);     // SPI to RTE
    
    /* enable FIRQ */
    neorv32_cpu_irq_enable(SPI_FIRQ_ENABLE);    // FIRQ6: SPI nterrupt
    
    /* enable global interrupt */
    neorv32_cpu_eint();                         // enable global interrupts
        
    /* Entry Message */ 
    neorv32_uart0_printf("SPI-Spike-Demo\n");

    /* SPI Flash */
        // Flow Control
    neorv32_spi_init(&g_neorv32_spi);
        // SPI Core
    neorv32_spi_disable();
        // neorv32_spi_setup(uint8_t prsc, uint8_t clk_phase, uint8_t clk_polarity, uint8_t data_size)
    neorv32_spi_setup(0, 0, 0, 0, 3);  // spi mode 0, 8bit, 11: TX fifo empty
    neorv32_spi_enable();
        
    /* Demo Transfer */
    for ( uint8_t i=0; i<32; i++ ) {
        /* Preapre */
        memset(uint8MemBuf, 0, sizeof(uint8MemBuf)/sizeof(uint8MemBuf[0])); // make empty
        uint8MemBuf[0] = 0x03;  // uint8FlashIstRdData
        uint8MemBuf[2] = i;     // middle byte of address, is page
        /* Request SPI */
            // int neorv32_spi_rw(t_neorv32_spi *self, void *spi, uint8_t csn, uint32_t num_elem, uint8_t data_byte)
        neorv32_spi_rw(&g_neorv32_spi, uint8MemBuf, 0, 12, 1);
        /* Wait for processing */
        while ( neorv32_spi_rw_busy(&g_neorv32_spi) ) {
            __asm("nop");
        }
    }
        
    /* stop programm counter */
    while ( 1 ) { }
    return 0;
}
```

The belonging spikes are still present, but i would more say it's one cycle of system clock

[image]

BR,
Andreas

stnolting, 2022-11-07:
Hey Andreas.

I just tested the code you posted. In my (behavioral) simulation (default testbench, running for 1ms) there are no "spikes" at all. The data output signal of the SPI controller (`sdo`) only changes it's level at the falling edge of the `sck` signal and keeps it's level for at least one full clock period of `sck`.

You code example is quite complex. Do you have a minimalist one? For example just single SPI transmission that makes those spikes to appear?

andkae, 2022-11-08:
Hi Stephan,

> I just tested the code you posted. In my (behavioral) simulation (default testbench, running for 1ms) there are no "spikes" at all.
Same for me, in RTL sim i see also not such an behaviour.

> Do you have a minimalist one?
How good this example look? Writing directly to the data register? Skipping the ISR? What do you have in mind?

In addition i can also try to make an timing simulation. Perhaps is there anything to see.

BR,
Andreas

stnolting, 2022-11-09:
> How should this example look? Writing directly to the data register? Skipping the ISR? What do you have in mind?

I am not sure. I have tested several "invalid" cases (like writing the data register when a transmission is in progress while not having a real FIFO) but still I cannot find any "spikes" in the functional simulation.

> In addition i can also try to make an timing simulation. Perhaps is there anything to see.

That might be a good idea :wink:

Btw, may I ask what kind of FPGA you are using?

stnolting, 2023-01-21:
Can we close this?

andkae, 2023-01-22:
Hi Stephan, yes close it. Through limit capacity i had no time to do further investigations. Perhaps i reopen it, when i have new inputs.

BR,
Andreas

closed 2023-01-22 by andkae