stnolting/neorv32 #444: XIRQ: irq_run inhibits next detection

andkae, 2022-11-20:
Hi Stephan,

for I2C communication is use the [IICMB](https://github.com/sshuv/iicmb). This controller clears automatically the IRQ after register access. When two IRQs come in a very short time difference, then is the seqend IRQ forward to the FIRQ inhibited by _irq_run_. I use core revision 15276eb5. 

[image]

i use following XIRQ config:
```vhdl
XIRQ_NUM_CH                  => 3,  -- number of external IRQ channels (0..32)
XIRQ_TRIGGER_TYPE            => x"ffffffff",   -- trigger type: 0=level, 1=edge
XIRQ_TRIGGER_POLARITY        => x"ffffffff",   -- trigger polarity: 0=low-level/falling-edge, 1=high-level/rising-edge
```

In my opinion could help a caputring of incoming edges. In case of irq_run is active, the next incoming edge is captured and after irq_run is low, fires the controller again to the FIRQ. 

Idea for Implementation: The Rise edge of _irq_run_ clears the RS-FF, with dominant set input. if a new IRQ edge arrives, the RSFF will not cleared and will start the next FIRQ after processing the last.

With the current behaviour hangs up my SW stack cause the one IRQ to go forward is missing.

Whats your opinion?

BR,
Andreas

stnolting, 2022-11-20:
I think this is a general problem: what if the interrupt fires again before the actual handler has started?

Of course we could add some kind of queue to capture further interrupts, but then we would need a buffer with an something like an infinite depth to make sure we capture _all_ interrupts at _any_ time...

This is what the XIRQ currently does:
* if an external interrupt fires, this is stored into an internal buffer (`irq_buf`)
* if any bit in this buffer is set the CPU interrupt is triggered
* the CPU starts the interrupt handler, which acknowledges the external interrupt and also clears the "pending" external interrupt source (inside the XIRQ module)

If the same interrupt that triggered the CPU interrupt fires again **before** the interrupt handler acknowledges the current IRQ, this interrupt trigger is lost, unfortunately.

One option might be to strip down the complexity of your interrupt handler (less code in the actual handler function).

> Idea for Implementation: The Rise edge of irq_run clears the RS-FF, with dominant set input. if a new IRQ edge arrives, the RSFF will not cleared and will start the next FIRQ after processing the last.

Without the "locking" mechanism of the `irq_run` signal we cannot ensure to precisely capture the actual source of the external interrupt that has triggered the current CPU interrupt.

A dirty workaround might be to connect your external interrupt source to two XIRQ inputs. But in the worst case this might give you two CPU interrupts were only one should occur.

GideonZ, 2022-11-20:
Hi Andreas,

So basically what you are suggesting is implementing a level triggered
interrupt. (I think latching an edge is exactly that..)

I will be honest, exactly for the reason you are describing, I have never
really understood the benefit of edge triggered interrupts. You can always
implement edge triggering with level triggered interrupts using software,
but you can't implement level triggered interrupts if you only have edge
triggered interrupt pins.

Best regards,
Gideon

GideonZ, 2022-11-20:
Correction to my previous statement:

Edge triggered interrupts are needed for sources that output a pulse.
Peripherals usually output a 'condition', saying "serve me", and it keeps saying this until the condition is cleared by software. In these cases you _thus_ use a level triggered interrupt. That includes the IICMB.

Gideon

andkae, 2022-11-21:
Hi Gideon, good hint, that solves for me :)

Thanks.

BR,
Andreas

closed 2022-11-21 by andkae