stnolting/neorv32 #459: make pdfin build folder not work

quantrpeter, 2022-12-18:
Hi all, please help

```
/home/peter/workspace/neorv32/docs>make pdf
[ -f revnumber.txt ] && REVNUMBER='-a revnumber='"$(cat revnumber.txt)" || unset REVNUMBER; \
asciidoctor-pdf $REVNUMBER \
  -a allow-uri-read \
  -a pdf-theme=neorv32-theme.yml \
  -r asciidoctor-diagram \
  datasheet/main.adoc \
  --out-file public/pdf/NEORV32.pdf
Traceback (most recent call last):
        6: from /usr/bin/asciidoctor-pdf:25:in `<main>'
        5: from /usr/lib/ruby/vendor_ruby/asciidoctor/cli/options.rb:274:in `parse!'
        4: from /usr/lib/ruby/vendor_ruby/asciidoctor/cli/options.rb:274:in `each'
        3: from /usr/lib/ruby/vendor_ruby/asciidoctor/cli/options.rb:276:in `block in parse!'
        2: from /usr/lib/ruby/vendor_ruby/polyglot.rb:65:in `require'
        1: from /usr/lib/ruby/2.7.0/rubygems/core_ext/kernel_require.rb:92:in `require'
/usr/lib/ruby/2.7.0/rubygems/core_ext/kernel_require.rb:92:in `require': cannot load such file -- asciidoctor-diagram (LoadError)
        20: from /usr/bin/asciidoctor-pdf:25:in `<main>'
        19: from /usr/lib/ruby/vendor_ruby/asciidoctor/cli/options.rb:274:in `parse!'
        18: from /usr/lib/ruby/vendor_ruby/asciidoctor/cli/options.rb:274:in `each'
        17: from /usr/lib/ruby/vendor_ruby/asciidoctor/cli/options.rb:276:in `block in parse!'
        16: from /usr/lib/ruby/vendor_ruby/polyglot.rb:65:in `require'
        15: from /usr/lib/ruby/2.7.0/rubygems/core_ext/kernel_require.rb:156:in `require'
        14: from /usr/lib/ruby/2.7.0/rubygems/core_ext/kernel_require.rb:168:in `rescue in require'
        13: from /usr/lib/ruby/2.7.0/rubygems/core_ext/kernel_require.rb:168:in `require'
        12: from /var/lib/gems/2.7.0/gems/asciidoctor-diagram-2.2.3/lib/asciidoctor-diagram.rb:1:in `<top (required)>'
        11: from /var/lib/gems/2.7.0/gems/asciidoctor-diagram-2.2.3/lib/asciidoctor-diagram.rb:1:in `require_relative'
        10: from /var/lib/gems/2.7.0/gems/asciidoctor-diagram-2.2.3/lib/asciidoctor-diagram/a2s.rb:2:in `<top (required)>'
         9: from /var/lib/gems/2.7.0/gems/asciidoctor-diagram-2.2.3/lib/asciidoctor-diagram/a2s.rb:2:in `require_relative'
         8: from /var/lib/gems/2.7.0/gems/asciidoctor-diagram-2.2.3/lib/asciidoctor-diagram/a2s/extension.rb:1:in `<top (required)>'
         7: from /var/lib/gems/2.7.0/gems/asciidoctor-diagram-2.2.3/lib/asciidoctor-diagram/a2s/extension.rb:1:in `require_relative'
         6: from /var/lib/gems/2.7.0/gems/asciidoctor-diagram-2.2.3/lib/asciidoctor-diagram/a2s/converter.rb:2:in `<top (required)>'
         5: from /var/lib/gems/2.7.0/gems/asciidoctor-diagram-2.2.3/lib/asciidoctor-diagram/a2s/converter.rb:2:in `require_relative'
         4: from /var/lib/gems/2.7.0/gems/asciidoctor-diagram-2.2.3/lib/asciidoctor-diagram/util/cli_generator.rb:2:in `<top (required)>'
         3: from /usr/lib/ruby/vendor_ruby/polyglot.rb:65:in `require'
         2: from /usr/lib/ruby/2.7.0/rubygems/core_ext/kernel_require.rb:92:in `require'
         1: from /usr/lib/ruby/2.7.0/rubygems/core_ext/kernel_require.rb:92:in `require'
/usr/lib/ruby/2.7.0/tempfile.rb:121:in `<top (required)>': superclass mismatch for class Tempfile (TypeError)
make: *** [Makefile:9: pdf] Error 1
```

stnolting, 2022-12-18:
This looks similar to https://github.com/stnolting/neorv32/issues/250, so it might be a problem with your asciidoctor installation.

Does `make html` work as expected?

Btw, you can find prebuilt PDF version of the latest data sheet [here](https://github.com/stnolting/neorv32/releases/tag/nightly).

quantrpeter, 2022-12-18:
it works in other ubuntu now, thanks

closed 2022-12-18 by quantrpeter